zephyrproject-rtos/zephyr #56595: wifi: Scan results dumping broken with new net_mgmt event design

chaitu-tk, 2023-04-06:
**Describe the bug**
PR https://github.com/zephyrproject-rtos/zephyr/pull/54701 has modified the network management event processing design, the event additions and callbacks can now run in parallel, so, if the callbacks take more time then the event queue can overflow easily. Previously the event additions used to wait for the lock to be release by callbacks, so, event queue will not overflow.

This is quite evident with Wi-Fi scan, the shell prints the results to UART which takes time causing queue overflow.

**To Reproduce**
Steps to reproduce the behavior: (I have reproed using downstream Wi-Fi chipset, so, these steps are untested)
1. Build `samples/net/wifi/`
2. Issue `wifi scan` in a crowded environment
3. You can see the queue overflow warnings and scan results are too low (as events are dropped)

**Expected behavior**
No warnings and high scan results.

**Impact**
Mainly those samples that print scan results to UART.

**Logs and console output**
```
 wifi scan
Scan requested
uart:~$ 17   | PHY_TEAM_ASUS24                  15    | 11   (2.4GHz) | -25  | WPA2-PSK        | 24:4B:FE:AD:EB:B0
uart:~$ 18   | VHT-5.2.1                        9     | 36   (5GHz  ) | -35  | UNKNOWN         | 8C:FD:F0:83:44:17
uart:~$ 19   | NETGEAR54                        9     | 1    (2.4GHz) | -35  | WPA2-PSK        | 9C:C9:EB:FA:E9:D8
uart:~$ 20   | ASUS_RTAX88U24G_11ax20WPA3SAE2   30    | 7    (2.4GHz) | -37  | WPA3-SAE        | 3C:7C:3F:DA:DF:38
uart:~$ 21   | ASUS_RTAX88U24G_11ax20WPA3SAE76  31    | 4    (2.4GHz) | -38  | WPA3-SAE        | A0:36:BC:B6:BD:E8
uart:~$ 22   | Broadcom                         8     | 11   (2.4GHz) | -39  | OPEN            | 00:90:4C:27:F0:B6
uart:~$ 23   | ASUS_RTAX88U24G_11ax20WPA3SAE80  31    | 3    (2.4GHz) | -40  | WPA3-SAE        | 3C:7C:3F:54:8D:50
uart:~$ 24   | NETIS_WF278024G11g20WPA2         24    | 1    (2.4GHz) | -40  | WPA2-PSK        | 08:10:74:00:00:01
uart:~$ 25   | ASUS_twt_divya                   14    | 9    (2.4GHz) | -41  | WPA2-PSK        | A0:36:BC:14:E7:70
uart:~$ 26   | ASUS_RTAX88U5G_11ax20WPA3SAE6    29    | 161  (5GHz  ) | -45  | WPA3-SAE        | A0:36:BC:56:41:E4
uart:~$ 27   | ASUS_RTAX88U5G_11ax20WPA3SAE53   30    | 153  (5GHz  ) | -46  | WPA3-SAE        | 3C:7C:3F:54:8D:54
uart:~$ [00:02:31.188,385] <wrn> net_mgmt: Failure to push event (3512074241), try increasing the 'CONFIG_NET_MGMT_EVENT_QUEUE_SIZE'
uart:~$ [00:02:31.199,218] <wrn> net_mgmt: Failure to push event (3512074241), try increasing the 'CONFIG_NET_MGMT_EVENT_QUEUE_SIZE'
uart:~$ [00:02:31.210,510] <wrn> net_mgmt: Failure to push event (3512074241), try increasing the 'CONFIG_NET_MGMT_EVENT_QUEUE_SIZE'
uart:~$ [00:02:31.222,564] <wrn> net_mgmt: Failure to push event (3512074241), try increasing the 'CONFIG_NET_MGMT_EVENT_QUEUE_SIZE'
uart:~$ [00:02:31.234,619] <wrn> net_mgmt: Failure to push event (3512074241), try increasing the 'CONFIG_NET_MGMT_EVENT_QUEUE_SIZE'
uart:~$ [00:02:31.245,971] <wrn> net_mgmt: Failure to push event (3512074241), try increasing the 'CONFIG_NET_MGMT_EVENT_QUEUE_SIZE'
uart:~$ [00:02:31.257,995] <wrn> net_mgmt: Failure to push event (3512074241), try increasing the 'CONFIG_NET_MGMT_EVENT_QUEUE_SIZE'
uart:~$ [00:02:31.270,080] <wrn> net_mgmt: Failure to push event (3512074241), try increasing the 'CONFIG_NET_MGMT_EVENT_QUEUE_SIZE'
uart:~$ [00:02:31.281,372] <wrn> net_mgmt: Failure to push event (3512074241), try increasing the 'CONFIG_NET_MGMT_EVENT_QUEUE_SIZE'
uart:~$ [00:02:31.293,457] <wrn> net_mgmt: Failure to push event (3512074241), try increasing the 'CONFIG_NET_MGMT_EVENT_QUEUE_SIZE'
uart:~$ [00:02:31.304,779] <wrn> net_mgmt: Failure to push event (3512074241), try increasing the 'CONFIG_NET_MGMT_EVENT_QUEUE_SIZE'
uart:~$ [00:02:31.316,833] <wrn> net_mgmt: Failure to push event (3512074241), try increasing the 'CONFIG_NET_MGMT_EVENT_QUEUE_SIZE'
uart:~$ [00:02:31.326,904] <wrn> net_mgmt: Failure to push event (3512074241), try increasing the 'CONFIG_NET_MGMT_EVENT_QUEUE_SIZE'
uart:~$ [00:02:31.336,975] <wrn> net_mgmt: Failure to push event (3512074241), try increasing the 'CONFIG_NET_MGMT_EVENT_QUEUE_SIZE'
uart:~$ [00:02:31.347,045] <wrn> net_mgmt: Failure to push event (3512074241), try increasing the 'CONFIG_NET_MGMT_EVENT_QUEUE_SIZE'
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.15.2
 - Downstream fork

@rlubos @JvanDooren

chaitu-tk, 2023-04-06:
Below are options to fix this:

1. Increase  `CONFIG_NET_MGMT_EVENT_QUEUE_SIZE` as suggested by the warning, but as UART is slow, this would mean this has to be very high, in our experiments, `150` works well (equal to the scan results).
2. Modify `wifi_shell` to batch the scan results and print them after scan is completed, but LOG_* macros might confuse the users because of timings and don't always print to UART depending on Kconfig. So, might need a `shell_fprintf_buffered` API.

rlubos, 2023-04-06:
It feels to me like a poor Wi-Fi shell design, just like with system workqueue, you should not hog on resources that are used system-wide (in this case `net_mgmt` thread time). 

Anyway, perhaps an easy fix for this would be to increase the allocation timeout for the net_mgmt events? Currently there's a fixed timeout of 10ms, but as this example proves it doesn't scale well. I'm taking about this line:
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/ip/net_mgmt.c#L82
I'd try to increase this value, and if it helps, consider adding a Kconfig entry for it.

trantanen, 2023-04-06:
Increasing timeout of 10ms seems to work. However, I guess I'd still think about not using net_mgmt thread for shell printing as all additional configs that we need to tweak for specific purposes that are not obvious, will be inconvenient.

krish2718, 2023-04-06:
> It feels to me like a poor Wi-Fi shell design, just like with system workqueue, you should not hog on resources that are used system-wide (in this case `net_mgmt` thread time).
>
> Anyway, perhaps an [...]

But doesn't it defeat the purpose of keeping adding and processing independent? We basically go back to old design where new events are waiting for lock and the current event is waiting for callbacks to complete and free up the queue.

>  However, I guess I'd still think about not using net_mgmt thread for shell printing as all additional configs that we need to tweak for specific purposes that are not obvious, will be inconvenient.

That is what I was suggesting in #2 in my proposed solution to avoid delaying net_mgmt thread either by finding a way batch UART access or as you suggest deferring prints to a workqueue or something.

rlubos, 2023-04-06:
It all boils down to whether we want to:
* Slow down the event generation at source, by increasing the allocation timeout,
* Or to allocate extra memory to store the event data in-between generation/print. UART is the bottleneck here, so even if the printing was handled by a separate thread/workqueue, you'd need some space to store the data to keep up with the quickly generated events. 

The question is, which tradeoff is a better choice here.

krish2718, 2023-04-06:
Yes, as this is not a real world problem (i.e., applications don't print out scan results to UART), we could simply bump the queue size to 200 (most in a crowded channel)?

rlubos, 2023-04-06:
This is one solution. But this also means, we need to allocate `200 * sizeof(struct mgmt_event_entry)` bytes of RAM statically, for this rare case you scan in a crowded environment. 

I'd be personally leaning towards making the timeout configurable, and extend it if Wi-Fi scan is used. Note, that such a massive event generation is a really rare scenario, and in regular operation the timeout does not matter anyway.

krish2718, 2023-04-06:
Yes, Wi-Fi scan is the stress test case for net_mgmt events. Based on previous profiling, each access to UART was taking about ~200ms (4-5 scan results), so, the timeout would be significantly larger, let me run few tests and check. @trantanen what was the timeout that solved the issue for you?

trantanen, 2023-04-06:
I have tried only with 1000ms timeout. In my office environment, we have ~35 APs available.

krish2718, 2023-04-06:
ok, let me try increasing the timeout and will raise a PR, thanks.

jukkar, 2023-04-06:
One additional solution is to have for example space for 10 entries and if that fills up, then lock the queue until there is more free space. Of course this would mean that after 10 entries, the queue would handle only 1 entry at a time. If that is not desired, then the queue could be flushed a bit more like for example 3-4 entries at a time.

closed 2023-04-28 by MaureenHelm