zephyrproject-rtos/zephyr #56604: [v2.7] application booted from v3.x mcuboot uptime resets

nordicjm, 2023-04-06:
**Describe the bug**
It seems that with https://github.com/zephyrproject-rtos/zephyr/pull/56528 applied to mcuboot, a 2.7-built application which uses uptime will ungracefully seem to lose it's uptime value after about 8.5 minutes, that is, the device will continue functioning but e.g. using `kernel uptime` from the shell will wrongly output an uptime value of seconds rather than 10 minutes which is wrong.

**To Reproduce**

 - Build mcuboot with above PR for nrf52840dk_nrf52840 and flash to board
 - Checkout v2.7-branch, west update, then build samples/subsys/shell/shell_module with mcuboot enabled and flash
 - Check `kernel uptime` and `kernel threads` output every minute, after the 8.5 minute mark, seemingly the uptime has set to 0.

**Expected behavior**
Uptime to continue incrementing

**Impact**
Showstopper, applications seemingly are unaware of the true uptime. Note that this only happens when older images e.g. 2.7 are booted from new versions of mcuboot e.g. from zephyr 3.x onwards

**Environment (please complete the following information):**
 - Commit SHA or Version used: `main` with above PR for mcuboot and `v2.7-branch` for application

cfriedt, 2023-04-11:
@nordicjm - any chance you could provide a fix?

nordicjm, 2023-04-12:
> @nordicjm - any chance you could provide a fix?

Not likely, can't determine cause

nordicjm, 2023-04-12:
@anangl is there a PR that needs to be ported?

anangl, 2023-04-12:
> @anangl is there a PR that needs to be ported?

I have no idea, to be honest. I don't know what could cause such behavior.

nordicjm, 2023-04-12:
Assigning to @anangl and @nordic-krch it seems to be something wrong with `drivers/timer/nrf_rtc_timer.c` and that file has been vastly changed in 3.x, if I add debug out then it seems to prevent the issue (albeit I get a continuous stream of output) so maybe something is not being handled properly like an overflow or next count or something.

nordicjm, 2023-06-12:
Ping on @anangl and @nordic-krch

anangl, 2023-06-12:
The issue is not actually related to mcuboot. It can be also observed in `samples/subsys/shell/shell_module` built with `v2.7-branch` without using mcuboot. The behavior is caused by specific construction of the application. Details in #59154.

PR #59154 (2023-06-13): [v2.7-branch] drivers: nrf_rtc_timer: Always set an initial timeout
In the tickless kernel mode, the nRF system timer does not schedule any timeout on initialization. This can lead to a situation that for certain applications no timeout is scheduled at all (for example, when an application does not create any threads, it exits `main()` without any sleeping and only handles interrupts) and in consequence `sys_clock_announce()` is never called (the nRF system timer calls this function only from the timeout handler). This in turn causes that uptime is reported correctly only until the RTC used as [... cut]
closed 2023-06-13 by anangl