zephyrproject-rtos/zephyr #56605: [ignore] [RFC] Add extensible version infrastructure

nordicjm, 2023-04-06:
### Problem description

At present, there is a way that applications can get various versions of things in zephyhr - they can get the version of zephyr that is checked out, they can get the version of a board that is being compiled for, but from an application perspective there is no unified way of getting a version of the application itself

### Problem description

Why do we want this change and what problem are we trying to address?

### Proposed change

A brief summary of the proposed change - the 10,000 ft view on what it will
change once this change is implemented.

## Detailed RFC

In this section of the document the target audience is the dev team. Upon
reading this section each engineer should have a rather clear picture of what
needs to be done in order to implement the described feature.

### Proposed change (Detailed)

This section is freeform - you should describe your change in as much detail
as possible. Please also ensure to include any context or background info here.
For example, do we have existing components which can be reused or altered.

By reading this section, each team member should be able to know what exactly
you're planning to change and how.

### Dependencies

Highlight how the change may affect the rest of the project (new components,
modifications in other areas), or other teams/projects.

### Concerns and Unresolved Questions

List any concerns, unknowns, and generally unresolved questions etc.

## Alternatives

List any alternatives considered, and the reasons for choosing this option
over them.

nordicjm, 2023-04-06:
Kconfig initial test PR: https://github.com/zephyrproject-rtos/zephyr/pull/55814
Alternative version file test PR: https://github.com/zephyrproject-rtos/zephyr/pull/56141

Laczen, 2023-04-07:
@nordicjm, from your RFC it is clear that there should be some definition of `CONFIG_MCUBOOT_APP_VERSION` that is used for signing an application.

I don't understand the need to promote this to an application version.

tejlmand, 2023-04-11:
as stated here:
> @tejlmand @nashif: too MCUboot-centered, let's make this more generic, decouple it from imgtool and MCUboot

ref: https://github.com/zephyrproject-rtos/zephyr/pull/55814#issuecomment-1496169537

this RFC is still too imgtool centric in its description as to why an application version should be introduced.

Many users of Zephyr are interested in an application version so it should be possible to describe such a feature for itself.

> the format of this field would not be inherently restricted to any specific format but format restrictions could be applied depending upon the signing system used if it has strict requirements (i.e. imgtool does not support alphanumeric characters as part of a version).

I strongly disagree here.
Zephyr should decide how an application version format can described, and users should be able to choose the fields they like to use semantic versioning, use of pre-release fields, like `rcX`.
If a signing tool has special restrictions compared to the version infrastructure provided by Zephyr, then it's the responsibility of such tool configuration to use just the portions of Zephyr app version that it understands, for example the `major.minor.patch` of such app version without the pre-release information.

No tool should restrict the version format that Zephyr version infrastructure supports.
Zephyr community should decide the version format we want to support, and then tools must use the field parts they understand or have their own version setting.

Laczen, 2023-04-11:
@tejlmand, I agree that there should be no restriction on the format of the application version, but I disagree that zephyr should decide on the application version format. This should be left to the user.

There is a very easy way out for this problem: define a `CONFIG_MCUBOOT_APP_VERSION` (as proposed by @nordicjm) and a `CONFIG_ZEPHYR_APP_VERSION` (as proposed by @tejlmand), ... and set (configurable) what is wanted into a `CONFIG_APPLICATION_VERSION`. This `CONFIG_APPLICATION_VERSION` can then be used by the user in reporting or storing the application version using the bindesc PR (##54464). For the signing `CONFIG_MCUBOOT_APP_VERSION` can be used, for something else the `CONFIG_ZEPHYR_APP_VERSION` could be used. When users wish to deviate from the proposed standards they can be defining there own `CONFIG_APPLICATION_VERSION`.

For applications that are using mcuboot the `CONFIG_APPLICATION_VERSION` could default be set to `CONFIG_MCUBOOT_APP_VERSION`.

For applications that are not using mcuboot the `CONFIG_APPLICATION_VERSION` could default be set to `CONFIG_ZEPHYR_APP_VERSION`.

nordicjm, 2023-04-11:
> If a signing tool has special restrictions compared to the version infrastructure provided by Zephyr, then it's the responsibility of such tool configuration to use just the portions of Zephyr app [...]

And this is where it causes a problem, if there is a bootloader which has one integer field for a version, this would mean that the major version of the application would be used. Now if we use an example based on a very popular piece of software, let's say windows 10, that has 14 major versions (to date) each of which must have thousands of minor versions - so we're saying it's fine to just completely bin the minor version because it's not needed? I think end users might disagree with that view, I know I do.

tejlmand, 2023-04-19:
>  I think end users might disagree with that view, I know I do.

i'm leaving it for the end-user to decide.
End users create their application and decides their versioning and their signing tool.

If the sign tool only supports a single digit and application version infrastructure supports `major.minor.patch-preRelease`, then the user must decide whether to use all fields or just a subset. Or if the signing tool should use a completely different numbering scheme. **That is for the user to decide**.

To me it appears that if the signing tool has special restrictions, then you want the Zephyr's app version infrastructure to be limited.
I disagree on that. Let the user decide. For the signing tool Zephyr natively supports and where the tool doesn't support all fields, there we can default the sign version of the tool based on a subset of app version.
And then it is for the user developing an application to decide if the app version only should use the fields supported by the tool, or how the signing version should be configured.

@nordicjm that means that in your case, you can choose to just use the `major.minor.patch` in your application, and imgtool will use those fields. No one forces you to use `-preRelease` field.

Another user can decide to use `major.minor.pacth-rcX` and if they are happy that imgtool uses only the` major.minor.patch` portion for signing, then everything is fine.
If they are unhappy, they must decide whether to only use `major.minor.patch` (as @nordicjm would do) or set the imgtool Kconfig sign version to what they want.

That has the benefit of re-using existing version infrastructure, and have a dedicated imgtool sign version with a sane default value.

I prefer having a good default value that will suit users needing just `major.minor.patch` with imgtool, while still allow end-users that wants additional app version information like `-rcX` to do so.
And I'm highly in favor of just a single implementation for creation of version header files and CMake variable handling, instead of multiple implementations.

nordicjm, 2023-05-02:
> Building on that, we should also consider how to embed the version in the binary header, so we can verify the version of a built binary.

@yonsch has a PR for adding binary descriptor support, it gets the version that it embeds from Kconfig: https://github.com/zephyrproject-rtos/zephyr/pull/54464/files#diff-cc6728571822866d8ccc320879d622948d055562d5affed1c824e0cc88addfa0

PR #56141 (2023-05-17): Application version handling.
This PR is inspired by #55814 proposal.

It tries to address the concerns raised in the review.

Main parts of this proposal :
- Zephyr version infrastructure is improved for better reuse, this allows a single infrastructure instead of duplicated files
  This means the same APP_VERSION_x variables similar to KERNEL_VERSION_x are available, providing consistency.
- APP_BUILD_VERSION is supported which includes extra build info through `git describe`, but executed in the application repo.
   For example:
   `#define KERNEL_BUILD_VERSION zephyr-v3.3.0-1081-gb6f97a492 [... cut]
closed 2023-05-17 by carlescufi