zephyrproject-rtos/zephyr #56610: BLE: ESP32C3: extended advertisement

pawelosyp-hid, 2023-04-06:
I have started development on M5Stamp C3U - based on ESP32-C3. After starting beacon sample I was trying to get an extended advertisement functionality. It turns out that there is no support for extended advertisement. Even I have tried to send raw command to controller (BT_HCI_OP_LE_READ_NUM_ADV_SETS) with responce command disallowed. It looks like the controller blob do not have enabled that feature.

As the ESP32-C3 supports extended advertisement the functionality should be exposed to Zephyr BLE host. 

Am I missing any config to enable that one or is there any ongoing work in that area?

sylvioalves, 2023-04-12:
Well, it should work. `samples/bluetooth/periodic_adv` uses extended advertising:  
```c
Starting Periodic Advertising Demo
[00:00:00.000,000] <inf> esp32_bt_adapter: esp_bt_controller_init: BT controller compile version [d913766]
[00:00:00.166,000] <inf> bt_hci_core: bt_dev_show_info: Identity: CD:96:80:55:E7:32 (random)
[00:00:00.166,000] <inf> bt_hci_core: bt_dev_show_info: HCI: version 5.0 (0x09) revision 0x0016, manufacturer 0x0060
[00:00:00.167,000] <inf> bt_hci_core: bt_dev_show_info: LMP: version 5.0 (0x09) subver 0x0016
Start Extended Advertising...done.
```

Does this sample code work as you need? Or you need something else?

pawelosyp-hid, 2023-04-19:
Thank you for the response, I had some wrong configs, and the demo you have proposed works.
Have you tried to run legacy advertisements through an extended API? The goal is to run multiple legacy advertising sets independent of the main application. 
I have observed that when in advertising parameter options are set as `BT_LE_ADV_OPT_CONNECTABLE | BT_LE_ADV_OPT_SCANNABLE` then I have got assert and I can not debug:
`assert co_utils.c 728, param 000000ff 000000ff`
The same situation is every time if there is no `BT_LE_ADV_OPT_EXT_ADV` option selected.

xyzzy42, 2023-06-03:
I'm able to duplicate this same assert, which is from the Espressif binary blob.

A trivial BLE example that does nothing but enable legacy advertising will trigger it, `bt_le_adv_start(BT_LE_ADV_CONN_NAME, ad, ARRAY_SIZE(ad), NULL, 0);`

If `BT_EXT_ADV` is disabled, then this works fine.  It Legacy advertises as it should.  But once `BT_EXT_ADV` is enabled, then the assert is triggered by the `bt_le_adv_start()` call.

OTOH, using extended advertising (only) does work, i.e. `bt_le_ext_adv_create()` + `bt_le_ext_adv_start()`.

I'm not aware that enabling extended advertising support in the build config is suppose to make the legacy advertising API unavailable.  It's certainly isn't documented anywhere I can find.  And even legacy advertising should be unavailable, I'd expect the call to `bt_le_adv_start()` to return `-ENOTSUP` rather than assert.

jadejr, 2024-07-09:
This seems to work now. I'm using 2 advertising sets with legacy advertising on the current zephyr main branch. I haven't yet tried it with 3.6.*'s release though.

jadejr, 2024-09-03:
It does work in 3.7.

sylvioalves, 2025-06-26:
Fix merged in #https://github.com/zephyrproject-rtos/zephyr/pull/85792

closed 2025-06-26 by sylvioalves