zephyrproject-rtos/zephyr #56613: zephyrproject-rtos/action-s3-cache@v1 is using deprecated features

gmarull, 2023-04-06:
**Describe the bug**

It looks like `zephyrproject-rtos/action-s3-cache@v1` is using deprecated Github features:

```
twister-build (2)Node.js 12 actions are deprecated. Please update the following actions to use Node.js 16: zephyrproject-rtos/action-s3-cache@v1.
twister-build (2)The `save-state` command is deprecated and will be disabled soon. Please upgrade to using Environment Files.
twister-build (2)The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files.
```

Action should be modernized before the deprecation period ends (May/Summer 2023)

Ref. https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
Ref. https://github.blog/changelog/2022-09-22-github-actions-all-actions-will-begin-running-on-node16-instead-of-node12/

**To Reproduce**
Steps to reproduce the behavior:
1. Go to any twister run page, e.g. https://github.com/zephyrproject-rtos/zephyr/actions/runs/4626953526

**Expected behavior**
No deprecated features are used

**Impact**
Just warnings for now, CI disruption when deprecation period ends.

**Logs and console output**
N/A

**Environment (please complete the following information):**
- Github CI

**Additional context**
N/A

stephanosio, 2023-04-13:
Note that `zephyrproject-rtos/action-s3-cache@v1` is not the only action using these deprecated commands:

* `backport.yml` / `zephyrproject-rtos/action-backport@v1.1.1-3` (e.g. https://github.com/zephyrproject-rtos/zephyr/actions/runs/4687971991)
* `clang.yml` / likely multiple actions; need in-depth investigation (e.g. https://github.com/zephyrproject-rtos/zephyr/actions/runs/4688050255)
* `stale_issues.yml` / `actions/stale@v3` (e.g. https://github.com/zephyrproject-rtos/zephyr/actions/runs/4684196269)
* `codecov.yaml` / likely multiple actions; need in-depth investigation (e.g. https://github.com/zephyrproject-rtos/zephyr/actions/runs/4641518076)
* `release.yml` / likely multiple actions; need in-depth investigation (e.g. https://github.com/zephyrproject-rtos/zephyr/actions/runs/4214467422)
* `doc-publish.yml` / likely multiple actions; need in-depth investigation (e.g. https://github.com/zephyrproject-rtos/zephyr/actions/runs/4687836026)
* `doc-publish-pr.yml` / likely multiple actions; need in-depth investigation (e.g. https://github.com/zephyrproject-rtos/zephyr/actions/runs/4688091557)
* `footprint-tracking.yml` / likely multiple actions; need in-depth investigation (e.g. https://github.com/zephyrproject-rtos/zephyr/actions/runs/4684756720)
* `issue_count.yml` / likely multiple actions; need in-depth investigation (e.g. https://github.com/zephyrproject-rtos/zephyr/actions/runs/4688102172)
* `bsim-tests-publish.yml` / likely multiple actions; need in-depth investigation (e.g. https://github.com/zephyrproject-rtos/zephyr/actions/runs/4687517225)
* `daily_test_version.yml` / `aws-actions/configure-aws-credentials@v1` (e.g. https://github.com/zephyrproject-rtos/zephyr/actions/runs/4683622111)
* `labeler.yml` / `actions/labeler@v2.1.1` (e.g. https://github.com/zephyrproject-rtos/zephyr/actions/runs/4637233263)
* `twister.yml` / likely multiple actions; need in-depth investigation (e.g. https://github.com/zephyrproject-rtos/zephyr/actions/runs/4687657040)
* `stale-workflow-queue-cleanup.yml` / `MajorScruffy/delete-old-workflow-runs@v0.3.0` (e.g. https://github.com/zephyrproject-rtos/zephyr/actions/runs/4679802293)

Changing to high priority since we only have approximately 1.5 month to fix all these and not doing so will result in catastrophic CI failures when these commands get completely disabled by GitHub.

Note that all the workflows using the affected actions need to be updated in every active release branch (and not just the `main` branch).

Also note that any downstream users relying on these actions will need to update their workflows as well; so, this needs to be fixed well before the deadline (31 May 2023).

stephanosio, 2023-05-11:
https://github.blog/changelog/2023-05-04-github-actions-all-actions-will-run-on-node16-instead-of-node12/

> Following on from our [warning in workflows using Node 12](https://github.blog/changelog/2022-09-22-github-actions-all-actions-will-begin-running-on-node16-instead-of-node12/), we will start enforcing the use of Node16 rather than Node12 on the 18th of May.

18th May is the deadline.

stephanosio, 2023-05-31:
Downgrading to medium priority since all the major and most critical workflows have already been migrated.

p.s. Even after the aforementioned deadline (May 18), the Node12 actions seem to be still working.

closed 2023-08-15 by github-actions[bot]