zephyrproject-rtos/zephyr #56625: mgmt/MCUmgr: Improve logging in image management group

de-nordic, 2023-04-06:
**Is your enhancement proposal related to a problem? Please describe.**
There are a lot of places where image group processing returns with error codes like `MGMT_ERR_UNKNOWN` with not additional information given.
Other error codes could also use some additional logging, like debug or error.

**Describe the solution you'd like**
More logging, specifically around `MGMG_ERR_UNKNOWN`

**Describe alternatives you've considered**
No alternatives.

**Additional context**
Image group already declares logging:
https://github.com/zephyrproject-rtos/zephyr/blob/4bb8b2523fa4cc6b51b7d568404b988e2391e0de/subsys/mgmt/mcumgr/grp/img_mgmt/src/zephyr_img_mgmt.c#L24
so it is only needed to be used in img_mgmt.c and img_mgmt_sate.c

nordicjm, 2023-04-11:
I'm actually against this, well, not against it - more logging is nice, but I would far rather have https://github.com/zephyrproject-rtos/zephyr/issues/53955 implemented with proper command result codes, that way clients will actually know what the real error is because yes the current error reporting is absolutely worthless and I think with proper error reporting, logs detailing the error are still important but not as important because then users would no longer need a CLI attached to see what is going wrong, it should be obvious from the code they get on their firmware update mechanism in their boxed product with has no display/CLI output.

nordicjm, 2023-04-11:
Though note for the above, it needs to be implemented soonish since we really need to mark MCUmgr as stable for 3.4 and it needs to be changed prior to that being announced.

de-nordic, 2023-04-11:
> I'm actually against this, well, not against it - more logging is nice, but I would far rather have #53955 implemented with proper command result codes, that way clients will actually know what the [...]

That is OK, but we will not be able to cover everything with error codes and, sometimes, we will be translating something that comes from upper level to ERR_MGMT codes. In such cases it might be worth to log why are we returning certain code.

PR #84872 (2025-02-19): mcumgr: Add error logs for all error cases
Errors were logged for some cases in zephyr_img_mgmt.c, but not all. This commit adds error logs for all error cases.

Fixes #56625