zephyrproject-rtos/zephyr #56627: Bluetooth: build fails with `CONFIG_BT_LOG=n`

mniestroj, 2023-04-06:
**Describe the bug**
Build of Bluetooth subsystem fails when `CONFIG_BT_LOG=n` is selected by user, in order to reduce footprint size. Everything is fine when this option is left enabled (default when `CONFIG_LOG=y && CONFIG_BT=y`).

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nrf52840dk_nrf52840 samples/bluetooth/peripheral -- -DCONFIG_BT_LOG=n
2. Build error `zephyr/subsys/bluetooth/host/l2cap.c:30:31: error: 'CONFIG_BT_L2CAP_LOG_LEVEL' undeclared here (not in a function)` occurs.

**Expected behavior**
Build should be successful after disabling logs for Bluetooth subsystem.

**Impact**
Annoyance, as Bluetooth logging can use lots of flash size. Bluetooth logs cannot be easily disabled without this option.

**Logs and console output**
Build error:
```
In file included from zephyr/include/zephyr/sys/util_macro.h:34,
                 from zephyr/include/zephyr/sys/atomic.h:16,
                 from zephyr/include/zephyr/kernel_includes.h:21,
                 from zephyr/include/zephyr/kernel.h:17,
                 from zephyr/subsys/bluetooth/host/l2cap.c:9:
zephyr/subsys/bluetooth/host/l2cap.c:30:31: error: 'CONFIG_BT_L2CAP_LOG_LEVEL' undeclared here (not in a function); did you mean 'CONFIG_USBC_LOG_LEVEL'?
   30 | LOG_MODULE_REGISTER(bt_l2cap, CONFIG_BT_L2CAP_LOG_LEVEL);
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~
zephyr/include/zephyr/sys/util_internal.h:72:26: note: in definition of macro '__DEBRACKET'
   72 | #define __DEBRACKET(...) __VA_ARGS__
      |                          ^~~~~~~~~~~
zephyr/include/zephyr/sys/util_internal.h:64:9: note: in expansion of macro '__GET_ARG2_DEBRACKET'
   64 |         __GET_ARG2_DEBRACKET(one_or_two_args _if_code, _else_code)
      |         ^~~~~~~~~~~~~~~~~~~~
zephyr/include/zephyr/sys/util_internal.h:59:9: note: in expansion of macro '__COND_CODE'
   59 |         __COND_CODE(_XXXX##_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~
zephyr/include/zephyr/sys/util_macro.h:180:9: note: in expansion of macro 'Z_COND_CODE_1'
  180 |         Z_COND_CODE_1(_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~~~
zephyr/include/zephyr/logging/log.h:364:9: note: in expansion of macro 'COND_CODE_1'
  364 |         COND_CODE_1(                                                    \
      |         ^~~~~~~~~~~
zephyr/include/zephyr/logging/log.h:320:9: note: in expansion of macro '_LOG_MODULE_CONST_DATA_CREATE'
  320 |         _LOG_MODULE_CONST_DATA_CREATE(_name, _level);           \
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
zephyr/include/zephyr/logging/log.h:366:18: note: in expansion of macro '_LOG_MODULE_DATA_CREATE'
  366 |                 (_LOG_MODULE_DATA_CREATE(GET_ARG_N(1, __VA_ARGS__),     \
      |                  ^~~~~~~~~~~~~~~~~~~~~~~
zephyr/include/zephyr/sys/util_internal.h:69:53: note: in expansion of macro '__DEBRACKET'
   69 | #define __GET_ARG2_DEBRACKET(ignore_this, val, ...) __DEBRACKET val
      |                                                     ^~~~~~~~~~~
zephyr/include/zephyr/sys/util_internal.h:64:9: note: in expansion of macro '__GET_ARG2_DEBRACKET'
   64 |         __GET_ARG2_DEBRACKET(one_or_two_args _if_code, _else_code)
      |         ^~~~~~~~~~~~~~~~~~~~
zephyr/include/zephyr/logging/log_core.h:85:9: note: in expansion of macro '__COND_CODE'
   85 |         __COND_CODE(_LOG_ZZZZ##_eval_level, _iftrue, _iffalse)
      |         ^~~~~~~~~~~
zephyr/include/zephyr/logging/log_core.h:82:9: note: in expansion of macro 'Z_LOG_EVAL1'
   82 |         Z_LOG_EVAL1(_eval_level, _iftrue, _iffalse)
      |         ^~~~~~~~~~~
zephyr/include/zephyr/logging/log.h:291:9: note: in expansion of macro 'Z_LOG_EVAL'
  291 |         Z_LOG_EVAL(LOG_LEVEL, \
      |         ^~~~~~~~~~
zephyr/include/zephyr/sys/util_macro.h:331:27: note: in expansion of macro 'Z_GET_ARG_2'
  331 | #define GET_ARG_N(N, ...) Z_GET_ARG_##N(__VA_ARGS__)
      |                           ^~~~~~~~~~
zephyr/include/zephyr/logging/log.h:293:20: note: in expansion of macro 'GET_ARG_N'
  293 |                   (GET_ARG_N(2, __VA_ARGS__, CONFIG_LOG_DEFAULT_LEVEL)))
      |                    ^~~~~~~~~
zephyr/include/zephyr/logging/log.h:367:39: note: in expansion of macro '_LOG_LEVEL_RESOLVE'
  367 |                                       _LOG_LEVEL_RESOLVE(__VA_ARGS__))),\
      |                                       ^~~~~~~~~~~~~~~~~~
zephyr/subsys/bluetooth/host/l2cap.c:30:1: note: in expansion of macro 'LOG_MODULE_REGISTER'
   30 | LOG_MODULE_REGISTER(bt_l2cap, CONFIG_BT_L2CAP_LOG_LEVEL);
      | ^~~~~~~~~~~~~~~~~~~
[202/282] Building C object zephyr/subsys/bluetooth/host/CMakeFiles/subsys__bluetooth__host.dir/hci_core.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /bin/cmake --build build
```

**Environment (please complete the following information):**
 - Toolchain: Zephyr SDK 0.16.0
 - Commit SHA: 4bb8b2523fa4cc6b51b7d568404b988e2391e0de

**Additional context**
This was introduced as part of commit af01a0f31398.

mniestroj, 2023-04-06:
FYI @theob-pro.

theob-pro, 2023-04-11:
My idea with the recent update was to use the logging symbols instead custom subsystem one. Can you try disabling BT logging using `BT_LOG_LEVEL_OFF=y` instead of `CONFIG_BT_LOG=n`?

mniestroj, 2023-04-11:
> My idea with the recent update was to use the logging symbols instead custom subsystem one. Can you try disabling BT logging using `BT_LOG_LEVEL_OFF=y` instead of `CONFIG_BT_LOG=n`?

There are no build issues with `BT_LOG_LEVEL_OFF=y`. It seems like a workaround for now. If that should always be the only way to disable BT logs, then `BT_LOG` should be made a ["hidden"](https://docs.zephyrproject.org/latest/build/kconfig/tips.html#assignments-to-promptless-symbols-in-configuration-files) Kconfig option.

theob-pro, 2023-04-11:
> There are no build issues with `BT_LOG_LEVEL_OFF=y`. It seems like a workaround for now. If that should always be the only way to disable BT logs, then `BT_LOG` should be made a [...]

Using `BT_LOG_LEVEL_OFF=y` is actually not a workaround but the right way to disable Bluetooth logging. But you're right, `BT_LOG` and `BT_LOG_LEGACY` should be hidden to the user. I created a PR to fix that.

PR #56728 (2023-04-12): Bluetooth: Logging: Make `BT_LOG` and `BT_LOG_LEGACY` hidden symbols
Make `BT_LOG` and `BT_LOG_LEGACY` hidden Kconfig symbols.

They should not be used by the user to configure the Bluetooth logging system. If the user want to completely disable Bluetooth logging, they should use `BT_LOG_LEVEL_OFF=y`.

Fixes: #56627
closed 2023-04-12 by carlescufi