zephyrproject-rtos/zephyr #56635: Introduce a Battery Charging API

rriveramcrus, 2023-04-06:
## Introduction

Add a kernel device driver API for battery charging control and status reporting.

### Problem description

Charging devices are application specific power converters with built in protection and sensing for safe battery charging and system power delivery. ~~There is currently no suitable subsystem, API, or example of a charging IC in Zephyr.~~ Some out-of-tree charging drivers leverage the sensor APIs channels and attributes. While this approach is suitable for more simple charging applications and platforms, more complicated power trees could have multiple charging ICs, a fuel gauge, power muxes, and input source detection components. These more complicated power trees require inter and intra subsystem communication via the host controller running Zephyr. 

### Proposed change

Adding a battery charging device driver API that facilitates communication to the necessary components of the power tree to complete a charge cycle. This proposed subsystem would be akin to the recently merged fuel gauge subsystem (#44847).

## Detailed RFC

### Proposed change (Detailed)

An external draft PR ([Creates Charging Subsystem](https://github.com/CirrusLogic/zephyr-drivers/pull/2)) is available that resembles the Zephyr fuel gauge API and the Linux mainline power supply subsystem ([Linux Kernel's power supply subsystem documentation](https://docs.kernel.org/power/power_supply_class.html)).

### Dependencies

No upstreamed clients would be affected by this change.

### Next Steps Towards a Future Power Supply Subsystem
The creation of this kernel space API would be the logical next step towards a fuller subsystem capable of performing a full battery charge cycle via an application.

### Concerns and Unresolved Questions

- Architectural consideration must be given to how the charging and fuel gauge APIs will interact.
- Some highly integrated ICs do offer the ability to gauge as well as charge, but most do not.

## Alternatives

### Status Quo

Pros
- No additional code base or complexity

Cons
- No true home for charging ICs
- Power Supply Subsystem remains incomplete

nashif, 2023-04-06:
@aaronemassey @teburd  FYI

nashif, 2023-04-06:
@rriveramcrus

> An external draft PR (https://github.com/CirrusLogic/zephyr-drivers/pull/2)

Maybe post this against the zephyr repo to get more visibility and reviews? Thank you for the posting this RFC and sharing the link to the PR.

teburd, 2023-04-06:
I think this could slot right in given the diagram @aaronemassey provided in the linked RFC (#44847) which is great.

It may require some tweaks and such but the premise looks like a good one to me.

Please send the PR to the main tree so we can look and review.

rriveramcrus, 2023-04-07:
Created a proper PR, I'm going to go ahead and close the original external PR.

closed 2024-04-05 by rriveramcrus