zephyrproject-rtos/zephyr #56643: support default toolchains based on arch in twister

galak, 2023-04-06:
## Introduction

Rather than having to always specify common toolchains for a given architecture, allow twister to be aware of common toolchains so they don't have to be setup in <BOARD>.yaml

### Problem description

Rather than having to update every <BOARD>.yaml when we add support for a new toolchain that is supported architecture wide, we can just add it one place.

### Proposed change

Have a default set of toolchains in supported_toolchains and remove the need to set these toolchains in <BOARD>.yaml.   Allow removal of the defaults from all <BOARD>.yaml.

galak, 2023-04-06:
See as an example https://github.com/zephyrproject-rtos/zephyr/pull/56642

nashif, 2023-08-17:
AFAIK, this was already implemented.

closed 2023-08-17 by nashif