zephyrproject-rtos/zephyr #56646: Multicast not working unless in promiscuous

cguimaraes, 2023-04-07:
**Describe the bug**
I have been using Zephyr v2.7.1 (mostly because it is the latest version support on PlatformIO), and multicast has been working just fine.
Lately, I have been moving to Zephyr v3.3.x (using `west`) and since then all the multicast code stopped working, unless I enabled NET_PROMISCUOUS_MODE in `prj.conf`

It looks to me that this is more of a workaround, and not so much how it is intended to work on Zephyr v3.3.x.
 
This has been tested and verified in: `nucleo_f767zi`,  `reel_board`, and  `s32z270dc2_rtu0_r52`. 

**To Reproduce**
I was able to strip down the code to a small testing application that fails:
```C
#include <arpa/inet.h>
#include <errno.h>
#include <netdb.h>
#include <netinet/in.h>
#include <stdbool.h>
#include <stddef.h>
#include <stdio.h>
#include <sys/socket.h>

int8_t _z_create_endpoint_udp(struct addrinfo *ep, const char *s_address, const char *s_port) {
    struct addrinfo hints;
    (void)memset(&hints, 0, sizeof(hints));
    hints.ai_family = PF_UNSPEC;  // Allow IPv4 or IPv6
    hints.ai_socktype = SOCK_DGRAM;
    hints.ai_flags = 0;
    hints.ai_protocol = IPPROTO_UDP;

    if (getaddrinfo(s_address, s_port, &hints, &ep) < 0) {
        return -1;
    }

    return 0;
}

void _free_endpoint_udp(struct addrinfo *ep) { freeaddrinfo(ep); }

int8_t _z_listen_udp_multicast(int *sock, struct sockaddr_in rep) {
    struct sockaddr *lsockaddr = NULL;
    unsigned int addrlen = 0;
    lsockaddr = (struct sockaddr *)malloc(sizeof(struct sockaddr_in));
    (void)memset(lsockaddr, 0, sizeof(struct sockaddr_in));
    addrlen = sizeof(struct sockaddr_in);

    struct sockaddr_in *c_laddr = (struct sockaddr_in *)lsockaddr;
    c_laddr->sin_family = AF_INET;
    c_laddr->sin_addr.s_addr = htons(INADDR_ANY);
    c_laddr->sin_port = rep.sin_port;

    *sock = socket(rep.sin_family, SOCK_DGRAM, IPPROTO_UDP);
    if (*sock != -1) {
        int optflag = 1;
        if (setsockopt(*sock, SOL_SOCKET, SO_REUSEADDR, (char *)&optflag, sizeof(optflag)) < 0) {
            return -1;
        }

        if (bind(*sock, lsockaddr, addrlen) < 0) {
            return -1;
        }

        // FIXME: iface passed into the locator is being ignored
        //        default if used instead
        struct net_if *ifa = NULL;
        ifa = net_if_get_default();
        if (ifa != NULL) {
            if (rep.sin_family == AF_INET) {
                struct net_if_mcast_addr *mcast = NULL;
                mcast = net_if_ipv4_maddr_add(ifa, &rep.sin_addr);
                if (!mcast) {
                    return -1;
                }
                net_ipv4_igmp_join(mcast);
            }
        } else {
            free(lsockaddr);
            return -1;
        }

        free(lsockaddr);
    } else {
        return -1;
    }

    return 0;
}

size_t _z_read_udp_multicast(int sock, uint8_t *ptr, size_t len) {
    struct sockaddr_storage raddr;
    unsigned int raddrlen = sizeof(struct sockaddr_storage);

    ssize_t rb = 0;
    do {
        rb = recvfrom(sock, ptr, len, 0, (struct sockaddr *)&raddr, &raddrlen);
        if (rb < (ssize_t)0) {
            rb = SIZE_MAX;
            break;
        }
    } while (1);

    return rb;
}

size_t _z_send_udp_multicast(int sock, const uint8_t *ptr, size_t len, struct addrinfo rep) {
    return sendto(sock, ptr, len, 0, rep.ai_addr, rep.ai_addrlen);
}

int main() {
    k_sleep(K_SECONDS(5));  // Sleep to make sure the network is up and running

    int sock = -1;

    struct sockaddr_in addr1;
    memset(&addr1, 0, sizeof(addr1));
    addr1.sin_family = AF_INET;
    addr1.sin_port = htons(7660);
    inet_pton(AF_INET, "239.255.0.1", &addr1.sin_addr);
    if (_z_listen_udp_multicast(&sock, addr1) < 0) {
        printf("Error opening the socket\n");
    }

    uint8_t buf[1500];
    int rb = _z_read_udp_multicast(sock, buf, 1500);
    printf("Received the following data [%d] %s", rb, buf);

    return 0;
}
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
It forced me to update all my build to contain the promiscuous.
Also, it might have unintended impact on the performance by enabling any packet to have to be processed.

**Logs and console output**
I can provide Wireshark captures of the node sending the multicast traffic if required.
It is being sent on the proper interface and, in fact, one could see the Ethernet interface blinking every time a packet is sent (or a burst of them).

Also, Zephyr net statistics do not catch anything w.r.t. received multicast packets, unless in promiscuous mode.

**Environment (please complete the following information):**
 - OS: Linux or MacOS
 - Toolchain: zephyr-sdk-0.16.0, zephyr 3.3.99

**Additional context**
It happens for both IPv4 and IPv6.

carlescufi, 2023-04-12:
@cguimaraes thanks for the report. Could you also test on the latest `main`?

cguimaraes, 2023-04-14:
@carlescufi sure!
I will make the tests next week, when I get the HW back.

rlubos, 2023-04-20:
@cguimaraes Which ethernet driver are you using?

I've tried to reproduce the problem (with the sample code you provide) on `nucleo_h723zg` board (`ETH_STM32_HAL` driver), but mutlicast reception works just fine, both with v3.3.0 or current main. No need to enable promiscuous mode.

Note, that your sample code won't print anything on reception, as the `printf()` is outside the receive loop, and the exit condition for the loop is a socket error (which is unlikely in normal circumstances).

cfriedt, 2023-04-20:
@cguimaraes - I would even go so far as to say, it's always best to add something to the testsuite that ensures the test can be reliably demonstrated on more than 1 platform, given that the network stack is expected to work the same on virtually every board. 

Actually, just reiterated the same thing in another ticket, but my philosophy is that the testsuite is often more valuable than the implementation it is actually testing.

We would be very grateful if you could contribute the issue you've detected to the testsuite in a PR, if possible. If it demonstrates the failure occuring in CI (or if multiple devs are able to reproduce it in HW), then by all means, it's a verified bug that should be fixed - but at least we have evidence that we will be preventing future regressions.

cguimaraes, 2023-04-20:
@rlubos indeed there is an issue with the snippet of code I pasted here. It was an artifact that I forgot to remove, since I wanted to provide a simple failing example (instead of the full library integrated). Today I got access at least to the `nucleo_f767zi` and I am trying the latest `main` branch.

@cfriedt I understand your point and I completely agree with it. However, note that before opening the issue I tried to debug it myself and got to a point I was not able to proceed further. Otherwise, I would have patched the code myself and made a Pull Request. I am more of a user of Zephyr, thus I am not aware of all the implementation details and/or testsuites. In fact, I followed the guidelines while opening an issue as indicated.

In any case, if you can point me out to a wiki page that explains how to add new tests to the testsuite I would be happy to contribute with a PR.

cfriedt, 2023-04-20:
@kartben - this may be a good opportunity to ad a bit of documentation that could pay off with a multiplicative effect. 

See my comment above and also the comment here
https://github.com/zephyrproject-rtos/zephyr/pull/57080/#pullrequestreview-1393915703

cguimaraes, 2023-04-20:
Thank you for the guidelines @cfriedt .
I will do as you say and include it as part of the testsuite :-)

In the meantime, I also updated the original post to fix the `recvfrom` issue.

cguimaraes, 2023-04-20:
> @cguimaraes thanks for the report. Could you also test on the latest `main`?

@carlescufi with the latest main it seems to be working again on `nucleo_f767zi`.

closed 2023-04-21 by carlescufi