zephyrproject-rtos/zephyr #56649: ESP32 wifi sample with MQTT over TLS crashes

marinjurjevic, 2023-04-07:
**Describe the bug**
I've created a custom sample which is combination of `samples/net/wifi` and `samples/net/mqtt_publisher`. I'm developing on ESP32-WROOM-32E (ESP32 devkitc v4). I've created a helper shell command for using MQTT commands (connect/disconnect/publish/subscribe..). MQTT connection is using TLS.

Randomly after sending shell commands I get fatal error. 

Sometimes I also get error below, but not always so I don't think it's relevant but listing it here as additional info.
```
[00:04:42.993,000] <err> net_pkt: Data buffer (1203) allocation failed.
[00:04:42.993,000] <err> net_tcp: conn: 0x3fffcb24 packet allocation failed, len=1155
[00:04:42.993,000] <err> net_tcp: TCP failed to allocate buffer in retransmission
```

I tried increasing various thread stack sizes (shell, workqueue) but to no avail.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b esp32
2. west flash --esp-device /dev/ttyUSB0 && west espressif monitor -p /dev/ttyUSB0
3. Connect to AP and MQTT broker, publish subscribe, disconnect, wifi scan
4. Error happens randomly

**Expected behavior**
I don't expect to see any errors.

**Impact**
Showstopper

**Logs and console output**

<details>
<summary>shell commands and logs</summary>
<p>

```
uart:~$ wifi connect "*****" *****
Connection requested
[00:04:24.983,000] <dbg> net_wifi_mgmt: wifi_connect: (shell_uart): ch 255 sec 1
[00:04:26.971,000] <dbg> net_dhcpv4: dhcpv4_start_internal: (esp_event): iface 0x3ffb1c68 state=init
[00:04:26.971,000] <dbg> net_dhcpv4: dhcpv4_start_internal: (esp_event): wait timeout=5s
[00:04:26.971,000] <dbg> net_dhcpv4: dhcpv4_set_timeout: (esp_event): sched timeout dhcvp4=0x3ffb1d08 timeout=5s
[00:04:26.971,000] <dbg> net_dhcpv4: dhcpv4_manage_timers: (sysworkq): iface 0x3ffb1c68 dhcpv4=0x3ffb1d08 state=init timeleft=5
[00:04:26.971,000] <dbg> net_dhcpv4: dhcpv4_timeout: (sysworkq): Waiting for 5s
[00:04:31.972,000] <dbg> net_dhcpv4: dhcpv4_manage_timers: (sysworkq): iface 0x3ffb1c68 dhcpv4=0x3ffb1d08 state=init timeleft=0
[00:04:31.972,000] <dbg> net_dhcpv4: dhcpv4_enter_selecting: (sysworkq): enter state=selecting
[00:04:31.973,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3fffb424 len 291
[00:04:31.973,000] <dbg> net_dhcpv4: dhcpv4_set_timeout: (sysworkq): sched timeout dhcvp4=0x3ffb1d08 timeout=4s
[00:04:31.973,000] <dbg> net_dhcpv4: dhcpv4_send_discover: (sysworkq): send discover xid=0xdfe0571a timeout=4s
[00:04:31.973,000] <dbg> net_dhcpv4: dhcpv4_timeout: (sysworkq): Waiting for 4s
[00:04:31.973,000] <dbg> net_dhcpv4: dhcpv4_manage_timers: (sysworkq): iface 0x3ffb1c68 dhcpv4=0x3ffb1d08 state=selecting timeleft=4
[00:04:31.973,000] <dbg> net_dhcpv4: dhcpv4_timeout: (sysworkq): Waiting for 4s
[00:04:32.153,000] <dbg> net_dhcpv4: net_dhcpv4_input: (rx_q[0]): Received dhcp msg [op=0x2 htype=0x1 hlen=6 xid=0xdfe0571a secs=0 flags=0x80 chaddr=94:3C:C6:CD:D7:60
[00:04:32.153,000] <dbg> net_dhcpv4: net_dhcpv4_input: (rx_q[0]):   ciaddr=0.0.0.0
[00:04:32.153,000] <dbg> net_dhcpv4: net_dhcpv4_input: (rx_q[0]):   yiaddr=192.168.43.97
[00:04:32.153,000] <dbg> net_dhcpv4: net_dhcpv4_input: (rx_q[0]):   siaddr=192.168.43.1
[00:04:32.153,000] <dbg> net_dhcpv4: net_dhcpv4_input: (rx_q[0]):   giaddr=0.0.0.0]
[00:04:32.153,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): options_server_id: 192.168.43.1
[00:04:32.153,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): options_lease_time: 3600
[00:04:32.153,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): options_renewal: 1800
[00:04:32.153,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): options_rebinding: 3150
[00:04:32.154,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): options_subnet_mask 255.255.255.0
[00:04:32.154,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): option unknown: 28
[00:04:32.154,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): options_router: 192.168.43.1
[00:04:32.154,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): option unknown: 43
[00:04:32.154,000] <dbg> net_dhcpv4: dhcpv4_parse_options: (rx_q[0]): options_end
[00:04:32.154,000] <dbg> net_dhcpv4: dhcpv4_handle_reply: (rx_q[0]): state=selecting msg=offer
[00:04:32.154,000] <dbg> net_dhcpv4: dhcpv4_enter_requesting: (rx_q[0]): enter state=requesting
[00:04:32.154,000] <dbg> net_dhcpv4: dhcpv4_set_timeout: (rx_q[0]): sched timeout dhcvp4=0x3ffb1d08 timeout=4s
[00:04:32.155,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3fffb424 len 298
[00:04:32.155,000] <dbg> net_dhcpv4: dhcpv4_send_request: (rx_q[0]): send request dst=255.255.255.255 xid=0xdfe0571b ciaddr=<unknown> +server-id +requested-ip timeout=4s
[00:04:32.155,000] <dbg> net_dhcpv4: dhcpv4_manage_timers: (sysworkq): iface 0x3ffb1c68 dhcpv4=0x3ffb1d08 state=requesting timeleft=4
[... 209 lines cut ...]

CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=4096
CONFIG_SHELL_STACK_SIZE=4096

# MQTT TLS
CONFIG_MAIN_STACK_SIZE=4096
CONFIG_MQTT_LIB_TLS=y
CONFIG_NET_SOCKETS_SOCKOPT_TLS=y

# TLS configuration
CONFIG_MBEDTLS=y
CONFIG_MBEDTLS_BUILTIN=y
CONFIG_MBEDTLS_ENABLE_HEAP=y
CONFIG_MBEDTLS_HEAP_SIZE=70000
CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=5120

CONFIG_MQTT_SERVICE_SERVER_PORT=8883
CONFIG_MQTT_SERVICE_LOG_LEVEL_DBG=y

CONFIG_NET_MGMT_EVENT_STACK_SIZE=2048
CONFIG_NET_MGMT_EVENT_QUEUE_SIZE=20
```

</p>
</details>

marinjurjevic, 2023-04-07:
Here's another error, I left the device running and it occured after 43minutes:
```
[00:43:17.081,000] <err> os:  ** FATAL EXCEPTION
[00:43:17.081,000] <err> os:  ** CPU 0 EXCCAUSE 0 (illegal instruction)
[00:43:17.081,000] <err> os:  **  PC 0x40083a1c VADDR (nil)
[00:43:17.081,000] <err> os:  **  PS 0x60920
[00:43:17.081,000] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:9 CALLINC:2)
[00:43:17.081,000] <wrn> cbprintf_package: (unsigned) char * used for %p argument. It's recommended to cast it to void * because it may cause misbehavior in certain configurations. String:" **  A0 %p  SP %p  A2 %p  A3 %p" argument:1
[00:43:17.081,000] <err> os:  **  A0 0x781  SP 0x3ffe9590  A2 0x3ffe81cc  A3 0x3ffb9650
[00:43:17.081,000] <err> os:  **  A4 (nil)  A5 0x3fffba20  A6 0x608  A7 (nil)
[00:43:17.081,000] <err> os:  **  A8 0x8008558b  A9 0x3ffe9560 A10 (nil) A11 0x3ffb1bb8
[00:43:17.081,000] <err> os:  ** A12 (nil) A13 0xffff A14 0xfffffffe A15 0x3fffb404
[00:43:17.081,000] <err> os:  ** LBEG 0x400f40c8 LEND 0x400f40d9 LCOUNT (nil)
[00:43:17.081,000] <err> os:  ** SAR 0x4

Ba0x40083a19:0x30x0000077e:0x3 |<-C

[00:43:17.081,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:43:17.081,000] <err> os: Current thread: 0x3ffb95c0 (rx_q[0])
[00:43:17.256,000] <err> os: Halting system
```

tomsalpietro, 2023-04-10:
I am getting similar errors using an ESP32 DevkitC for Wifi. I am actually trying to do something very similar where I connect to wifi and upload to a dashboard via mqtt. Here is my fatal error code: 
```

[00:01:33.298,000] <err> os:  ** FATAL EXCEPTION
[00:01:33.298,000] <err> os:  ** CPU 0 EXCCAUSE 28 (load prohibited)
[00:01:33.298,000] <err> os:  **  PC 0x40085638 VADDR 0x8c
0x40085638: z_swap at /home/tomsalpietro/embedded/zephyrproject/zephyr/kernel/include/kswap.h:170

[00:01:33.298,000] <err> os:  **  PS 0x60420
[00:01:33.298,000] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:4 CALLINC:2)
[00:01:33.298,000] <wrn> cbprintf_package: (unsigned) char * used for %p argument. It's recommended to cast it to void * because it may cause misbehavior in certain configurations. String:" **  A0 %p  SP %p  A2 %p  A3 %p" argument:1
[00:01:33.298,000] <err> os:  **  A0 0x80085924  SP 0x3ffe7680  A2 (nil)  A3 0x60420
[00:01:33.298,000] <err> os:  **  A4 0x3ffb1a1c  A5 0x3ffb7b44  A6 0x60820  A7 (nil)
[00:01:33.298,000] <err> os:  **  A8 0x80085630  A9 0x3ffe7670 A10 0x3ffe69cc A11 0x3ffcc888
[00:01:33.298,000] <err> os:  ** A12 (nil) A13 0x3fffa0c4 A14 0xfffff9f8 A15 0x1
[00:01:33.298,000] <err> os:  ** LBEG 0x400eb0ac LEND 0x400eb0f0 LCOUNT 0x3

[00:01:33.298,000] <err> os:  ** SAR 0x3

[00:01:33.298,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:01:33.298,000] <err> os: Current thread: 0x3ffcc7f8 (rx_q[0])
[00:01:33.480,000] <err> os: Halting system
```
It looks very similar with the cbprintf warning and I also experience this randomly; maybe after a minute of running.
Would love to find a solution for these

jgl-meta, 2023-04-11:
It's probably worth checking the number of buffers allocated. This seems like a config error.

jgl-meta, 2023-04-11:
@sylvioalves

sylvioalves, 2023-04-11:
@marinjurjevic Do you have a working branch/repo so I can test it?

marinjurjevic, 2023-04-12:
Hi, you can find the branch [here](https://github.com/marinjurjevic/zephyr/tree/samples/net/wifi/esp32_crash).

You can use following commands:
```bash

uart:~$ wifi connect "<ssid>" <password>
uart:~$ mqtt connect
uart:~$ mqtt publish <topic_name>
uart:~$ mqtt subscribie <topic_name>
```

tomsalpietro, 2023-04-20:
I might be able to provide some more information on my problem. It looks like some threads are getting corrupted? not too sure but this seems relevant to the thread.
```
[00:09:18.717,000] <err> os:  **  A0 0xac0  SP 0x3ffe8090  A2 0x3ffe573c  A3 0x3ffb49e8
[00:09:18.717,000] <err> os:  **  A4 (nil)  A5 (nil)  A6 0x3ffb62d4  A7 (nil)
[00:09:18.717,000] <err> os:  **  A8 0x800852bf  A9 0x3ffe8060 A10 (nil) A11 0x3fff8b64
[00:09:18.717,000] <err> os:  ** A12 (nil) A13 0x3ffb5e1c A14 0x1e0 A15 0x3ffe7fd0
[00:09:18.717,000] <err> os:  ** LBEG 0x400e6b1c LEND 0x400e6b60 LCOUNT 0x7
0x400e6b1c: net_context_is_used at /home/tomsalpietro/embedded/zephyrproject/zephyr/include/zephyr/net/net_context.h:353
 (inlined by) find_context at /home/tomsalpietro/embedded/zephyrproject/zephyr/subsys/net/ip/net_context.c:830
 (inlined by) net_context_packet_received at /home/tomsalpietro/embedded/zephyrproject/zephyr/subsys/net/ip/net_context.c:1901

0x400e6b60: find_context at /home/tomsalpietro/embedded/zephyrproject/zephyr/subsys/net/ip/net_context.c:829
 (inlined by) net_context_packet_received at /home/tomsalpietro/embedded/zephyrproject/zephyr/subsys/net/ip/net_context.c:1901

[00:09:18.717,000] <err> os:  ** SAR 0x1a

Backtrace:0x40083a01:0x3ffe8090 0x00000abd:0x3ffe80a0  |<-CORRUPTED
0x40083a01: xtensa_switch at /home/tomsalpietro/embedded/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2-util.S:271

[00:09:18.717,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:09:18.717,000] <err> os: Current thread: 0x3ffb4998 (unknown)
[00:09:18.842,000] <err> os: Halting system
```
Probably something to do with context switching in ESP32 that zephyr isn't fully supporting yet?

sylvioalves, 2023-04-21:
@tomsalpietro @marinjurjevic thanks for the inputs, I'll take a look on it.

xyzzy42, 2023-04-28:
I found a trivial wifi example (just shell) on ESP32 will overflow the stacks for the network management system.

The `net_mgmt` thread's default stack of 768 isn't enough, once wifi is up and dhcp completes.  And if the shell's network monitor is on, then the `event_mon` thread's stack, which is hard-coded to 1024 bytes, will also overflow.

sylvioalves, 2023-04-28:
@xyzzy42 I planned to submit a wifi sample code and add custom overlays/changes to update those stacks you mentioned. @marinjurjevic, I have a working branch to fix your MQTT+TLS issue. Would you mind trying that out? Run `west update` once pulling this. https://github.com/zephyrproject-rtos/zephyr/pull/57391

xyzzy42, 2023-04-28:
I wonder how the stack usage on those threads compares to other systems with wifi?  I.e., are those stacks unrealistically small or does ESP32 have high stack usage?

sylvioalves, 2023-04-29:
@xyzzy42 Can you share the logging regarding those overflows you mentioned? Or same sample code?

xyzzy42, 2023-04-29:
There is no need for any code.  An empty `main()` is sufficient.  After connecting wifi the net_mgmt thread will have used its entire stack.  If `NET_MGMT_EVENT_MONITOR` and `ESP32_WIFI_STA_AUTO_DHCPV4` are on, then the `event_mon` thread will use up its stack too.

```
0x3ffcbc48 event_mon                        (real size 2048):   unused  956     usage 1092 / 2048 (53 %)
0x3ffcb998 net_mgmt                         (real size 1536):   unused  428     usage 1108 / 1536 (72 %)
```

The size of these stacks is normally 1024 and 768 bytes, respectively.

marinjurjevic, 2023-05-02:
> @xyzzy42 I planned to submit a wifi sample code and add custom overlays/changes to update those stacks you mentioned. @marinjurjevic, I have a working branch to fix your MQTT+TLS issue. Would you mind trying that out? Run `west update` once pulling this. #57391

Hi, I've been ill last week so I've only got around this today. I pulled your changes and so far I didn't experience hard fault.
Although I do get frequent disconnects. Not sure if this is connected to `tcp_work` stack being at 100%? That doesn't look normal.

```
uart:~$ mqtt connect
[00:13:26.209,000] <inf> mqtt_service: Resolving: test.mosquitto.org
[00:13:26.213,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3fffb820 len 78
[00:13:26.266,000] <inf> mqtt_service: IPv4 Address found 91.121.93.94
[00:13:26.270,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3fffb7a0 len 58
[00:13:26.308,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3fffb7e0 len 54
[00:13:26.316,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3fffb760 len 173
[00:13:26.354,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3fffb7e0 len 54
[00:13:26.362,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3fffb7e0 len 54
[00:13:26.406,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3fffb7e0 len 54
[00:13:26.415,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3fffb760 len 1018
[00:13:26.584,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3fffb7a0 len 321
[00:13:27.781,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3fffb760 len 323
[00:13:27.846,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3fffb7a0 len 145
[00:13:27.889,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3fffb7e0 len 54
[00:13:27.903,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3fffb760 len 155
[00:13:27.903,000] <inf> net_mqtt: Connect completed
[00:13:27.904,000] <dbg> mqtt_service: mqtt_service_poll: MQTT broker connection request sent
[00:13:27.942,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3fffb7e0 len 54
[00:13:27.947,000] <dbg> mqtt_service: mqtt_service_evt_handler: MQTT client connected!
[00:13:27.947,000] <inf> wifi_sample: MQTT Client connected!
[00:13:29.134,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3fffb7e0 len 54
[00:13:29.141,000] <dbg> mqtt_service: mqtt_service_evt_handler: MQTT publish received 0, 13 bytes
[00:13:29.141,000] <dbg> mqtt_service: mqtt_service_evt_handler:  id: 0, qos: 0
[00:13:29.141,000] <dbg> mqtt_service: mqtt_service_evt_handler: MQTT message payload:
                                       48 65 6c 6c 6f 20 66 72  6f 6d 20 50 43          |Hello fr om PC   
[00:13:29.141,000] <inf> wifi_sample: Received 13 bytes from topic: my_topic/
[00:13:29.141,000] <inf> wifi_sample: Message
                                      48 65 6c 6c 6f 20 66 72  6f 6d 20 50 43          |Hello fr om PC   
uart:~$ mqtt unsubscribe my_topic/
Unsubscribing from my_topic/ topic.
[00:13:37.491,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3fffb760 len 123
[00:13:37.530,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3fffb7e0 len 54
[00:13:37.535,000] <dbg> mqtt_service: mqtt_service_evt_handler: UNSUBACK packet
uart:~$ mqtt subscribe my_topic/
Subscribing to my_topic/ topic with QoS: 0
[00:13:49.662,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3fffb760 len 139
[00:13:49.701,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3fffb7e0 len 54
[00:13:49.707,000] <dbg> mqtt_service: mqtt_service_evt_handler: SUBACK packet
[00:13:51.867,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3fffb7e0 len 54
[00:13:51.874,000] <dbg> mqtt_service: mqtt_service_evt_handler: MQTT publish received 0, 13 bytes
[00:13:51.874,000] <dbg> mqtt_service: mqtt_service_evt_handler:  id: 0, qos: 0
[00:13:51.874,000] <dbg> mqtt_service: mqtt_service_evt_handler: MQTT message payload:
                                       48 65 6c 6c 6f 20 66 72  6f 6d 20 50 43          |Hello fr om PC   
[00:13:51.874,000] <inf> wifi_sample: Received 13 bytes from topic: my_topic/
[00:13:51.874,000] <inf> wifi_sample: Message
                                      48 65 6c 6c 6f 20 66 72  6f 6d 20 50 43          |Hello fr om PC   
uart:~$ kernel stacks
0x3ffb9120 mqtt_poll_thread                 (real size 4096):	unused 1480	usage 2616 / 4096 (63 %)
0x3ffb9a68 rx_q[0]                          (real size 2048):	unused  472	usage 1576 / 2048 (76 %)
0x3ffb98b0 wifi                             (real size 3584):	unused 1688	usage 1896 / 3584 (52 %)
0x3ffb99b0 net_mgmt                         (real size 2048):	unused  904	usage 1144 / 2048 (55 %)
0x3ffb9b40 tcp_work                         (real size 1024):	unused    0	usage 1024 / 1024 (100 %)
0x3ffb9cf0 esp_event                        (real size 4096):	unused 3304	usage  792 / 4096 (19 %)
0x3ffb9800 esp_timer                        (real size 4096):	unused 3544	usage  552 / 4096 (13 %)
0x3ffb9ef0 sysworkq                         (real size 4096):	unused 3184	usage  912 / 4096 (22 %)
0x3ffb9728 shell_uart                       (real size 4096):	unused 1600	usage 2496 / 4096 (60 %)
0x3ffb91c0 logging                          (real size 1024):	unused  696	usage  328 / 1024 (32 %)
0x3ffb9d90 idle                             (real size 1024):	unused  824	usage  200 / 1024 (19 %)
0x3ffeaa30 IRQ 00                           (real size 2048):	unused 1760	usage  288 / 2048 (14 %)
[00:14:49.668,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3fffb760 len 123
[... cut ...]
[00:15:49.767,000] <inf> net_mqtt_sock_tls: Closing socket 0
[00:15:49.770,000] <dbg> mqtt_service: mqtt_service_evt_handler: MQTT client disconnected -5
[00:15:49.770,000] <wrn> wifi_sample: MQTT Client disconnected! Reason: 3
[00:15:49.770,000] <err> mqtt_service: Cloud MQTT input error: -5
[00:15:49.770,000] <dbg> mqtt_service: mqtt_service_poll: The cloud socket is already closed.
```

In the meantime we moved to ESP32-C3 and we don't experience any issues. I'll leave this issue open for now.

sylvioalves, 2023-05-02:
@marinjurjevic, great, thanks. Yes, you have to increase that `tcp_work` stack (`CONFIG_NET_TCP_WORKQ_STACK_SIZE=4096`).  
I was not able to get anymore crashes. Will submit this.

sylvioalves, 2023-05-06:
Fixed at #57289. Open again if necessary.

closed 2023-05-06 by sylvioalves