zephyrproject-rtos/zephyr #56657: Incorrect IPv6 MSS size calculation leads to send failures

wirehell, 2023-04-07:
**Describe the bug**

In my setup, I'm using TCP connections with IPv6 through Openthread. MTU is by default 1280

What happens is that this line returns a correct MSS of 1220
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/ip/tcp.c#L3368
But this line, sets it (incorrectly) to 1280.
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/ip/tcp.c#L3371

This later causes this line to fail with OT_ERROR_PARSE (6), caused by Header parsing validation failing on too large message.
https://github.com/zephyrproject-rtos/zephyr/blob/ab410807bcc73fe9e88fb6c2dba393e2750aa4c1/modules/openthread/platform/radio.c#L445
It turns out that the message is 1340 bytes and does not fit into the MTU.

**To Reproduce**
Send an IPv6 TCP packet with size between 1221 and 1280.

**Expected behavior**
I expect the maximum MSS size for MTU 1280 to be 1220

**Impact**
I think this is a serious bug which probably impact more people than me. I can work around this by either patching my version (fix is trivial), or sending data with TCP_NODELAY and limited packet size.

**Environment (please complete the following information):**
Using main branch as of 2022-04-07

wirehell, 2023-04-07:
```diff --git a/subsys/net/ip/tcp.c b/subsys/net/ip/tcp.c
index eae947e1eb..87fdaa1fa8 100644
--- a/subsys/net/ip/tcp.c
+++ b/subsys/net/ip/tcp.c
@@ -3280,13 +3280,20 @@ uint16_t net_tcp_get_supported_mss(const struct tcp *conn)
        if (family == AF_INET) {
 #if defined(CONFIG_NET_IPV4)
                struct net_if *iface = net_context_get_iface(conn->context);
+               int mss = 0;
 
                if (iface && net_if_get_mtu(iface) >= NET_IPV4TCPH_LEN) {
                        /* Detect MSS based on interface MTU minus "TCP,IP
                         * header size"
                         */
-                       return net_if_get_mtu(iface) - NET_IPV4TCPH_LEN;
+                       mss = net_if_get_mtu(iface) - NET_IPV4TCPH_LEN;
+               }
+
+               if (mss == 0) {
+                       mss = NET_IPV4_MTU - NET_IPV4TCPH_LEN;
                }
+
+               return mss;
 #else
                return 0;
 #endif /* CONFIG_NET_IPV4 */
@@ -3303,8 +3310,8 @@ uint16_t net_tcp_get_supported_mss(const struct tcp *conn)
                        mss = net_if_get_mtu(iface) - NET_IPV6TCPH_LEN;
                }
 
-               if (mss < NET_IPV6_MTU) {
-                       mss = NET_IPV6_MTU;
+               if (mss == 0) {
+                       mss = NET_IPV6_MTU - NET_IPV6TCPH_LEN;
                }
 
                return mss;
```

carlescufi, 2023-04-12:
> This is the patch I apply locally against v3.3.0.
> 
> ```diff
> index eae947e1eb..87fdaa1fa8 100644
> --- a/subsys/net/ip/tcp.c
> +++ b/subsys/net/ip/tcp.c
> @@ -3280,13 +3280,20 @@ uint16_t net_tcp_get_supported_mss(const struct tcp *conn)
>         if (family == AF_INET) {
>  #if defined(CONFIG_NET_IPV4)
>                 struct net_if *iface = net_context_get_iface(conn->context);
> +               int mss = 0;
>  
>                 if (iface && net_if_get_mtu(iface) >= NET_IPV4TCPH_LEN) {
>                         /* Detect MSS based on interface MTU minus "TCP,IP
>                          * header size"
>                          */
> -                       return net_if_get_mtu(iface) - NET_IPV4TCPH_LEN;
> +                       mss = net_if_get_mtu(iface) - NET_IPV4TCPH_LEN;
> +               }
> +
> +               if (mss == 0) {
> +                       mss = NET_IPV4_MTU - NET_IPV4TCPH_LEN;
>                 }
> +
> +               return mss;
>  #else
>                 return 0;
>  #endif /* CONFIG_NET_IPV4 */
> @@ -3303,8 +3310,8 @@ uint16_t net_tcp_get_supported_mss(const struct tcp *conn)
>                         mss = net_if_get_mtu(iface) - NET_IPV6TCPH_LEN;
>                 }
>  
> -               if (mss < NET_IPV6_MTU) {
> -                       mss = NET_IPV6_MTU;
> +               if (mss == 0) {
> +                       mss = NET_IPV6_MTU - NET_IPV6TCPH_LEN;
>                 }
>  
>                 return mss;
> ```

Thanks for the patch! Why not open a Pull Request with it? Then we can review and merge to fix this.

commit 353e5f25 (2023-04-17): net: tcp: Fix IPv6 TCP MSS issue

Fixes #56657, incorrect TCP MSS calculation for IPv6.

Previously the supported MSS would incorrectly returned as the MTU if it
was less than the MTU, probably to account for the case where
the network interface MTU check returns 0.

New behaviour is to return the supported MSS as MTU minus header length,
using default MTU for this calculation if network interface MTU check
returns 0.
closed 2023-04-17 by carlescufi