zephyrproject-rtos/zephyr #56716: STM32F303C8: wrong measurements on some ADC channels

or-rhssk, 2023-04-11:
When building `samples/drivers/adc` with mutiple ADC channels enabled, some of them always measure close to VREF, instead of the actual voltage on the pin.

This was observed on a custom board with a STM32F303C8T6 SoC.

Here's the output of the flashed ADC sample project (with enabled UART console). During this run all the relevant analog inputs were briefly connected to ground. Channels 1, 2 and 4 appear to register the change while channels 11, 12 and 13 do not.
```
*** Booting Zephyr OS build zephyr-v3.3.0-2454-g65101ec68397 ***
ADC reading[0]:
- adc@50000000, channel 1: 4089 = 2994 mV
- adc@50000000, channel 2: 4080 = 2988 mV
- adc@50000000, channel 4: 4041 = 2959 mV
- adc@50000000, channel 11: 4093 = 2997 mV
- adc@50000000, channel 12: 4095 = 2999 mV
- adc@50000000, channel 13: 4095 = 2999 mV
ADC reading[1]:
- adc@50000000, channel 1: 4078 = 2986 mV
- adc@50000000, channel 2: 4071 = 2981 mV
- adc@50000000, channel 4: 4034 = 2954 mV
- adc@50000000, channel 11: 4092 = 2997 mV
- adc@50000000, channel 12: 4094 = 2998 mV
- adc@50000000, channel 13: 4093 = 2997 mV
ADC reading[2]:
- adc@50000000, channel 1: 4095 = 2999 mV
- adc@50000000, channel 2: 4078 = 2986 mV
- adc@50000000, channel 4: 4059 = 2972 mV
- adc@50000000, channel 11: 4092 = 2997 mV
- adc@50000000, channel 12: 4095 = 2999 mV
- adc@50000000, channel 13: 4095 = 2999 mV
ADC reading[3]:
- adc@50000000, channel 1: 4095 = 2999 mV
- adc@50000000, channel 2: 4077 = 2986 mV
- adc@50000000, channel 4: 4055 = 2969 mV
- adc@50000000, channel 11: 4095 = 2999 mV
- adc@50000000, channel 12: 4091 = 2996 mV
- adc@50000000, channel 13: 4092 = 2997 mV
ADC reading[4]:
- adc@50000000, channel 1: 4095 = 2999 mV
- adc@50000000, channel 2: 4092 = 2997 mV
- adc@50000000, channel 4: 4067 = 2978 mV
- adc@50000000, channel 11: 4093 = 2997 mV
- adc@50000000, channel 12: 4094 = 2998 mV
- adc@50000000, channel 13: 4092 = 2997 mV
ADC reading[5]:
- adc@50000000, channel 1: 4095 = 2999 mV
- adc@50000000, channel 2: 4072 = 2982 mV
- adc@50000000, channel 4: 4063 = 2975 mV
- adc@50000000, channel 11: 4095 = 2999 mV
- adc@50000000, channel 12: 4093 = 2997 mV
- adc@50000000, channel 13: 4094 = 2998 mV
ADC reading[6]:
- adc@50000000, channel 1: 4071 = 2981 mV
- adc@50000000, channel 2: 4087 = 2993 mV
- adc@50000000, channel 4: 18 = 13 mV
- adc@50000000, channel 11: 4095 = 2999 mV
- adc@50000000, channel 12: 4094 = 2998 mV
- adc@50000000, channel 13: 4094 = 2998 mV
ADC reading[7]:
- adc@50000000, channel 1: 4095 = 2999 mV
- adc@50000000, channel 2: 4095 = 2999 mV
- adc@50000000, channel 4: 16 = 11 mV
- adc@50000000, channel 11: 4095 = 2999 mV
- adc@50000000, channel 12: 4093 = 2997 mV
- adc@50000000, channel 13: 4093 = 2997 mV
ADC reading[8]:
- adc@50000000, channel 1: 4095 = 2999 mV
- adc@50000000, channel 2: 4095 = 2999 mV
- adc@50000000, channel 4: 16 = 11 mV
- adc@50000000, channel 11: 4094 = 2998 mV
- adc@50000000, channel 12: 4091 = 2996 mV
- adc@50000000, channel 13: 4095 = 2999 mV
[... 267 lines cut ...]
	channel@11 {
		reg = <11>;
		zephyr,gain = "ADC_GAIN_1";
		zephyr,reference = "ADC_REF_INTERNAL";
		zephyr,acquisition-time = <ADC_ACQ_TIME_MAX>;
		zephyr,resolution = <12>;
	};

	channel@12 {
		reg = <12>;
		zephyr,gain = "ADC_GAIN_1";
		zephyr,reference = "ADC_REF_INTERNAL";
		zephyr,acquisition-time = <ADC_ACQ_TIME_MAX>;
		zephyr,resolution = <12>;
	};

	channel@13 {
		reg = <13>;
		zephyr,gain = "ADC_GAIN_1";
		zephyr,reference = "ADC_REF_INTERNAL";
		zephyr,acquisition-time = <ADC_ACQ_TIME_MAX>;
		zephyr,resolution = <12>;
	};
};
```

gautierg-st, 2023-04-11:
I think that in channel@xx, xx is a value in hexadecimal. Could you try again by changing to channel@b, channel@c, and channel@d ?

or-rhssk, 2023-04-11:
Ahh, you're completely right, this can also be seen in the provided overlays for the same sample project (d'oh!).
Nevertheless, after changing the values to hex, the same 3 channels keep measuring near VREF, regardless of pin voltage.

gautierg-st, 2023-04-11:
I have quickly run the samples/drivers/adc on a Nucleo F303K8 (without channels 12 and 13 as they are not present on this chip), and it works as expected, including for channel 11. So I'm not sure what's wrong on your side.
Are you sure you have properly rebuilt your code after the dts change of the channels? Could there be anything HW related on your custom board that could interfere?
What Zephyr version are you using? I'm for my part on latest main.

or-rhssk, 2023-04-11:
Yes, I made a clean build (deleted build folder).
Not sure what HW details might cause this issue. Normally the problematic analog inputs are not connected to anything and for testing I briefly connect them to GND with a jumper wire. Not much that can go wrong there.
I'm also on main: `Booting Zephyr OS build zephyr-v3.3.0-2454-g65101ec68397`

For what it's worth, yesterday I also built the same sample project but on NUCLEO-WB55RG. Added an overlay with 10 ADC channels and had no issues whatsoever. So my initial guess was that the issue is specific to the F303C8 series.

IvanVnucec, 2023-04-16:
> Yes, I made a clean build (deleted build folder). Not sure what HW details might cause this issue. Normally the problematic analog inputs are not connected to anything and for testing I briefly [...]
Did you check F303C8 series errata documents?

gautierg-st, 2023-05-23:
Hi @or-rhssk,

I've tried again to reproduce your issue but I still can't see it. I'm using samples/drivers/adc on a Nucleo F303K8 with the following overlay:
```
#include <zephyr/dt-bindings/adc/adc.h>

/ {
	zephyr,user {
		/* adjust channel number according to pinmux in board.dts */
		io-channels = <&adc1 1>, <&adc1 2>, <&adc1 4>, <&adc1 11>, <&adc1 12>;
	};
};

&adc1 {
	pinctrl-0 = <&adc1_in1_pa0 &adc1_in2_pa1 &adc1_in4_pa3 &adc1_in11_pb0 &adc1_in12_pb1>;
	#address-cells = <1>;
	#size-cells = <0>;

	channel@1 {
		reg = <1>;
		zephyr,gain = "ADC_GAIN_1";
		zephyr,reference = "ADC_REF_INTERNAL";
		zephyr,acquisition-time = <ADC_ACQ_TIME_MAX>;
		zephyr,resolution = <12>;
	};

	channel@2 {
		reg = <2>;
		zephyr,gain = "ADC_GAIN_1";
		zephyr,reference = "ADC_REF_INTERNAL";
		zephyr,acquisition-time = <ADC_ACQ_TIME_MAX>;
		zephyr,resolution = <12>;
	};

	channel@4 {
		reg = <4>;
		zephyr,gain = "ADC_GAIN_1";
		zephyr,reference = "ADC_REF_INTERNAL";
		zephyr,acquisition-time = <ADC_ACQ_TIME_MAX>;
		zephyr,resolution = <12>;
	};

	channel@b {
		reg = <11>;
		zephyr,gain = "ADC_GAIN_1";
		zephyr,reference = "ADC_REF_INTERNAL";
		zephyr,acquisition-time = <ADC_ACQ_TIME_MAX>;
		zephyr,resolution = <12>;
	};

	channel@c {
		reg = <12>;
		zephyr,gain = "ADC_GAIN_1";
		zephyr,reference = "ADC_REF_INTERNAL";
		zephyr,acquisition-time = <ADC_ACQ_TIME_MAX>;
		zephyr,resolution = <12>;
	};
};
```
And here are the results I obtain. I've connected all five channels to the ground successively, and I see them all pass to 0.
```
*** Booting Zephyr OS build zephyr-v3.3.0-4247-g51cc9e0708f2 ***
ADC reading[1]:
- adc@50000000, channel 1: 0 = 0 mV
- adc@50000000, channel 2: 1259 = 922 mV
- adc@50000000, channel 4: 2545 = 1864 mV
- adc@50000000, channel 11: 2528 = 1851 mV
- adc@50000000, channel 12: 2134 = 1562 mV
ADC reading[2]:
- adc@50000000, channel 1: 0 = 0 mV
- adc@50000000, channel 2: 1476 = 1081 mV
- adc@50000000, channel 4: 2818 = 2063 mV
- adc@50000000, channel 11: 1911 = 1399 mV
- adc@50000000, channel 12: 2939 = 2152 mV
ADC reading[3]:
- adc@50000000, channel 1: 2309 = 1691 mV
- adc@50000000, channel 2: 1876 = 1374 mV
- adc@50000000, channel 4: 2791 = 2044 mV
- adc@50000000, channel 11: 1026 = 751 mV
- adc@50000000, channel 12: 2573 = 1884 mV
ADC reading[4]:
- adc@50000000, channel 1: 3328 = 2437 mV
- adc@50000000, channel 2: 0 = 0 mV
- adc@50000000, channel 4: 2624 = 1921 mV
- adc@50000000, channel 11: 965 = 706 mV
- adc@50000000, channel 12: 2989 = 2189 mV
ADC reading[5]:
- adc@50000000, channel 1: 3688 = 2701 mV
- adc@50000000, channel 2: 0 = 0 mV
- adc@50000000, channel 4: 2655 = 1944 mV
- adc@50000000, channel 11: 949 = 695 mV
- adc@50000000, channel 12: 2956 = 2165 mV
ADC reading[6]:
- adc@50000000, channel 1: 3830 = 2805 mV
- adc@50000000, channel 2: 2318 = 1697 mV
- adc@50000000, channel 4: 0 = 0 mV
- adc@50000000, channel 11: 659 = 482 mV
- adc@50000000, channel 12: 2891 = 2117 mV
ADC reading[7]:
- adc@50000000, channel 1: 3893 = 2851 mV
- adc@50000000, channel 2: 2560 = 1875 mV
- adc@50000000, channel 4: 0 = 0 mV
- adc@50000000, channel 11: 1438 = 1053 mV
- adc@50000000, channel 12: 3296 = 2414 mV
ADC reading[8]:
- adc@50000000, channel 1: 3910 = 2863 mV
- adc@50000000, channel 2: 2604 = 1907 mV
- adc@50000000, channel 4: 2558 = 1873 mV
- adc@50000000, channel 11: 952 = 697 mV
- adc@50000000, channel 12: 2985 = 2186 mV
ADC reading[9]:
- adc@50000000, channel 1: 3917 = 2868 mV
- adc@50000000, channel 2: 2647 = 1938 mV
- adc@50000000, channel 4: 2925 = 2142 mV
- adc@50000000, channel 11: 998 = 730 mV
- adc@50000000, channel 12: 2830 = 2072 mV
ADC reading[10]:
- adc@50000000, channel 1: 3917 = 2868 mV
- adc@50000000, channel 2: 2653 = 1943 mV
- adc@50000000, channel 4: 2991 = 2190 mV
- adc@50000000, channel 11: 0 = 0 mV
- adc@50000000, channel 12: 2797 = 2048 mV
ADC reading[11]:
- adc@50000000, channel 1: 3919 = 2870 mV
- adc@50000000, channel 2: 2655 = 1944 mV
- adc@50000000, channel 4: 3006 = 2201 mV
- adc@50000000, channel 11: 0 = 0 mV
- adc@50000000, channel 12: 3014 = 2207 mV
ADC reading[12]:
- adc@50000000, channel 1: 3922 = 2872 mV
- adc@50000000, channel 2: 2663 = 1950 mV
- adc@50000000, channel 4: 3015 = 2208 mV
- adc@50000000, channel 11: 996 = 729 mV
- adc@50000000, channel 12: 3095 = 2266 mV
ADC reading[13]:
- adc@50000000, channel 1: 3917 = 2868 mV
- adc@50000000, channel 2: 2662 = 1949 mV
- adc@50000000, channel 4: 3009 = 2203 mV
- adc@50000000, channel 11: 2636 = 1930 mV
- adc@50000000, channel 12: 0 = 0 mV
ADC reading[14]:
- adc@50000000, channel 1: 3921 = 2871 mV
- adc@50000000, channel 2: 2652 = 1942 mV
- adc@50000000, channel 4: 2999 = 2196 mV
- adc@50000000, channel 11: 1662 = 1217 mV
- adc@50000000, channel 12: 0 = 0 mV
```

or-rhssk, 2023-05-23:
Thanks for the additional test. I'm out of ideas about what else to try and am switching to a different SoC anyways.
So I'm writing this off to some sort of board issue/misconfiguration on my end.

closed 2023-05-23 by or-rhssk