zephyrproject-rtos/zephyr #56727: uart: ITE board semaphore timeouts in UART async API

LukaszMrugala, 2023-04-11:
**Describe the bug**
The UART async API tests ( `drivers.uart.drivers.uart.async_api` ), added to the ITE test suite this Friday, fail on that board. The direct cause is a timeout on the semaphores.

 - Target platform: it8xxx2_evb
 - The issue has been reproduced on at least two boards.
 - `git bisect` points to 2d03aaf99f326ec0c916da1ca7b8725c8b9fdbc8 as the cause.

**To Reproduce**
Steps to reproduce the behavior:
1. `cd <path/to>zephyrproject/zephyr`
2. `source zephyr-env.sh`
3. `twister -p it8xxx2_evb --device-testing -T tests/drivers/uart/uart_async_api/ --west-flash=<path/to>/itetool/loader.sh --device-serial=/dev/<serial/path> --fixture=gpio_loopback`

**Expected behavior**
The tests pass or are still skipped, as before.

**Impact**
Muddles test results, UART async API seems unusable.

**Logs and console output**
handler.log:
```
***** delaying boot 1ms (per build configuration) *****
*** Booting Zephyr OS build zephyr-v3.3.0-2461-gd5f83d2a7f1d (delayed boot 1ms) ***
Running TESTSUITE uart_async_chain_read
===================================================================
START - test_chained_read
M
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:295: uart_async_chain_read_test_chained_read: (k_sem_take(&rx_rdy, K_MSEC(1000)) not equal to 0)
RX_RDY timeout
FAIL - test_chained_read in 1.032 seconds
===================================================================
TESTSUITE uart_async_chain_read failed.
Running TESTSUITE uart_async_chain_write
===================================================================
START - test_chained_write
M
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:664: uart_async_chain_write_test_chained_write: (k_sem_take(&tx_done, K_MSEC(100)) not equal to 0)
TX_DONE timeout
FAIL - test_chained_write in 0.121 seconds
===================================================================
TESTSUITE uart_async_chain_write failed.
Running TESTSUITE uart_async_double_buf
===================================================================
START - test_double_buffer
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:352: uart_async_double_buf_test_double_buffer: (uart_rx_enable(uart_dev, double_buffer[0], sizeof(double_buffer[0]), 50 * USEC_PER_MSEC) not equal to 0)
Failed to enable receiving
FAIL - test_double_buffer in 0.026 seconds
===================================================================
TESTSUITE uart_async_double_buf failed.
Running TESTSUITE uart_async_long_buf
===================================================================
START - test_long_buffers

Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:742: uart_async_long_buf_test_long_buffers: (k_sem_take(&rx_rdy, K_MSEC(200)) not equal to 0)
RX_RDY timeout
FAIL - test_long_buffers in 0.264 seconds
===================================================================
TESTSUITE uart_async_long_buf failed.
Running TESTSUITE uart_async_multi_rx
===================================================================
START - test_multiple_rx_enable
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:159: uart_async_multi_rx_test_multiple_rx_enable: (ret not equal to 0)
uart_rx_enable failed
FAIL - test_multiple_rx_enable in 0.018 seconds
===================================================================
TESTSUITE uart_async_multi_rx failed.
Running TESTSUITE uart_async_read_abort
===================================================================
START - test_read_abort

Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:439: uart_async_read_abort_test_read_abort: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
FAIL - test_read_abort in 0.121 seconds
===================================================================
TESTSUITE uart_async_read_abort failed.
Running TESTSUITE uart_async_single_read
===================================================================
START - test_single_read
t
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:107: uart_async_single_read_test_single_read: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
FAIL - test_single_read in 0.221 seconds
===================================================================
TESTSUITE uart_async_single_read failed.
Running TESTSUITE uart_async_timeout
===================================================================
START - test_forever_timeout

Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:596: uart_async_timeout_test_forever_timeout: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
FAIL - test_forever_timeout in 3.124 seconds
===================================================================
TESTSUITE uart_async_timeout failed.
Running TESTSUITE uart_async_write_abort
===================================================================
START - test_write_abort

Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:516: uart_async_write_abort_test_write_abort: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
FAIL - test_write_abort in 0.121 seconds
===================================================================
TESTSUITE uart_async_write_abort failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -   0.00% [uart_async_chain_read]: pass = 0, fail = 1, skip = 0, total = 1 duration = 1.032 seconds
- FAIL - [uart_async_chain_read.test_chained_read] duration = 1.032 seconds
SUITE FAIL -   0.00% [uart_async_chain_write]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.121 seconds
- FAIL - [uart_async_chain_write.test_chained_write] duration = 0.121 seconds
SUITE FAIL -   0.00% [uart_async_double_buf]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.026 seconds
- FAIL - [uart_async_double_buf.test_double_buffer] duration = 0.026 seconds
SUITE FAIL -   0.00% [uart_async_long_buf]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.264 seconds
- FAIL - [uart_async_long_buf.test_long_buffers] duration = 0.264 seconds
SUITE FAIL -   0.00% [uart_async_multi_rx]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.018 seconds
- FAIL - [uart_async_multi_rx.test_multiple_rx_enable] duration = 0.018 seconds
SUITE FAIL -   0.00% [uart_async_read_abort]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.121 seconds
- FAIL - [uart_async_read_abort.test_read_abort] duration = 0.121 seconds
SUITE FAIL -   0.00% [uart_async_single_read]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.221 seconds
- FAIL - [uart_async_single_read.test_single_read] duration = 0.221 seconds
SUITE FAIL -   0.00% [uart_async_timeout]: pass = 0, fail = 1, skip = 0, total = 1 duration = 3.124 seconds
- FAIL - [uart_async_timeout.test_forever_timeout] duration = 3.124 seconds
SUITE FAIL -   0.00% [uart_async_write_abort]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.121 seconds
- FAIL - [uart_async_write_abort.test_write_abort] duration = 0.121 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: 36bc67bb90464042214819a013d9ba34
PROJECT EXECUTION FAILED
```

device.log
```
[sudo] password for <user>:
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.5 LTS 
 - SDK: Zephyr SDK
 - Commit: d5f83d2a7f1d3c7031bc2eec6b72e7261234bcba

Dino-Li, 2023-04-21:
Since this test requires the TX/RX to be connected together, I connected them together and the first test (test_chained_read) no longer had a semaphore timeout. But the test still fails with the incorrect read length.
Does ns16550's async api set correct rx length for event? Did this test pass on other boards using the ns16550 driver?

Dino-Li, 2023-04-24:
Duplicate: https://github.com/zephyrproject-rtos/zephyr/issues/57103

closed 2023-06-06 by pkunieck