zephyrproject-rtos/zephyr #56734: mgmt/MCUmgr: Compile time option blocking image confirmation on secondary slot

de-nordic, 2023-04-11:
**Is your feature request related to a problem? Please describe.**
MCUmgr is now able to confirm image in secondary slot, making it permanent boot application from the next boot.
This is ok, but in case when that application is swapped to the primary slot and fails to boot, the device gets bricked.

To avoid above issue user should only mark application in secondary slot for test and allow it to confirm itself once if is able to boot and do self-check or only confirm app in first slot. This is so far not enforced in any way, so mistakes can happen.

**Describe the solution you'd like**
This feature request proposes addition of Kconfig option that would disable confirming application images in secondary slot, allowing to avoid confirming image that could not boot, as MCUboot is, in most cases, able to revert to previously working application.

**Describe alternatives you've considered**
Leaving as is.

mateusz-, 2024-02-15:
Hopefully not a stupid question, not super familiar with the firmware update process.

Would it make sense to extend this feature request to block image pending as well? That way, the application would be fully in control of the process once image is received.

Seems like a feature that could potentially be desirable?

nordicjm, 2024-02-16:
> Hopefully not a stupid question, not super familiar with the firmware update process.

All questions are welcome

> Would it make sense to extend this feature request to block image pending as well? That way, the application would be fully in control of the process once image is received.

Not sure what you mean by this, can you elaborate? An update would be loaded, then marked for test (or confirmed outright), the device reboots and mcuboot swaps the images, if the image was only marked for test then the newly booted application needs to confirm itself or it will be reverted on the next reboot

mateusz-, 2024-02-16:
Basically, what I am thinking is **in addition to** disabling confirmation of application images, disabling pending state as well from MCUmgr, and letting the firmware handle **both** things. Maybe 2 different KConfig options?

The flow would be:

1. Firmware update transfer is completed.
2. Firmware application sets pending and reboots.
3. MCUboot swaps the application.
4. New firmware application boots and confirms.

Once 1) happens, MCUmgr is done.

nordicjm, 2024-02-19:
A Kconfig to set that would be a nice addition

de-nordic, 2024-02-19:
> Basically, what I am thinking is **in addition to** disabling confirmation of application images, disabling pending state as well from MCUmgr, and letting the firmware handle **both** things. Maybe 2 different KConfig options?

Yeah, why not.

> 
>     2. Firmware application sets pending and reboots.

For that part we already have a mechanism: 
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/mgmt/mcumgr/grp/img_mgmt/src/img_mgmt.c#L691

nordicjm, 2025-02-03:
Perplexed with `CONFIG_MCUMGR_GRP_IMG_ALLOW_CONFIRM_NON_ACTIVE_IMAGE_SECONDARY` and `CONFIG_MCUMGR_GRP_IMG_ALLOW_CONFIRM_NON_ACTIVE_IMAGE_ANY` Kconfig options, they do not seem to be working, in the default configuration where `CONFIG_MCUMGR_GRP_IMG_ALLOW_CONFIRM_NON_ACTIVE_IMAGE_SECONDARY` is enabled and the other isn't, if 2 images are uploaded to secondaries, both can be marked as confirmed prior to rebooting which doesn't match the Kconfig descriptions