zephyrproject-rtos/zephyr #56749: Bluetooth: Host: Audio SDU fragmented when ISO Tx MTU equals Controller's ISO Tx Buffer Size

cvinayak, 2023-04-12:
**Describe the bug**
> Observations
To allow 251 bytes of ISO payload over the HCI interface the CONFIG_BT_CTLR_ISO_TX_BUFFER_SIZE needs to be set to a value higher than 251 bytes (as that is the value returned in read_buffer_size_v2). but the name suggests the HCI buffer limit is also used as ISO TX limit, which seems odd and wrong - the MAX ISO TX PDU is 251 bytes.
If this value is set to 251 it forces the host to fragment the 251 byte ISO data payload and thereby use 2 credits per ISO PDU.

Refer to https://github.com/zephyrproject-rtos/zephyr/issues/52055#issue-1439858139

Using a `CONFIG_BT_ISO_TX_MTU=40` and `CONFIG_BT_CTLR_ISO_TX_BUFFER_SIZE=40`, after couple PDUs transmitted over air, no more PDUs are transmitted. SDU is fragment into multiple HCI ISO Data.

In the HCI specification, Data_Total_Length, which can be less than or equal to ISO_Data_Packet_Length returned by the controller, excludes the packet header. Hence, ISO Tx MTU size of 40 bytes shall not be fragmented when ISO Tx buffer size returned by Controller is 40 bytes.

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir -p build/broadcast_audio_source; cd build/broadcast_audio_source
2. cmake -GNinja -DBOARD=nrf52_bsim -DCONFIG_BT_ISO_TX_MTU=40 -DCONFIG_BT_CTLR_ISO_TX_BUFFER_SIZE=40 -DCONFIG_BT_CTLR_ADV_ISO_PDU_LEN_MAX=40 -DCONFIG_BT_CTLR_ADV_ISO_STREAM_MAX=2 -DCONFIG_BT_CTLR_ISOAL_SOURCES=2 ../../samples/bluetooth/broadcast_audio_source
3. ninja
4. ./zephyr/zephyr.exe -s=iso -d=0 &
5. ~/bsim/bin/bs_2G4_phy_v1 -s=iso -D=1 -sim_length=2e6
6. Observe the PDUs transmitted by analysing the babblesim exported trace

**Expected behavior**
SDU shall not be fragmented when ISO Tx MTU equals Controller's ISO Tx Buffer Size.

**Impact**
showstopper

**Logs and console output**

[Broadcast_audio_source_40_40.zip](https://github.com/zephyrproject-rtos/zephyr/files/11208195/Broadcast_audio_source_40_40.zip)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: b76bd15

**Additional context**
None

closed 2023-05-03 by carlescufi

ivaniushkov, 2023-09-29:
It looks like CONFIG_BT_CTLR_ISO_TX_BUFFER_SIZE and CONFIG_BT_ISO_TX_MTU values have different meaning in Host and in Controller:
CONFIG_BT_ISO_TX_MTU in Zephyr Host includes only actual user data transmitted in SDU (i.e., it is data transmitted in ISO_SDU_Fragment fields of HCI ISO Data packets).

BT_CTLR_ISO_TX_BUFFER_SIZE in Zephyr Controller is the buffer size that Controller returns to Host in response to HCI LE Read Buffer Size V2 command. According to Core Spec, this value should include optional fields of HCI ISO Data packet: ISO_SDU_Length, Packet_Sequence_Number, ISO_SDU_Length and Packet_Status_Flag.

So I think the changes made in https://github.com/zephyrproject-rtos/zephyr/pull/57177 are not solving the actual issue. I am partially reverting these changes in separate PR: https://github.com/zephyrproject-rtos/zephyr/pull/63244 because these changes introduced bugs related to SDU fragmentation to multiple HCI packets

cvinayak, 2023-09-29:
@ivaniushkov  please read the definition as in the Spec:

[image]

Data_Total_Length: is Length of the packet, excluding the packet header, in octets.

ivaniushkov, 2023-09-29:
@cvinayak 
Thank you for clarifiaction!
According to Spec, packet header is first 4 octets of packet only:
[image]

So I think you are right, Data_Total_Length field doesn't include header length. It only includes ISO_SDU_Fragment + optional fields (Time_Stamp, Packet_Sequence_Number, ISO_SDU_Length, RFUs).

In other words, for the HCI ISO Data packet including the whole SDU without timestamp (PB_Flag=0b10, TS_Flag=0) with 4 bytes of data in ISO_SDU_Fragment, the following values should be set in length fields:
- Data_Total_Length=8
- ISO_SDU_Length=4

cvinayak, 2023-09-29:
You are correct, `The HCI ISO Data packet header is the first 4 octets of the packet.`

Which implies that to transmit a ISO PDU of 251 bytes, the LE Read Buffer Size has to return a value of 255.

Hence, I acknowledge the fix in #63244, but the test should keep  `CONFIG_BT_ISO_TX_MTU=251`, `CONFIG_BT_ISO_RX_MTU=251` and increase `CONFIG_BT_CTLR_ISO_TX_BUFFER_SIZE` to 255.

PR #57177 (2023-05-03): Bluetooth: host: don't fragment ISO if len <= MTU
MTU doesn't count against the ISO and ISO data headers.

Then a config with CONFIG_BT_ISO_TX_MTU ==
CONFIG_BT_CTLR_ISO_TX_BUFFER_SIZE should not fragment SDUs over HCI.

Also set the TS_Flag bit if a timestamp is present.

Fixes #56749
commit e460847b (2023-05-03): Bluetooth: host: don't fragment ISO if len <= MTU

MTU doesn't count against the ISO and ISO data headers.

Then a config with CONFIG_BT_ISO_TX_MTU ==
CONFIG_BT_CTLR_ISO_TX_BUFFER_SIZE should not fragment SDUs over HCI.

Also set the TS_Flag bit if a timestamp is present.

Fixes #56749