zephyrproject-rtos/zephyr #56769: test: mimxrt595_evk_cm33: tests/kernel/usage/thread_runtime_stats/ test fail

hakehuang, 2023-04-12:
The testcase "tests/kernel/usage/thread_runtime_stats/" fails with assertion.

**Describe the bug**
run the kernel usage case

Please also mention any information which could help others to understand
the problem you're facing:
 
this case only fails on mimxrt595_evk_cm33, and bisect found it is caused by below commit

```
commit b79e74404373827619425912fec5e4ab59bd692e
Author: Mahesh Mahadevan <mahesh.mahadevan@nxp.com>
Date:   Tue Nov 29 16:13:21 2022 -0600

    boards: mimxrt595evk: Setup the regulator for Run mode

    Setup the appropriate voltage in the PMIC based on the
    core frequency.
    Add labels to the regulators for easy access to the nodes.
    Add CONFIG_REGULATOR to the defconfig so we can setup the
    PMIC voltages during platform startup.

    Signed-off-by: Mahesh Mahadevan <mahesh.mahadevan@nxp.com>

 boards/arm/mimxrt595_evk/board.c                   | 74 ++++++++++++++++++++++
 boards/arm/mimxrt595_evk/mimxrt595_evk_cm33.dts    |  8 +--
 .../arm/mimxrt595_evk/mimxrt595_evk_cm33_defconfig |  2 +-
 3 files changed, 79 insertions(+), 5 deletions(-)

```

**To Reproduce**
Steps to reproduce the behavior:
1. cd tests/kernel/usage/thread_runtime_stats
2. west build -b mimxrt595_evk_cm33
3. west flash
4. See error

**Expected behavior**
clock count for this board is not stable after setting regulator

**Impact**
clock stable issue

**Logs and console output**

```

*** Booting Zephyr OS build zephyr-v3.2.0-2259-g6cb5c5519e4d ***
Running TESTSUITE usage_api
===================================================================
START - test_all_stats_usage
    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/usage/thread_runtime_stats/src/test_thread_runtime_stats.c:108: usage_api_test_all_stats_usage: stats1.execution_cycles == stats1.total_cycles is false
 FAIL - test_all_stats_usage in 0.021 seconds
===================================================================
START - test_sys_stats_enable_disable
 PASS - test_sys_stats_enable_disable in 0.001 seconds
===================================================================
START - test_thread_stats_enable_disable
 PASS - test_thread_stats_enable_disable in 0.002 seconds
===================================================================
START - test_thread_stats_usage
 PASS - test_thread_stats_usage in 0.006 seconds
===================================================================
TESTSUITE usage_api failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -  75.00% [usage_api]: pass = 3, fail = 1, skip = 0, total = 4 duration = 0.030 seconds
 - FAIL - [usage_api.test_all_stats_usage] duration = 0.021 seconds
 - PASS - [usage_api.test_sys_stats_enable_disable] duration = 0.001 seconds
 - PASS - [usage_api.test_thread_stats_enable_disable] duration = 0.002 seconds
 - PASS - [usage_api.test_thread_stats_usage] duration = 0.006 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
PROJECT EXECUTION FAILED

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux,)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: zephyr-v3.3.0-2500-gafaf4cddd2

**Additional context**
this is a long existing bug, but hidded by other timer issues. so I did not report it earily.

hakehuang, 2023-04-12:
@dleach02 , @mmahadevan108 , @danieldegrasse .
this is a long existing bug, but hidded by other timer issues. so I did not report it earily. Sorry for this missing

nashif, 2023-04-18:
@peter-mitsis  any idea why this might be failing?

peter-mitsis, 2023-04-18:
Missed this. Will take a look.

peter-mitsis, 2023-04-18:
Broadly speaking, the cause of the failure can be thought to be either in the code for gathering the statistics, or the test code. Given both that the stats sampling is performed is generic kernel C code (and not architecture assembly), and that the test failure is isolated to the mimxrt595_evk_cm33, I am leaning towards that an assumption made in the test code is not holding true for this board.

What assumptions? One of them is that the system did not idle prior to this test. If this assumption ever gets violated, then that could cause the `zassert_true(stats1.execution_cycles == stats1.total_cycles);` check to fail. Unfortunately, I do not have the hardware to validate/test this.

We can test if the above accurately describes the above with by modifying the test to add the following lines to the start of test_all_stats_usage().

```
        struct k_thread_runtime_stats baseline;
        k_thread_runtime_stats_all_get(&baseline);   /* Get baseline stats */
        printk("Baseline execution: %lld, total; %lld\n", baseline.execution_cycles, baseline.total_cycles);
```
If the two values are different, then it strongly suggests that somewhere during the initialisation before the test ran, some thread had a delay or sleep.

If this turns out to show that the two values are different at this baseline stage, then there are two options.
1. Track down where the idle is coming from.
2. Update the test to perform its checks relative to the baseline.

hakehuang, 2023-04-19:
@peter-mitsis , thanks for this analysis, here is the log, which shows the values are different.

```
Running TESTSUITE usage_api
===================================================================
START - test_all_stats_usage
Baseline execution: 20149, total; 17477

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/usage/thread_runtime_stats/src/test_thread_runtime_stats.c:112: usage_api_test_all_stats_usage: stats1.execution_cycles == stats1.total_cycles is false

 FAIL - test_all_stats_usage in 0.024 seconds
===================================================================
START - test_sys_stats_enable_disable
 PASS - test_sys_stats_enable_disable in 0.001 seconds
===================================================================
START - test_thread_stats_enable_disable
 PASS - test_thread_stats_enable_disable in 0.002 seconds
===================================================================
START - test_thread_stats_usage
 PASS - test_thread_stats_usage in 0.006 seconds
===================================================================
TESTSUITE usage_api failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -  75.00% [usage_api]: pass = 3, fail = 1, skip = 0, total = 4 duration = 0.033 seconds
 - FAIL - [usage_api.test_all_stats_usage] duration = 0.024 seconds
 - PASS - [usage_api.test_sys_stats_enable_disable] duration = 0.001 seconds
 - PASS - [usage_api.test_thread_stats_enable_disable] duration = 0.002 seconds
 - PASS - [usage_api.test_thread_stats_usage] duration = 0.006 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION FAILED

```

peter-mitsis, 2023-04-24:
@hakehuang - As a followup on this item, would you happen to know, or be able to identify the cause of this discrepancy between execution and total cycles? I ask because that will ultimately direct our course of action on how to best resolve it. For example, if the discrepancy is of a legitimate origin, then updating the test would probably be the best bet. However, if it is not of a legitimate origin, then updating the test could merely mask the problem.  I realise that this also leaves open the question "What is a legitimate origin?", but for now, if we can find out the cause of the discrepancy, that would be a step in the right direction.

hakehuang, 2023-04-25:
@peter-mitsis , I assume this would be some driver calll the sleep in initialization, as you seeumption 1
Track down where the idle is coming from.
I first review the related code, and did not find any sleep call, and there are some mutex called in drivers, do you have any suggestion on narrow down the issue? Thanks in advance.

mmahadevan108, 2023-04-25:
@hakehuang, should we reassign this issue?

hakehuang, 2023-04-26:
> @hakehuang, should we reassign this issue?

I guess not for now, still in analysis the root cause with peter.

hakehuang, 2023-04-26:
> don't think that Zephyr is ready to issue a call to k_sem_take() at this point. 

@peter-mitsis , even in the INIT_LEVEL_POST_KERNEL stage, sem can not use? it is True? what is the condition for semaphar to be usable?

```
#0  mcux_flexcomm_configure (dev=dev@entry=0x1800b588 <__device_dts_ord_75>, dev_config_raw=20) at /home/shared/disk/zephyr_project/zephyr_test/zephyr/drivers/i2c/i2c_mcux_flexcomm.c:53
#1  0x18004894 in mcux_flexcomm_init (dev=0x1800b588 <__device_dts_ord_75>) at /home/shared/disk/zephyr_project/zephyr_test/zephyr/drivers/i2c/i2c_mcux_flexcomm.c:371
#2  0x180077b0 in z_sys_init_run_level (level=level@entry=INIT_LEVEL_POST_KERNEL) at /home/shared/disk/zephyr_project/zephyr_test/zephyr/kernel/init.c:258

```

peter-mitsis, 2023-04-26:
The initialisation system is one of my weaker areas, so if I get something wrong there, it is not too surprising to me. But hey, it is a chance to learn, so hopefully I won't make the same mistake in the future. :) That being said, it looks like I was wrong--semaphores can be used at that point in time.

My initial look at this suggests that it is dependent upon I2C and the initialization routine does some I2C work which in turn does some semaphore work that can block. I am not familiar with this block of code, but my gut says that it is legitimate. This in turn would suggest that the kernel/usage/thread_runtime_stats should be updated to account for this behavior.

@gmarull - Can you confirm that the regulator_pca9420.c initialization code may legitimately result in blocking behavior that causes the system to idle for a period of time?

gmarull, 2023-04-26:
> @gmarull - Can you confirm that the regulator_pca9420.c initialization code may legitimately result in blocking behavior that causes the system to idle for a period of time?

regulators are potentially blocking, so it's possible.

peter-mitsis, 2023-04-26:
I would not expect it to matter whether we use a semaphore or mutex in this instance.

hakehuang, 2023-04-27:
> I would not expect it to matter whether we use a semaphore or mutex in this instance.

in current regulator implement, for each regulator domain, we use a different i2c instance, which means we may initial the semaphor multiply times, although we set the lock count as 1, but we have multiply instance for lock with 1 count, and sharing with one physical i2c bus. If my understanding is true, then the sem does not take any effects for regulator usage case.

hakehuang, 2023-05-11:
@mmahadevan108 , there are many other kernel cases which will enter deep sleep mode, like kernel/timer/time_api, can we by default add the wakeup source to RT595 and RT685. otherwise the CI testing will fail due to board is in deep sleep and can not wakeup.

mmahadevan108, 2023-05-12:
@hakehuang, I add a PR to make this the default on RT595 and RT685 EVK boards.

hakehuang, 2023-07-14:
@mmahadevan108 , as we disable the regulator by  1222b5abeca6ee35c2b5a778886ad8f3cb87fa47, this issue is hidden. maybe we can close it for now.

closed 2023-09-13 by hakehuang