zephyrproject-rtos/zephyr #56774: net: lwm2m: Registration retry behavior on timeout

mlasch, 2023-04-12:
There are some confusing and in my opinion redundant LwM2M retry settings in the LwM2M client code.

First, there is the `CONFIG_COAP_MAX_RETRANSMIT` Kconfig setting which controls the number of retransmissions for CON transactions like the registration message. This works as expected. I set  `CONFIG_COAP_MAX_RETRANSMIT=2` and the client does only two retransmissions before starting over with a new transaction.

On the other hand there is the `CONFIG_LWM2M_RD_CLIENT_MAX_RETRIES` Kconfig entry for which according to the description, should do the same as `CONFIG_COAP_MAX_RETRANSMIT` but only for registration messages? After tests with the current code, I can't figure out what `CONFIG_LWM2M_RD_CLIENT_MAX_RETRIES` is intended to do, it seems to be ignored.

Timeouts do not result in `ENGINE_NETWORK_ERROR` as indicated in the `CONFIG_LWM2M_RD_CLIENT_MAX_RETRIES` help text:

> Specify maximum number of registration retries, before the application
> is notified about the network failure. Once application is notified,
> it's up to the application to handle this situation in a way
> appropriate for the specific use-case (for instance by waiting for
> LTE link to be re-established).

On reaching the `CONFIG_COAP_MAX_RETRANSMIT` limit, two events are processed: `LWM2M_RD_CLIENT_EVENT_REG_TIMEOUT` and `LWM2M_RD_CLIENT_EVENT_DISCONNECT`.
```
[00:00:59.860,000] <wrn> net_lwm2m_rd_client: Registration Timeout
[00:00:59.860,000] <dbg> net_lwm2m_client_app: rd_client_event: Disconnected
[00:00:59.860,000] <dbg> net_lwm2m_client_app: rd_client_event: Registration timeout!

```

**Steps to reproduce the behavior:**

Build with different settings of `CONFIG_COAP_MAX_RETRANSMIT` and `CONFIG_LWM2M_RD_CLIENT_MAX_RETRIES` and observe the behavior:
```
west build -p auto -b native_posix -d build-native zephyr/samples/net/lwm2m_client -- -DCONFIG_NATIVE_UART_0_ON_STDINOUT=y -DCONFIG_COAP_MAX_RETRANSMIT=2 -DCONFIG_LWM2M_RD_CLIENT_MAX_RETRIES=1
```

Tested with release `3.2.0` and current `main`.

So to summarize, the question would be, whats does, or should `CONFIG_LWM2M_RD_CLIENT_MAX_RETRIES` do and how does it fit together with `CONFIG_COAP_MAX_RETRANSMIT`?

SeppoTakalo, 2023-04-19:
Thank you for the question @mlasch 

If you use non-secure LwM2M, these two values seem to do almost the same.
However, they do have different purposes and it is visible when you use DTLS.

CoAP retry can handle single packets dropped from the application layer traffic. If that can recover, then there is no network failures assumed.
But when network NAT timeout have happened, CoAP retry can not recover. But if there is still lifetime left, recovering the DTLS session might recover the LwM2M connection.
So the difference between CoAP retries and RD_CLIENT_MAX_RETRIES is that socket is closed in between RD client retries.

Here is a relevant piece from the state machine diagram:

[image]

SeppoTakalo, 2023-04-19:
Here is a Wireshark capture of the effect. Here CoAP retries fail to recover the connection, so we restart the DTLS and it recovers
[image]