zephyrproject-rtos/zephyr #56782: k_thread_cpu_pin function definition is wrong

rjuez00, 2023-04-12:
The definition of function k_thread_cpu_pin is:
`k_thread_cpu_pin(k_tid_t thread, int cpu)`
but this is wrong, the function actually needs the pointer to the k_tid_t.

In the file samples/syncronization/src/main.c we can see that the function is called like:		`k_thread_cpu_pin(&threadA_data, 0);`

and threadA_data is defined:
`static struct k_thread threadA_data;`

which means that k_thread_cpu_pin actually needs the pointer, not the variable.

credits to @Petrus97

Archie-Elvie, 2023-04-14:
I think the usage in the sample is a bit confusing but isn't wrong as `k_tid_t;` is a typedef of ` struct k_thread*`. So the sample should really be
```c
	k_tid_t threadA_id = k_thread_create(&threadA_data, threadA_stack_area,
			K_THREAD_STACK_SIZEOF(threadA_stack_area),
			threadA, NULL, NULL, NULL,
			PRIORITY, 0, K_FOREVER);
	k_thread_name_set(&threadA_data, "thread_a");
#if PIN_THREADS
	if (arch_num_cpus() > 1) {
		k_thread_cpu_pin(threadA_id , 0);
	}
#endif
```

andyross, 2023-04-14:
The `k_tid_t` type is a typedef of a pointer to a `struct k_thread`.  Indeed, you want to take the pointer to the thread, but that's what the API says.

And yes, it's a little confusing to have an API that mixes handles and pointers like this, which is why we generally (as Linux does) disfavor typedef handles in APIs.   But `k_tid_t` is a legacy thing from very early Zephyr versions.  We're stuck with it.

closed 2023-04-14 by andyross