zephyrproject-rtos/zephyr #56789: Application userspace/hello_world_user in SoC LiteX VexRiscv

KevinQhv, 2023-04-12:
Hi, 

I try an application provided by Zephyr `userspace/hello_world_user` on my SoC that i created with LiteX.

```
Configuration :
- OS : Ubuntu 22.04 LTS
- Board : Digilent Arty A7 100T
- LiteX Version 2022.08
- CPU : VexRiscv Variant : Secure, Linux, Full, Lite
- OS : ZephyrOS lastest 3.1.0
```

I tried the application with several variants Full, Lite, Linux and Secure for the VexRiscv CPU. But, I don't enter the thread in user mode. The `k_is_user_context() ` function does not return 1 for user mode for all variants. 

```
ASSERTION FAIL [k_is_user_context()] @ WEST_TOPDIR/zephyr/samples/userspace/hello_world_user/src/main.c:19
        User mode execution was expected
E:      a0: 00000004    t0: 00000000
E:      a1: 0000001c    t1: 00000000
E:      a2: 400053c8    t2: ffffffff
E:      a3: 40006d84    t3: 00000020
E:      a4: 00000000    t4: 00000023
E:      a5: f0003000    t5: ffffffff
E:      a6: 20000000    t6: ffffffff
E:      a7: 00000000
E:      ra: 40000ac0
E:    mepc: 40001da0
E: mstatus: 00001880
E: 
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x400063d0 (unknown)
E: Halting system
```

So, i got an ASSERTION FAIL with the function `__ASSERT(k_is_user_context()` but, I want to enter the user mode. The Linux, Full and Lite variants have three privilege modes.

Thank you for your help

fkokosinski, 2023-06-14:
Hi @KevinQhv!

I've looked into what you've described, and:
1. `__ASSERT(k_is_user_context()` will always fail if `CONFIG_ARCH_HAS_USERSPACE` is n-selected in Kconfig.
2. `CONFIG_ARCH_HAS_USERSPACE` is selected only by `CONFIG_RISCV_PMP`, which `litex_vexriscv` doesn't select.
3. `litex_vexriscv` targets the `standard` variant of VexRiscv, which doesn't support PMP.

So in summary - the sample seems to work as expected, and the assertion failure is to be expected because the current `litex_vexriscv` platform doesn't support PMP:
```
*** Booting Zephyr OS build v3.4.0-rc3-59-g3104ad0e393c ***
Hello World from privileged mode. (litex_vexriscv)
ASSERTION FAIL [k_is_user_context()] @ WEST_TOPDIR/zephyr/samples/usersp
ace/hello_world_user/src/main.c:19
 User mode execution was expected
E:      a0: 00000004    t0: 00000000
E:      a1: 00000013    t1: 00000000
E:      a2: 40008248    t2: 00000000
E:      a3: 4000a284    t3: 00000020
E:      a4: 00000000    t4: 00000023
E:      a5: e0002000    t5: 00000000
E:      a6: 10000000    t6: 00000000
E:      a7: 00000000
E:      ra: 40000af8
E:    mepc: 4000233c
E: mstatus: 00001880
E: 
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x400098c8 (unknown)
E: Halting system
```

fkokosinski, 2023-06-14:
> - OS : ZephyrOS lastest 3.1.0

Note that Zephyr 3.1.0 is not the latest one. :)

KevinQhv, 2023-06-15:
Hi ! 

I added `CONFIG_RISCV_PMP=y` to the `zephyr/boards/litex_vexriscv/litex_vexriscv_defconfig` file and now applications with `USERSPACE` work. 

Thanks

closed 2023-06-15 by KevinQhv