zephyrproject-rtos/zephyr #56795: Bluetooth: only one adv report for extended active scanning

jori-nordic, 2023-04-13:
**Describe the bug**
The Zephyr controller seems to not pend two reports for every scan request.

If one stops the advertiser as soon as it is scanned, the scanner will only ever get a report with the `BT_GAP_ADV_PROP_SCANNABLE | BT_GAP_ADV_PROP_EXT_ADV` bits set.

If the advertiser is left running, the scanner eventually gets a scan report with the `BT_GAP_ADV_PROP_SCAN_RESPONSE` bit, as expected.

The specification clearly shows that it should send one after `AUX_ADV_IND` and another one after `AUX_SCAN_RSP`.
Vol. 6, Part D 4.6 "ACTIVE SCANNING ON THE SECONDARY ADVERTISING PHYSICAL CHANNEL"
[image]

**To Reproduce**
Run the `tests/bsim/bluetooth/host/privacy/device`, after modifying it slightly:
in `tests/bsim/bluetooth/host/privacy/device/src/test_undirected_peripheral.c`, at the end of `test_peripheral_main()`, remove the special case for the zephyr controller that doesn't stop the advertiser.

**Expected behavior**
Two scan reports for each (`AUX_ADV_IND`, `AUX_SCAN_RSP`) tuple. 
One of those with the `BT_GAP_ADV_PROP_SCAN_RESPONSE` bit set.

**Impact**
Application has to work around this. This does not have a big impact

cvinayak, 2023-04-13:
What is values for `CONFIG_BT_BUF_EVT_RX_COUNT` and `CONFIG_BT_CTLR_RX_BUFFERS` ? Would increasing these help?

jori-nordic, 2023-04-13:
3 and 1 respectively.
Increasing to 30 and 10 didn't change anything.

cvinayak, 2023-04-13:
Zephyr Controller always stops the radio events immediately, does not wait for Transmissions to complete. This is to ensure deterministic command complete response time.

If the advertising is stopped on reception of scan req event, then the scan response will not be transmitted.

This is not an implementation issue, the Controller behaves as expected as requested by the test to stop the advertising state.