zephyrproject-rtos/zephyr #56841: Bluetooth: controller: Error in handling invalid PDU in LLCP, leading to ASSERT

erbr-ot, 2023-04-14:
A situation with a central not rejecting overlapping local initiated PHY_REQ, but instead sends PHY_UPD_IND has been seen, leading to a situation where 'our' peripheral will have both remote and local procedure expecting a PHY_UPD_IND PDU and will thus hit the ASSERT 'That cannot happen' in ull_cp_rx(). 

Solution is to refactor and handle the invalid behaviour as elsewhere - and terminate connection.
Also refactor to ensure that a local procedure that is 'overtaken' by a central initiated counterpart should no longer expect PDUs other that UNKNOWN_RSP and REJECTs

This is fixed by: https://github.com/zephyrproject-rtos/zephyr/pull/56840

henrikbrixandersen, 2023-04-17:
Please use our bug template when reporting bugs. You need to edit this issue to include the needed information from https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/bug_report.md

carlescufi, 2023-04-18:
Fixed in https://github.com/zephyrproject-rtos/zephyr/pull/56840

closed 2023-04-18 by carlescufi