zephyrproject-rtos/zephyr #56845: samples: drivers :adc  error in main.c  printk issue with twister test

Desvauxm-st, 2023-04-14:
Describe the bug

our twister test sample/drivers/adc  with all board are failed
due to the modification into sample/drivers/adc/src/main.c
in line 58:
printk("ADC reading[%u]:\n", count++);

The output on the console is no longer aligned with what is written in the sample.yaml
boards: all board
the bug is due to the PR  sensor: Add NTC thermistor sensor for TDK Robokit1 board #54947

To Reproduce
lunch any test in console with 

Expected behavior
our twister test sample/drivers/adc  with all board must be pass
and
The output on the console must be  aligned with what is written in the sample.yaml
boards: all board

Impact
 our twister test sample/drivers/adc  with all board are failed

Logs and console output
*** Booting Zephyr OS build zephyr-v3.3.0-2587-g9d041dd6f2d3 ***
ADC reading[0]:
- adc@40022000, channel 0: 3601 = 181 mV
ADC reading[1]:
- adc@40022000, channel 0: 2463 = 124 mV

Environment (please complete the following information):

OS: (e.g. Linux,)
Toolchain (e.g Zephyr SDK, ...)

Additional context
@asemjonovs
 the bug is due to the PR  sensor: Add NTC thermistor sensor for TDK Robokit1 board #54947

I suggest to change the sample.yaml with something like that ( remove : after "ADC reading )
before :
          tests:
            sample.drivers.adc:
              *****
                type: multi_line
                regex:
                  - "ADC reading:"                 
after :
          tests:
            sample.drivers.adc:
              *****
                type: multi_line
                regex:
                  - "ADC reading"

asemjonovs, 2023-04-17:
@Desvauxm-st - can you provide exact console command you're using to reproduce?  I'm observing tests passing with 
`west twister -T samples/drivers/adc/`.  

Your suggestion to remove the `:` from 
```
      regex:
        - "ADC reading:"  # <-- remove ':' here
```
in  `samples/drivers/adc/sample.yaml`
sounds reasonable to me.  Does that resolve your issue?

Desvauxm-st, 2023-04-18:
@asemjonovs 
yes  right

closed 2023-04-18 by nashif

asemjonovs, 2023-04-18:
@Desvauxm-st - looks like https://github.com/zephyrproject-rtos/zephyr/issues/56750 fixes your issue