zephyrproject-rtos/zephyr #56865: ESP32 crash in kernel shell when CONFIG_ASSERT=y

jmontgomery-cruise, 2023-04-14:
**Describe the bug**
The `kernel stack` and `kernel threads` commands crash on esp32s2_saola when CONFIG_ASSERT=y. 
Here are some screenshot logs demonstrating the issue:
[image]
[image]

Note that the problem will rarely manifest as above. It more commonly manifests as a simple crash, like this:
[image]

From an initial inspection, the problem appears to be related to printing and/or the UART. Enabling CONFIG_LOG_MODE_IMMEDIATE causes the board to stop crashing. Commenting out [L193 in kernel_service.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/shell/modules/kernel_service.c) also stops the crashing for the relevant command. Moreover, the issue does not appear to be related to the specific information being printed, as even "Hello, World!" in this function will manifest the crash. 

I have only tested this behavior on esp32s2_saola, but I suspect that it may be replicable on other platforms.

**To Reproduce**
Steps to reproduce the behavior:
1. append CONFIG_ASSERT=y to samples/subsys/shell/shell_module/prj.conf
2. `west build -b <board> samples/subsys/shell/shell_module --pristine`
3. Flash board and drop into UART
4. Run `kernel stacks` or `kernel threads`
5. Board will crash

**Expected behavior**
The system should print the kernel stack and thread info respectively.

**Impact**
This bug is only an annoyance right now, but may be indicative of deeper issues.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: Able to replicate from 3.2.0 onwards. Currently testing at main / 76078ea63ce77fa32260d82e913115569731f849

**Additional context**
The assert being implicated here ensures that there are no locks being held during context switches. I believe the lock violating that is held by the k_thread_foreach() call in the `kernel stacks` and `kernel threads` commands. However, I am not able to confirm that and I am not sure why deferred logging would cause a context switch.

carlescufi, 2023-04-18:
This is likely a stack overflow. `CONFIG_ASSERT=y` requires more stack.

jmontgomery-cruise, 2023-04-18:
I've raised the stacks of every thread in the system to 8k or beyond without seeing any change in behavior. Additionally, the canaries aren't being tripped in any way that I can observe. It's still possible the stack is overflowing, but somewhat less likely.

sam131208, 2023-04-22:
I'm also having a similar issue, perhaps related to the multiplexing feature of uart, which is also a programming interface.
   In addition, it has a problem, when burned, there is only one chance to connect to the uart output log, and if you try to open it again after closing the connection, it will cause Zephyr to crash.
  My simple response is to turn off uart after the test is complete：
   CONFIG_LOG=n
   CONFIG_STDOUT_CONSOLE=n
   CONFIG_CONSOLE=n
   CONFIG_SERIAL=n
   CONFIG_UART_CONSOLE=n

closed 2023-07-06 by github-actions[bot]