zephyrproject-rtos/zephyr #56883: Board init for particle_boron runs too late

weinholt, 2023-04-15:
**Describe the bug**
The `board_particle_boron_init()` function runs at init priority 99. It configures GPIOs that are needed for using the board's modem.

The board has a u-blox SARA modem that can be used with the `gsm_ppp.c` and `ublox-sara-r4.c` drivers. However, these by default have init priorities 42 and 80. They run before the board init has configured the GPIOs. 

**To Reproduce**
1. Use this overlay:
CONFIG_RTT_CONSOLE=y
CONFIG_USE_SEGGER_RTT=y
CONFIG_MODEM=y
CONFIG_MODEM_UBLOX_SARA=y
2. Build with `west build -b particle_boron samples/net/lwm2m_client/ -- -DOVERLAY_CONFIG=test.overlay`
3. Observe that the modem driver gets stuck at `Waiting for modem to respond`
4. Change the init priority for `board_particle_boron_init` to 41.
5. Observe that the modem driver no longer gets stuck.

**Expected behavior**
I expected the board init to run before the modem serial line is used.

**Impact**
The modem driver cannot communicate with the modem because the serial buffer (level shifter) is not powered on.

**Environment**
 - OS: Debian sid
 - Toolchain: crosscompiler
 - Commit SHA: e2d1fce6ca51807811e46f29e89c7ba4cf15df46

henrikbrixandersen, 2023-04-17:
> 4\. Change the init priority for `board_particle_boron_init` to 41.

Can you please submit a PR for this?

weinholt, 2023-04-17:
> Can you please submit a PR for this?

Will do.

PR #56951 (2023-05-30): boards: particle_boron: Initialize the board before the modem driver
The board initialization must be done before the modem driver is started because the power to the modem serial line's level shifter is enabled here.

Fixes: #56883
commit b6c80d16 (2023-05-30): boards: particle_boron: Fix init of modem pins

The Boron has a power switch that controls power to a level shifter
for the modem's serial line. Previously, this switch was being enabled
after the modem driver had already started. It is now controlled using
a fixed regulator.

The board initialization was also configuring the V_INT_DETECT pin.
However, this pin is already handled as mdm-vint-gpios in the dts when
the ublox-sara-r4 driver is used. Since it is not used otherwise,
there is no need to configure it in the board init.

Fixes: #56883
closed 2023-05-30 by nashif