zephyrproject-rtos/zephyr #56884: Add support for OSV-scanner

toonst, 2023-04-15:
I want to check whether the firmware I am running contains any existing security vulnerabilities.

Google released a tool some months ago that checks for vulnerabilities based on the SBOM.
https://github.com/google/osv-scanner

Zephyr already has support for generating SBOMs:
https://docs.zephyrproject.org/latest/develop/west/zephyr-cmds.html#software-bill-of-materials-west-spdx

Running osv-scanner on the generated spdx file gives the following result:

```
❯ osv-scanner  --sbom=build/spdx/zephyr.spdx
Scanned SPDX SBOM and found 0 packages
No package sources found, --help for usage information.
```

It seems the OSV scanner tool needs a package url (PURL) to locate a specific package, and this is not provided by the Zephyr generated SPDX file.

toonst, 2023-04-15:
@dkalowsk FYI

dkalowsk, 2023-04-19:
@swinslow was the original author of the SPDX work although I do not believe he is active on the Zephyr project anymore.  

@toonst  have you done any manual munging of the zephyr.spdx file to see what and where the URLs are needed?

toonst, 2023-04-19:
There's 3 SPDX files generated:
## 1. app.spdx
contains one package:
```
...
##### Package: app-sources

PackageName: app-sources
SPDXID: SPDXRef-app-sources
PackageDownloadLocation: NOASSERTION
PackageLicenseConcluded: NOASSERTION
PackageLicenseDeclared: NOASSERTION
PackageCopyrightText: NOASSERTION
PackageLicenseInfoFromFiles: NOASSERTION
FilesAnalyzed: true
PackageVerificationCode: cc425c9a748a6cccb15821d748fe82e12606b2ab

FileName: ./src/main.c
SPDXID: SPDXRef-File-main.c
FileChecksum: SHA1: 1f103f19d6392a3d712e32be37f737bfaf4142f0
FileChecksum: SHA256: 4fa53b946e2935016d1bf8fb237ef42aa7ad24da59802e6e0c368ce96bd2506d
LicenseConcluded: NOASSERTION
LicenseInfoInFile: NONE
FileCopyrightText: NOASSERTION
```
## 2. zephyr.spdx
```
...
##### Package: zephyr-sources

PackageName: zephyr-sources
SPDXID: SPDXRef-zephyr-sources
PackageDownloadLocation: NOASSERTION
PackageLicenseConcluded: Apache-2.0 AND BSD-3-Clause AND BSD-3-Clause-UC AND MIT
PackageLicenseDeclared: NOASSERTION
PackageCopyrightText: NOASSERTION
PackageLicenseInfoFromFiles: Apache-2.0
PackageLicenseInfoFromFiles: BSD-3-Clause
PackageLicenseInfoFromFiles: BSD-3-Clause-UC
PackageLicenseInfoFromFiles: MIT
FilesAnalyzed: true
PackageVerificationCode: 7d4ee00f611fbdac34ff0f00ee8318b458d125e5

...
Here a long list of FileName: blocks
...
```
## 3. build.spdx
Contains a long list of packages, I assume generated from Cmake targets:
```
##### Package: arch__common

PackageName: arch__common
SPDXID: SPDXRef-arch--common
PackageDownloadLocation: NOASSERTION
PackageLicenseConcluded: NOASSERTION
PackageLicenseDeclared: NOASSERTION
PackageCopyrightText: NOASSERTION
PackageLicenseInfoFromFiles: NOASSERTION
FilesAnalyzed: true
PackageVerificationCode: d3e8cc23336f3e17dece861a8300496f2b74d743

Relationship: SPDXRef-arch--common HAS_PREREQUISITE SPDXRef-isr-tables
Relationship: SPDXRef-arch--common HAS_PREREQUISITE SPDXRef-syscall-list-h-target
Relationship: SPDXRef-arch--common HAS_PREREQUISITE SPDXRef-driver-validation-h-target
Relationship: SPDXRef-arch--common HAS_PREREQUISITE SPDXRef-kobj-types-h-target
Relationship: SPDXRef-arch--common HAS_PREREQUISITE SPDXRef-zephyr-generated-headers

FileName: ./zephyr/arch/common/libarch__common.a
SPDXID: SPDXRef-File-libarch--common.a
FileChecksum: SHA1: 7da11875f46a3b7fa0db247e740e533a0ae55f0d
FileChecksum: SHA256: df58ffe726a71784351b3542550e27a27f8c83d62e6223849415d306864601e8
LicenseConcluded: NOASSERTION
LicenseInfoInFile: NONE
FileCopyrightText: NOASSERTION

Relationship: SPDXRef-File-libarch--common.a GENERATED_FROM DocumentRef-zephyr:SPDXRef-File-sw-isr-common.c
Relationship: SPDXRef-File-libarch--common.a STATIC_LINK SPDXRef-File-libisr-tables.a
...
list goes on and on
...
```
This doesn't seem to fit the OSV scanner usecase. What we probably want is a package entry in the spdx for each west module defined in the west.yml manifest.

Next to the right packages and package urls, what is also missing as far I'm aware, is a central package manager for modules used. From https://osv.dev/ you can see that each package is tied to a packaging ecosystem, which does not exist for embedded C/C++ libraries. Maybe something like https://vcpkg.io/ might be used there, but it's at least not currently supported by OSV.

The OSV scanner based on SPDX files approach seems a bit far fetched atm. 
If there's some other suggestions to get to the initial goal (I want to check whether the firmware I am running contains any existing security vulnerabilities) I'm eager to hear opinions.

ceolin, 2023-04-21:
> This doesn't seem to fit the OSV scanner usecase. What we probably want is a package entry in the spdx for each west module defined in the west.yml manifest.

Exactly! We need, at least, proper `PackageName` and `PackageVersion`  there for these tools be able to query CVEs databases. I got the same problem some time ago trying to use https://github.com/intel/cve-bin-tool.

I am afraid of the efficacy of Zephyr's SBOM to be honest, since it does not make proper distinction between components.

ceolin, 2023-04-21:
FIY, I was looking it in the context of this issue https://github.com/zephyrproject-rtos/zephyr/issues/53479

toonst, 2023-05-02:
Closing this in favor of the discussion in https://github.com/zephyrproject-rtos/zephyr/issues/53479

closed 2023-05-02 by toonst