zephyrproject-rtos/zephyr #56917: CMSIS-NN and CMSIS-DSP External Source Code

XenuIsWatching, 2023-04-17:
## Origin

https://github.com/ARM-software/CMSIS-NN
https://github.com/ARM-software/CMSIS-DSP

## Purpose

Previously CMSIS-NN and CMSIS-DSP lived inside of CMSIS_5. Since then CMSIS-NN and CMSIS-DSP have for been moved in to it's own realm 🧙 
Where CMSIS-NN used to be --> https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/NN
Where CMSIS-DSP used to be --> https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/DSP

It is also expected that the current zephyr fork for CMSIS (https://github.com/zephyrproject-rtos/cmsis) will have cmsis-nn and cmsis-dsp removed from them

## Mode of integration

These are to be 2 modules. One for `cmsis-dsp` and another for `cmsis-nn`.

## Maintainership

@stephanosio as he current maintainer? @XenuIsWatching (myself) will be working on the integration of it and I can be added as a collaborator

## Pull Request

// TODO

## Description

CMSIS-NN and CMSIS-DSP have moved in to their own realm outside of CMSIS_5. This would allow updates to more closely follow the upstream repositories. Previously, these have already been in https://github.com/zephyrproject-rtos/cmsis but they have been under old versions.
This also solves the issue as the latest versions have bug fixes (which include a major one that _somehow_ crashes 💥 GCC 12.2 during compiliation). There are also updates to ARM MVE for those working on processors that support that)

## Dependencies

An update to CMSIS-NN would also require the tflite-micro (https://github.com/zephyrproject-rtos/tflite-micro) to be update due to some headers moving around with cmsis-nn moving to it's own realm. Some small smake updates would also be needed to have it point to the new cmsis-nn module.

## Revision

// TODO

## License

### CMSIS-DSP
 
CMSIS-DSP has two licenses under it.

Apache v2 https://github.com/ARM-software/CMSIS-DSP/blob/main/LICENSE.txt

Compute Library is used for processors that have NEON support

MIT License https://github.com/ARM-software/CMSIS-DSP/blob/main/ComputeLibrary/LICENSE.txt

### CMSIS-NN

Apache v2 https://github.com/ARM-software/CMSIS-NN/blob/main/LICENSE.txt

stephanosio, 2023-04-17:
Note that the TSC has already approved creating a Zephyr module for CMSIS-DSP -- see https://github.com/zephyrproject-rtos/zephyr/issues/50928#issuecomment-1511153270.

So, we only need to discuss/decide on whether we should do the same for CMSIS-NN (I see no reason not to).

ifyall, 2023-04-26:
Voted on in TSC 26 April 2023.

stephanosio, 2023-05-02:
> Voted on in TSC 26 April 2023.

To clarify, the motion to create a module for CMSIS-NN passed. Removing the TSC label.

stephanosio, 2023-05-02:
The Zephyr CMSIS-NN module repository has been created based on the content of the upstream [CMSIS-NN repository](https://github.com/ARM-software/CMSIS-NN): https://github.com/zephyrproject-rtos/cmsis-nn

closed 2023-05-02 by stephanosio