zephyrproject-rtos/zephyr #56918: Flash fails on the nucleo_wb55rg board.

2songh, 2023-04-17:
**Describe the bug**
When Bluetooth is connected with Windows(PC or Laptop),
flash_erase & flash_write functions cannot be used and zephyr OS operation is stopped.

**Environment**
 - OS: Linux ubuntu 18.04
 - Toolchain:  zephyr-sdk-0.11.4
 - FUS Stack: stm32wb5x_FUS_fw.bin (version: 1.2.0.0)
 - BLE Stack: stm32wb5x_BLE_HCILayer_fw.bin (version: 1.13.0.5)

**Windows and Bluetooth Environment**
- Widows 10
- Intel(R) Wireless Bluetooth Driver 
  and Realtek Bluetooth 5.0 Adapter (Bluetooth Dongle).
  Both cause the same problem.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=nucleo_wb55rg ..
3. west flash
4. bluetooth connect with Windows
5. See error

**Impact**
What impact does this issue have on your progress: Problems with adding product features.

**Expected behavior**
There is no problem when bluetooth connected with mobile products.
(ex. iphone(ios), Galaxy(Android))
The problem occurs only when connected to Windows.
I want to use the flash function without problems even when connected bluetooth to Windows.

**Logs and console output**
[image]

Please also mention any information which could help others to understand
the problem you're facing:
[image]
 - Is that red part causing the problem?
 - Is the flash function unavailable while Bluetooth is connected with Windows?

Thank you for reading.

2songh, 2023-04-18:
I noticed one difference.
It is a Bluetooth device address type.

ios -> random
Windows -> public

Could this difference also be the cause?

2songh, 2023-04-19:
Could prj.conf file be related?
I wonder if there are any Config settings related to Windows.

Any further advice please.

erwango, 2023-04-21:
@2songh the difference of behavior might be due to Windows not accepting ble connection with long enough window interval for the erase to happen. 
Unfortunately, there is nothing that can be done on stm32wb55rg side to overcome this limitation.

2songh, 2023-04-27:
Thank you for answer.
I will admit that there is no workaround. 
Shall I close the issue?

erwango, 2023-05-02:
> Shall I close the issue?

Yes, txs

closed 2023-05-02 by 2songh