zephyrproject-rtos/zephyr #56919: Add ACPI bus driver support using acpica external open source project

najumon1980, 2023-04-17:
ACPI is an hardware abstraction interface for abstract platform HW complexity with the help of platform FW (eg: BIOS) and provide a unified interface to OS so that OS can be easily ported with minimum modification to different platforms (which are ACPI complaint). 

ACPI will significantly reduce Zephyr porting effort on newer platforms since it will identify supported devices and cpu/platform/device resource and automatically configure them dynamically during system boot. Please note that most of current DTS information can be removed since acpi will provide those information on run time. The device and platform configuration for these platforms normally provided via BIOS. Hence user can go and modify them via BIOS (or even BIOS menu in some cases) without re-build of Zephyr image.

Currently Zephyr does not have a full fledge acpi support and this limit its capability to supports platforms which are acpi complaint. 

> **Note**
Please note this already discussed in Arch WG few week before as part of https://github.com/zephyrproject-rtos/zephyr/pull/56347 PR review.

The solution here proposed using an open source project namely acpica and using it as external module for realize the acpi bus driver functionality. 

Following block diagram depict overall Zephyr BSP architecture based on ACPI sub system:

Please note that "zephyr acpi bus interface" will provide a generic interface to other Zephyr sub systems/drivers and also provide some glue logics for abstract under lying acpica instantiate and complexity. The acpica is an OS-independent implementation of ACPI and hence OS Service layer will provide necessary logic to port acpica into Zephyr. The OS Service layer for Zephyr already upstreamed into acpica opensource tree. 

[image]

Following are some of the ACPI bus driver functionalities:
- Enumerate PCI bus/bridge and acpi device such as WDT, RTC, GPIO, APIC, DMA etc.
- Enumerate device resources such as MMIO or IO address, IRQ details
- Retrieve IO configurations such as clk freq, boud rate, polarity etc. for SOC device like I2C, SPI, UART, SMBus etc.
- Bus and other config details for on board sensors
- Retrieve platform configuration such as PCI legacy interrupt routing, APIC
- Retrieve CPU configurations/Information
- Device/platform power management and configuration

**Pull Request Details**

The https://github.com/zephyrproject-rtos/zephyr/pull/56347 PR created for enable the acpi support for Zephyr using acpica open source project.

Following commits provided:

add acpi bus driver for zephyr: https://github.com/zephyrproject-rtos/zephyr/pull/56347/commits/1040137ecdc3bcee0e44e3c5c61b36dee93064ef

build: west: add acpica module into zephyr project: https://github.com/zephyrproject-rtos/zephyr/pull/56347/commits/4bacb86267e9cfb41018374f9ecb7398b1906785

board: x86: rpl_crb: added acpi support for raptor lake crb: https://github.com/zephyrproject-rtos/zephyr/pull/56347/commits/d2dd350094ba5c01c90595ccdb9780b5f1e19b23

drivers: watchdog: add acpi support for tco wdt: https://github.com/zephyrproject-rtos/zephyr/pull/56347/commits/ae9dcdd5edd5790e232924d8a98d3c50f90f53e3

drivers: pcie: add prt support for pci legacy interrupt: https://github.com/zephyrproject-rtos/zephyr/pull/56347/commits/952885a16adb9d5c4f17572968643b98325c57dc

hongshui3000, 2023-04-17:
except for x86 architectures, can other architectures implement ACPI? Or what should I do to porting such as cortex m architecture soc using acpi. Seems like it's a complicated framework

najumon1980, 2023-04-17:
ARM based platform already support ACPI and you can get more details.
https://developer.arm.com/Architectures/Advanced%20Configuration%20and%20Power%20Interface

ACPI normally support/help application processor based platform such as Cortex A series etc. where more complexly in porting OS and not for MCU series such as cortex m.

PR #56347 (2023-06-30): Add ACPI support based on ACPICA library
Add ACPI driver support for Zephyr using acpica open source project. ACPI driver use to discover and configure platform hardware components, perform power management (e.g. putting unused hardware components to sleep), auto configuration (e.g. Plug and Play and hot swapping) etc. Though acpica support full-fledged acpi driver functionalities, current scope limited to device enumeration and retrieve acpi device resource such as MMIO address and IRQ details, PCI interrupt routing information etc. 

Fixes #56919
closed 2023-06-30 by jhedberg