zephyrproject-rtos/zephyr #56920: Cyclic dependency in SPI bus

benediktibk, 2023-04-17:
It is possible to have a devicetree for an SPI bus which has a cyclic depencency:

    &spi1 {
    	cs-gpios = <&gpiof 10 GPIO_ACTIVE_LOW>,			
    		   <&temp_port_expander 0 GPIO_ACTIVE_LOW>;
    	...
    
    	temp_port_expander: temp_port_expander@0 {
    		compatible = "microchip,mcp23sxx";
    		...
    	};
    
    	temp_adc1: temp_adc1@1 {
    		compatible = "ti,ads114s08";
    		...
    	};
    };

This devicetree will then fail to build during the devicetree parsing and code generation, as cyclic depencency are forbidden. But this setup is a valid hardware configuration, hence it should be possible to use it in zephyr.

During the discussion of a PR for this issue (#56726) I learned that the long term goal is to move forward with replacing the device initialization levels and priorities with a initialization order based upon the device ordinals. This is the point where this SPI setup is problematic, because for the algorithm it is unclear which device should be initialized first.

As a solution I could for instance think of moving the chip selects from SPI controller to the bus devices, but this would of course be a major change.

benediktibk, 2023-04-17:
Maybe @mbolivar-nordic and @tbursztyka could give their opinions on this matter?

benediktibk, 2023-04-18:
> I think you have bigger problems than cyclic dependencies.
> 
> It is going to be difficult to call spi_context_cs_configure_all() when initializing your SPI controller if the CS GPIO is itself a SPI device on the same bus.

You are absolutely right, during my first test on this piece of hardware I stumbled upon exactly this problem. Which makes me more certain that it will be necessary to rework where the chip selects are defined in the devicetree and initialized to be able to support such a hardware configuration.

carlescufi, 2023-04-18:
@benediktibk is this still a valid issue then, or should we close it?

benediktibk, 2023-04-18:
It is absolutely still an open issue, it just got even more complicated or cumbersome to solve.

tbursztyka, 2023-04-19:
I don't have much to say on this atm. We still haven't a solution to get rid of level/prio, so this sort of exception is not ready to be tackled.

Though it's not fully circular: 

I removed the SPI label, as it is not related to SPI really: it's all about DTS and device initialization order.

benediktibk, 2023-04-19:
I now finally have the hardware with this configuration available, and we figured out that it actually won't work anyway. Due to the chip selects being activated or deactivated there will always be a certain period when two chip selects are active.

Therefore, it is actually nothing I will have to fix or support in software.

Thanks anyway for your inputs!