zephyrproject-rtos/zephyr #56962: sample project mentioned in the doc no longer exists

botbotty, 2023-04-18:
The doc says:
https://github.com/zephyrproject-rtos/zephyr/blob/f5b7752c64eba16bd00b45d920e78f2b2888ef5b/doc/kernel/code-relocation.rst?plain=1#L134-L140

but `$ZEPHYR_BASE/samples/application_development/code_relocation` does not exist

nordicjm, 2023-04-18:
Indeed it seems that it was replaced in https://github.com/zephyrproject-rtos/zephyr/pull/47771 with tests/application_development/code_relocation

@carlocaione is it a drop-in replacement that functions the same? Should the path in the documentation just be updated or does it operate differently?

carlocaione, 2023-04-18:
> @carlocaione is it a drop-in replacement that functions the same? Should the path in the documentation just be updated or does it operate differently?

Yes, that should be the same code just in a "test form".

PR #56993 (2023-04-19): doc: change path to code relocation test
A sample was moved to tests, adapt documentation accordingly.

Fixes #56962
commit 5238544d (2023-04-19): doc: change path to code relocation test

A sample was moved to tests, adapt documentation accordingly.

Fixes #56962
closed 2023-04-19 by nashif