zephyrproject-rtos/zephyr #56987: app.overlay does not seem to work when an app contains a board directory

Laczen, 2023-04-18:
**Describe the bug**
I am trying to build an application (in folder app) that has a board folder (app/boards).

The board folder contains a board.overlay specifying the flash partitions (containing a `sfsl_partition`).

In the app folder I placed an app.overlay file that specifies the partition the code should be generated for using:
```
/ {
	chosen {
		zephyr,code-partition = &sfsl_partition;
	};
};
```

The prj.conf file containes the line:
```
CONFIG_USE_DT_CODE_PARTITION=y
```

The build system does not respect the `chosen zephyr,code-partition` and seems to generate the build for the entire flash (the Region Size reported is the complete flash).

If the `chosen {...` is placed in board.overlay the build system generates the correct binary for `zephyr,code-partition`.

**Expected behavior**

The build system should respect both the app.overlay and the board.overlay.

bdunlay, 2023-04-19:
Hi @Laczen, I discovered this behavior too this week, but I am using mcuboot like you mentioned. With zephyr 3.3.0 / mcuboot 1.1.0, the app.overlay file _**does not**_ get pulled in with mcuboot if an overlay is present in the boards/ directory. 

I think this behavior is expected based on the [device tree overlay docs](https://docs.zephyrproject.org/3.3.0/build/dts/howtos.html#set-devicetree-overlays). Before mcuboot 1.1.10, the mcuboot CMake file was the one adding app.overlay files on top of other overlay files, but they've since [removed that logic](https://github.com/mcu-tools/mcuboot/commit/82ee5d0bb9ac68c14033b102705904ee9991ee3d#diff-73248e09fee3c44341d1841b649cbb65a47808f15cc2b273020386af925a30b5) and now it follows the default behavior which is what you are seeing.

I wanted to reuse my overlay file from my app without setting a chosen code partition, so I worked around it by passing both overlays in the `DTC_OVERLAY_FILE` flag to cmake:

```
west build -d mcuboot-build -b b_l4s5i_iot01a ~/zephyrproject/bootloader/mcuboot/boot/zephyr -- \
    -DDTC_OVERLAY_FILE="~/zephyrproject/bootloader/mcuboot/boot/zephyr/boards/b_l4s5i_iot01a.overlay;~/zephyrproject/bootloader/mcuboot/boot/zephyr/app.overlay"
```

nordicjm, 2023-04-19:
> The build system should respect both the app.overlay and the board.overlay.

No, it should not, this is documented here: https://docs.zephyrproject.org/latest/build/dts/howtos.html#set-devicetree-overlays

Closing since not a bug

closed 2023-04-19 by nordicjm

Laczen, 2023-04-19:
Hi @bdunlay, it was your question on discord that triggered me to test this and report it. I agree that it is expected behavior after reading the documentation again.

@nordicjm, maybe it is expected behavior, but it might require some attention:
* in mcuboot there are several overlay files that result in mcuboot being compiled for complete flash instead of boot_partition.
* when using retention memory areas or data retention areas a similar construct is used to assign the region to `zephyr,boot_mode`, this will also not work (the zephyr, boot_mode assignment will need to be made in the boards directory).

nordicjm, 2023-04-19:
There is only 1 possible board which has a dedicated .overlay file and that is this https://github.com/mcu-tools/mcuboot/blob/main/boot/zephyr/boards/intel_s1000_crb.overlay I cannot find that board in the zephyr tree so either it's an out of tree board or is not a specific board name. The other overlay files are not used because they are not the name of a board, they have partial names of boards in because they are designed for use with `-DDTC_OVERLAY_CONFIG` in which case they would also need to provide `app.overlay` as part of this argument.

And yes, this is already done: https://github.com/zephyrproject-rtos/zephyr/pull/55125/commits/99967a90dd4d4c4b3b2c02221a9d6f580870c168#diff-063371bd9ea8797dd286faeaf505fa13f82fe4da5ba2aed32e3814b1472014f2R6