zephyrproject-rtos/zephyr #56989: Bluetooth: Host: Invalid disconnect reason

hermabe, 2023-04-18:
**Describe the bug**
The Bluetooth Core Specification Version 5.4 | Vol 4, Part E, Section 7.1.6 "Disconnect command" specifies these allowed reasons:
> Authentication Failure error code (0x05),
Other End Terminated Connection error codes (0x13 to 0x15),
Unsupported Remote Feature error code (0x1A),
Pairing with Unit Key Not Supported error code (0x29) and
Unacceptable Connection Parameters error code (0x3B)

However, the zephyr host uses a different reason (`0x1f`) in at least on case:
https://github.com/zephyrproject-rtos/zephyr/blob/ed3f8761a725b9c261a0605e98e9484be43ea71e/subsys/bluetooth/host/hci_core.c#L1255

The zephyr host should use one of the allowed disconnect reasons instead.

**To Reproduce**
Inspect the zephyr source code and the Bluetooth core specification.

**Expected behavior**
The zephyr host to be compliant with the Bluetooth core specification.

**Impact**
The zephyr host is violating the Bluetooth specification and a compliant controller will reject the disconnection command.

alwa-nordic, 2023-04-18:
The PR introducing this behavior: https://github.com/zephyrproject-rtos/zephyr/pull/26465

hermabe, 2023-04-18:
> I think we can keep the priority low. I suspect this case of `bt_hci_disconnect(handle, BT_HCI_ERR_UNSPECIFIED);` is for when the host or controller misbehaves (as evidenced by the use of BT_ERR).

Agree. I wanted to use this error in my PAwR connection PR in case the host is unable to allocate the connection context. I will just use another one.

> Will the remote side really reject (i.e. ignore) the command? I expect it would disconnect if it considers the protocol violated, which is fine here.

The softdevice controller rejcets the HCI disconnect command with error 0x12 (Invalid parameters).

carlescufi, 2023-04-20:
Agreement: Use REMOTE USER TERMINATED CONNECTION (0x13) instead.

PR #57365 (2023-04-28): Bluetooth: Host: Use a valid disconnect reason
"Unspecified Error" is not a disconnect reason allowed by spec.

Fixes #56989
commit 58936061 (2023-04-28): Bluetooth: Host: Use a valid disconnect reason

"Unspecified Error" is not a disconnect reason allowed by spec.

Fixes #56989
closed 2023-04-28 by carlescufi