zephyrproject-rtos/zephyr #56995: [mbedtls/tf-m] PSA API conflicts

tpennors, 2023-04-18:
**Describe the bug**
When migrating to Zephyr 3.3.0 which uses mbedtls 3.2.1 and TF-m 1.7.0, we are having PSA API conflicts during build.
After investigation we saw in release note that there is a "PSA API conflicts that can not be immediately resolved between Zephyr's
instance of MbedTLS in the NS environment, and the TF-M PSA APIs" in following commit :
https://github.com/zephyrproject-rtos/zephyr/commit/c63fb217601fb

Does anyone have any roadmap or indications to solve this issue ?

**Expected behavior**
Working as with Zephyr 3.2.0

**Impact**
Currently unable to migrate to Zephyr 3.3.0 but we need it to use TF-m 1.7.0 with new psa_fwu API

**Logs and console output**
Here is an output of our build: 
```
In file included from /home/e.basle/projects/lpp21/lpp4_dl/lpp4_zephyr_3_3_0_integration/platform/zephyr/../modules/crypto/mbedtls/include/mbedtls/ssl_ciphersuites.h:28,
                 from /home/e.basle/projects/lpp21/lpp4_dl/lpp4_zephyr_3_3_0_integration/platform/zephyr/../modules/crypto/mbedtls/include/mbedtls/ssl.h:32,
                 from /home/e.basle/projects/lpp21/lpp4_dl/lpp4_zephyr_3_3_0_integration/framework/base/subsys/common/cert_mngr/lib_cert_mngr/tests_u/src/main_ztest.c:21:
/home/e.basle/projects/lpp21/lpp4_dl/lpp4_zephyr_3_3_0_integration/platform/zephyr/../modules/crypto/mbedtls/include/mbedtls/pk.h:328:36: error: unknown type name 'mbedtls_svc_key_id_t'
  328 |                              const mbedtls_svc_key_id_t key );
      |                                    ^~~~~~~~~~~~~~~~~~~~
/home/e.basle/projects/lpp21/lpp4_dl/lpp4_zephyr_3_3_0_integration/platform/zephyr/../modules/crypto/mbedtls/include/mbedtls/pk.h:994:32: error: unknown type name 'mbedtls_svc_key_id_t'
  994 |                                mbedtls_svc_key_id_t *key,
      |                                ^~~~~~~~~~~~~~~~~~~~
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Zephyr tag 3.3.0

microbuilder, 2023-04-19:
There are some changes in the works to change the way that MbedTLS and TF-M are integrated into Zephyr, to build MbedTLS with three distinct libraries (Crypto, X.509, TLS, with Crypto linked against TF-M on the secure side), but this require changes to MbedTLS and TF-M that will be part of the next releases of both projects. As such, this is likely to only be resolved when we migrate to TF-M 1.8.0 and MbedTLS 3.4.0, with a goal to complete the work before the Zephyr feature freeze late May.

There is also no guarantee we will add a sample for the new FWU service. This may be something you can contribute if you are looking at using that service in Zephyr(?).

microbuilder, 2023-04-19:
@joerchan @rajkan01 FYI

closed 2023-05-26 by nashif