zephyrproject-rtos/zephyr #56999: gpio: some drivers still use GPIO_PORT_PIN_MASK_FROM_NGPIOS macro the old way

avolkov-1221, 2023-04-18:
GPIO drivers **drivers/gpio/gpio_ite_it8xxx2.c** and  **drivers/gpio/gpio_kscan_ite_it8xxx2.c** still use GPIO_PORT_PIN_MASK_FROM_NGPIOS macro the old way, as shown below:

        .port_pin_mask = GPIO_PORT_PIN_MASK_FROM_NGPIOS(              \
                           DT_INST_PROP(inst, ngpios))

while this usage has been deprecated and replaced by the newer style (in v2.3.0):

        .port_pin_mask =  GPIO_PORT_PIN_MASK_FROM_DT_INST(inst)             \

Or, even better, in the latest way:

        .port_pin_mask =  GPIO_PORT_PIN_MASK_FROM_DT_INST(inst)             \

As result driver are ignoring handling of the 'gpio-reserved-ranges' DT property value.

The patch is attached.

[gpio_ite_fix.txt](https://github.com/zephyrproject-rtos/zephyr/files/11264896/gpio_ite_fix.txt)

nordicjm, 2023-04-19:
> The patch is attached.

We do not accept patches submitted like this, can you submit a PR with the fix?

avolkov-1221, 2023-04-19:
Hello Jamie,
Actually the diff is a part of PR  https://github.com/zephyrproject-rtos/zephyr/pull/56295 , it was sent as an example of the problem solution.

avolkov-1221, 2023-04-19:
BTW should I create the issues for another drivers affected the same problem?

nordicjm, 2023-04-19:
> BTW should I create the issues for another drivers affected the same problem?

If there's other drivers with the same problem, yes

henrikbrixandersen, 2023-04-20:
Please beware that `GPIO_PORT_PIN_MASK_FROM_NGPIOS()` is not deprecated in any way, so it is perfectly okay for a driver to use it.

avolkov-1221, 2023-04-20:
Ok, I've 'un-deprecated' it. Could you review the last commit?

PR #56295 (2023-07-13): drivers: gpio: add "gpio_reserved_ranges" property support

Added missing support of DT gpio's "gpio_reserved_ranges" property. The array of pin range pairs as shown below now works. The implementation constraint is inherited from common DT stuff: a maximum of 64 pairs can be used, i.e. no more than 128-bit gpio ports are supported.

Usage example:
The reserved mask , defined in the device tree as:
gpio0: gpio0 {
....
ngpios=<32>;
gpio-reserved-ranges = <0 8>, <9 5>, <15 17>;
....
};

and used in driver configuration as:
{
...
.gpio_reserved_ranges = GPIO_RESERVED_PIN_MASK_FROM_DT_NODE(
DT_NOD [... cut]
closed 2023-07-13 by carlescufi