zephyrproject-rtos/zephyr #57001: udp sockets do not respect port bind

bdunlay, 2023-04-18:
**Describe the bug**

UDP sockets do not respect port bindings. If you bind a socket to a specific port, or bind to 0, the port is ignored and datagrams appear to be sent from the destination port.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? B-L4S5I-IOT01A with offload sockets disabled
 - What have you tried to diagnose or workaround this issue? 
 
* Bind to 0, sendto
* Bind to 32100, sendto
* Bind to 0, connect, send
* Bind to 32100, send
* Created example on mac os using posix api to demonstrate expected behavior
  
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit? No

**To Reproduce**
Steps to reproduce the behavior:

See attached file for reference implementation.

**Expected behavior**

I expect that when I bind a socket to port 0, a random port is assigned and all sends using that socket are sent from that port. I also expect that when I bind a socket to a specific port, like 32100, all sends using that socket are sent from that port.

What wireshark has shown me is that packets sent from that socket have src port == dst port.

**Impact**
This is a major problem because I am not able to reestablish communications after reboots. I don't have any control over the source port. DTLS will ignore new handshake requests until server-side session expirations trigger. With a new port, I am able to initiate new handshakes.

**Logs and console output**

See attached for wireshark screencap.

**Environment (please complete the following information):**
 - OS: Mac OS
 - Toolchain: Zephyr SDK 0.15.2
 - Version: v3.3.0

bdunlay, 2023-04-18:
[udp.zip](https://github.com/zephyrproject-rtos/zephyr/files/11265238/udp.zip)
[image]

bdunlay, 2023-04-18:
This posix example demonstrates that providing the port in the sockaddr_in at bind associates the socket with that port. Wireshark confirms.

```#include <stdio.h>
#include <string.h>
#include <sys/socket.h>
#include <arpa/inet.h>

static char buf[] = "Hello";

int main() {
  int ret;

  struct sockaddr_in my_addr = {
    .sin_family = AF_INET,
    .sin_addr.s_addr = htonl(INADDR_ANY),
    .sin_port = htons(32100)
  };

  struct sockaddr_in server_addr = {
    .sin_family = AF_INET,
    .sin_port = htons(5684)
  };
  inet_pton(AF_INET, "192.168.1.92", &server_addr.sin_addr);

  int sock = socket(my_addr.sin_family, SOCK_DGRAM, IPPROTO_UDP);

  ret = bind(sock, &my_addr, sizeof(struct sockaddr_in));
  if (ret < 0) { printf("Bind: %d", ret);}

  if (connect(sock, (struct sockaddr *)&server_addr, sizeof(server_addr)) < 0) {
    printf("Failed to set socket peer address %s", "192.168.1.192:5684");
  }

  ret = send(sock, buf, sizeof(buf), 0);
  if (ret < 0) { printf("Sendto: %d", ret);}

  return 0;
}
```

[image]

bdunlay, 2023-04-18:
This looks like it is probably a bug with eswifi. The network stack is offloaded to the eswifi chip, and even if I try to disable network offloading, it is selected anyway.

bdunlay, 2023-04-18:
@loicpoulain I pulled in your latest change to bind (9afa73236b1a3b6937eb0e7ff04a326bb8f1b0f6), which starts a UDP server with `P5`. But then the next call, `sendto()`, starts a UDP client with `P6`. So the port bound to the udp server socket is not used.

bdunlay, 2023-04-18:
[bind-sendto-udp.zip](https://github.com/zephyrproject-rtos/zephyr/files/11266719/bind-sendto-udp.zip)

bdunlay, 2023-04-19:
I reached out to Inventek, and the response from them is that it is not possible to set the UDP client port.

I think this is a limitation that should be documented for this driver, or at least an error/log message should be returned if a bind/sendto occurs on the same port (c8f48681ed8161607b634d924061e9a946891673 will tear down the server silently, which is confusing if you expect bind/sendto to work).

carlescufi, 2023-04-21:
> I reached out to Inventek, and the response from them is that it is not possible to set the UDP client local port.
> 
> I think this is a limitation that should be documented for this driver, or at least an error/log message should be returned if a bind/sendto occurs on the same port. sendto will tear down the server ([c8f4868](https://github.com/zephyrproject-rtos/zephyr/commit/c8f48681ed8161607b634d924061e9a946891673)) started by bind ([54e8d10](https://github.com/zephyrproject-rtos/zephyr/commit/54e8d10e61a50ac9d89e6138a8a53105c4f3445a)) silently, which is confusing if you expect bind/sendto to work.

Thanks for investigating this. Would you mind opening a Pull Request that does what you suggest? (document the limitation and issue an error message)

closed 2023-07-09 by github-actions[bot]