zephyrproject-rtos/zephyr #57002: can: incorrect RTR flag handling in incoming frames

avolkov-1221, 2023-04-18:
Hello @henrikbrixandersen, @alexanderwachter, @martinjaeger 

I've just discovered that almost all CAN bus drivers assume that an incoming CAN frame with the RTR flag has a valid payload and copies data from FIFOs/registers up onto the stack.

Unfortunately this behavior is incorrect, since RTR mean that some data should be transferred FROM the node, not TO.
So trying to copy the CAN's frame data is wasting of resources in the best case, or is a security issue in the worst ("sensitive data can be copied to the wrong client").

Regards
Andrey VOLKOV
Linux Socketcan co-founder

avolkov-1221, 2023-04-18:
Hi @henrikbrixandersen,

I also checked the sending routines a little harder, and the situation there is also "not so good": almost all drivers are trying to send data while  RTR is active.

Unfortunately this can cause an almost guaranteed panic, since the actual payload length is 0 (so possible "read out of the buffer boundary" case). Luckily for now, almost all current drivers pre-allocate packets with a maximum payload length, so you (I mean the team) have not seen this problem before and why it's probably not so critical. Meanwhile this is not the case when the device acts as a CAN router, then both incoming and outgoing traffic will contain RTR packets with zero payload in the Zephyr's CAN stack. Those, we will have a classic ping of death.

I've seen such situation before on Linux when someone from Freescale broke my MSCAN driver in their BSP by "optimizing" the RTR flag check.

henrikbrixandersen, 2023-04-24:
Thank you for reporting these findings.

> I've just discovered that almost all CAN bus drivers assume that an incoming CAN frame with the RTR flag has a valid payload and copies data from FIFOs/registers up onto the stack.
> 
> Unfortunately this behavior is incorrect, since RTR mean that some data should be transferred FROM the node, not TO.

Right.

> Thus, trying to copy CAN's frame data in the case of RTR is nothing but a waste of resources at best, or a security problem at worst ("disclose sensitive data"), since the payload from some old frame will be copied almost everywhere and/or machine check can be raised on some odd hardware.

Zephyr is not a "multi-user" system, so the security problem is more theoretical than practical. However, we should not read potentially uninitialized data from the CAN controller MB/FIFO.

> I also checked the sending routines a little harder, and the situation there is also "not so good": almost all drivers are trying to send data while RTR is active.

Right.

> Unfortunately this can cause an almost guaranteed panic, since the actual payload length is 0 (so possible "read out of the buffer boundary" case). Luckily for now, almost all current drivers [...]

In Zephyr `struct can_frame` is always allocated with full data size, so reading the data buffer, even for RTR frames, will not lead to a panic. However, it might result in writing uninitialized data to the CAN controller MB/FIFO, which should not happen.

I will look into providing a fix.

avolkov-1221, 2023-04-25:
Under the "CAN router" I mean the following usage scenario:
 1. the chip with Zephir is receiving raw CAN frames from any other interface but CAN, as raw packets stream;
 2. the recv thread is calling the can_send for above stream without any intermediate copy;
 3. error. Out of bound reading for RTR packet.
 
BTW in theory the security problem could arise when working with the ARM sec/normal world modes frames handling, and not in the case of a "multi-user".

avolkov-1221, 2023-08-29:
FYI @grantramsay can_mcan.c is affected too.

henrikbrixandersen, 2023-08-29:
> FYI @grantramsay can_mcan.c is affected too.

Yes, I am working on a fix.

avolkov-1221, 2023-08-29:
> > FYI @grantramsay can_mcan.c is affected too.
> 
> Yes, I am working on a fix.

It was mainly an alert for @grantramsay :). He had just worked on MCAN driver and probably still has access to the hardware. Therefore, he can probably fix and check RTR. Moreover, for MCAN driver, this error is not so low priority, but looks like a "ping of death": data fetching from Rx FIFO/pointers updating is based on the wrong DLC, so content of all following frames will be unpredictable.

henrikbrixandersen, 2023-08-30:
> It was mainly an alert for @grantramsay :). He had just worked on MCAN driver and probably still has access to the hardware. Therefore, he can probably fix and check RTR.

No worries. I have access to a handful or more different boards from different vendors all implementing CAN controllers based on the Bosch M_CAN IP, allowing me to verify the changes across a wide range of different M_CAN IPs.

> Moreover, for MCAN driver, this error is not so low priority, but looks like a "ping of death": data fetching from Rx FIFO/pointers updating is based on the wrong DLC, so content of all following frames will be unpredictable.

That is not how the Rx FIFOs in the Bosch M_CAN work. FIFO elements are of a fixed size. Please avoid speculating the impact of issues like this as it just adds uncertainty and doubt.

This is purely an issue of reading and writing potentially unitialized memory regions.

PR #63802 (2023-10-13): drivers: can: only copy frame data for non-RTR frames
Only copy frame data for non-RTR frames as RTR frames do not carry any data.

Fixes: #57002
commit 27323a8a (2023-10-13): drivers: can: mcan: only copy frame data for non-RTR frames

Only copy frame data for non-RTR frames as RTR frames do not carry any
data.

Fixes: #57002
commit 443847f0 (2023-10-13): drivers: can: nxp: s32: only copy frame data for non-RTR frames

Only copy frame data for non-RTR frames as RTR frames do not carry any
data.

Fixes: #57002
commit 4d321448 (2023-10-13): drivers: can: mcp2515: only copy frame data for non-RTR frames

Only copy frame data for non-RTR frames as RTR frames do not carry any
data. Limit the amount of data copied to the actual DLC.

Fixes: #57002
closed 2023-10-13 by jhedberg