zephyrproject-rtos/zephyr #57020: tests: subsys: fs: littlefs: filesystem.littlefs.default and filesystem.littlefs.custom fails

nordic-piks, 2023-04-19:
**Describe the bug**
The tests/subsys/fs/littlefs/filesystem.littlefs.custom and  tests/subsys/fs/littlefs/filesystem.littlefs.default fails.

Observed for
- nrf52840dk_nrf52840

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf52840dk_nrf52840 connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/subsys/fs/littlefs -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
[2023-04-19T01:25:59.441Z] [0mERROR   - *** Booting Zephyr OS build zephyr-v3.3.0-2791-g4bbd91a9083a ***[0m
[2023-04-19T01:25:59.441Z] [0mRunning TESTSUITE littlefs[0m
[2023-04-19T01:25:59.441Z] [0m===================================================================[0m
[2023-04-19T01:25:59.441Z] [0mSTART - test_fs_mkfs_custom[0m
[2023-04-19T01:25:59.441Z] [0mClean /sml[0m
[2023-04-19T01:25:59.441Z] [0mErasing 65536 (0x10000) bytes[0m
[2023-04-19T01:25:59.441Z] [0mError wiping flash area 4 [-5][0m
[2023-04-19T01:25:59.441Z] [0mAssertion failed at ../../../../../../../../../zephyr/zephyr/tests/subsys/fs/littlefs/src/testfs_mkfs.c:28: cleanup: (testfs_lfs_wipe_partition(mp) not equal to TC_PASS)[0m
[2023-04-19T01:25:59.441Z] [0mFailed to clean partition[0m
[2023-04-19T01:25:59.441Z] [0mFAIL - test_fs_mkfs_custom in 1.123 seconds[0m
[2023-04-19T01:25:59.441Z] [0m===================================================================[0m
[2023-04-19T01:25:59.441Z] [0mSTART - test_fs_mkfs_ops_lfs[0m
[2023-04-19T01:25:59.441Z] [0mClean /sml[0m
[2023-04-19T01:25:59.441Z] [0mErasing 65536 (0x10000) bytes[0m
[2023-04-19T01:25:59.441Z] [0mError wiping flash area 4 [-5][0m
[2023-04-19T01:25:59.441Z] [0mAssertion failed at ../../../../../../../../../zephyr/zephyr/tests/subsys/fs/littlefs/src/testfs_mkfs.c:28: cleanup: (testfs_lfs_wipe_partition(mp) not equal to TC_PASS)[0m
[2023-04-19T01:25:59.441Z] [0mFailed to clean partition[0m
[2023-04-19T01:25:59.441Z] [0mFAIL - test_fs_mkfs_ops_lfs in 1.123 seconds[0m
[2023-04-19T01:25:59.441Z] [0m===================================================================[0m
[2023-04-19T01:25:59.441Z] [0mSTART - test_fs_mkfs_simple_lfs[0m
[2023-04-19T01:25:59.441Z] [0mClean /sml[0m
[2023-04-19T01:25:59.441Z] [0mErasing 65536 (0x10000) bytes[0m
[2023-04-19T01:25:59.441Z] [0mError wiping flash area 4 [-5][0m
[2023-04-19T01:25:59.441Z] [0mAssertion failed at ../../../../../../../../../zephyr/zephyr/tests/subsys/fs/littlefs/src/testfs_mkfs.c:28: cleanup: (testfs_lfs_wipe_partition(mp) not equal to TC_PASS)[0m
[2023-04-19T01:25:59.441Z] [0mFailed to clean partition[0m
[2023-04-19T01:25:59.441Z] [0mFAIL - test_fs_mkfs_simple_lfs in 1.123 seconds[0m
[2023-04-19T01:25:59.441Z] [0m===================================================================[0m
[2023-04-19T01:25:59.441Z] [0mSTART - test_fs_mount_flags[0m
[2023-04-19T01:25:59.441Z] [0mClean /sml[0m
[2023-04-19T01:25:59.441Z] [0mErasing 65536 (0x10000) bytes[0m
[2023-04-19T01:25:59.441Z] [0mError wiping flash area 4 [-5][0m
[2023-04-19T01:25:59.441Z] [0mAssertion failed at ../../../../../../../../../zephyr/zephyr/tests/subsys/fs/littlefs/src/testfs_mount_flags.c:16: cleanup: (testfs_lfs_wipe_partition(mp) not equal to TC_PASS)[0m
[2023-04-19T01:25:59.441Z] [0mFailed to clean partition[0m
[2023-04-19T01:25:59.441Z] [0mFAIL - test_fs_mount_flags in 1.123 seconds[0m
[2023-04-19T01:25:59.441Z] [0m===================================================================[0m
[2023-04-19T01:25:59.441Z] [0mSTART - test_fs_open_flags_lfs[0m
[2023-04-19T01:25:59.441Z] [0mClean /sml[0m
[2023-04-19T01:25:59.441Z] [0mErasing 65536 (0x10000) bytes[0m
[2023-04-19T01:25:59.441Z] [0mError wiping flash area 4 [-5][0m
[2023-04-19T01:25:59.441Z] [0mAssertion failed at ../../../../../../../../../zephyr/zephyr/tests/subsys/fs/littlefs/src/testfs_open_flags.c:35: cleanup: (testfs_lfs_wipe_partition(mp) not equal to TC_PASS)[0m
[2023-04-19T01:25:59.441Z] [0mFailed to clean partition[0m
[2023-04-19T01:25:59.441Z] [0mFAIL - test_fs_open_flags_lfs in 1.123 seconds[0m
[2023-04-19T01:25:59.441Z] [0m===================================================================[0m
[2023-04-19T01:25:59.441Z] [0mSTART - test_lfs_basic[0m
[2023-04-19T01:25:59.441Z] [0mclearing partition /sml[0m
[2023-04-19T01:25:59.441Z] [0mErasing 65536 (0x10000) bytes[0m
[2023-04-19T01:25:59.441Z] [0mError wiping flash area 4 [-5][0m
[2023-04-19T01:25:59.441Z] [0mAssertion failed at ../../../../../../../../../zephyr/zephyr/tests/subsys/fs/littlefs/src/test_lfs_basic.c:44: clear_partition: (testfs_lfs_wipe_partition(mp) not equal to TC_PASS)[0m
[2023-04-19T01:25:59.441Z] [0mfailed to wipe partition[0m
[... 38 lines cut ...]
[2023-04-19T01:25:59.441Z] [0mSUITE FAIL -  42.86% [littlefs]: pass = 6, fail = 8, skip = 0, total = 14 duration = 9.098 seconds[0m
[2023-04-19T01:25:59.441Z] [0m- FAIL - [littlefs.test_fs_mkfs_custom] duration = 1.123 seconds[0m
[2023-04-19T01:25:59.441Z] [0m- FAIL - [littlefs.test_fs_mkfs_ops_lfs] duration = 1.123 seconds[0m
[2023-04-19T01:25:59.441Z] [0m- FAIL - [littlefs.test_fs_mkfs_simple_lfs] duration = 1.123 seconds[0m
[2023-04-19T01:25:59.441Z] [0m- FAIL - [littlefs.test_fs_mount_flags] duration = 1.123 seconds[0m
[2023-04-19T01:25:59.441Z] [0m- FAIL - [littlefs.test_fs_open_flags_lfs] duration = 1.123 seconds[0m
[2023-04-19T01:25:59.441Z] [0m- FAIL - [littlefs.test_lfs_basic] duration = 1.125 seconds[0m
[2023-04-19T01:25:59.441Z] [0m- FAIL - [littlefs.test_lfs_dirops] duration = 1.124 seconds[0m
[2023-04-19T01:25:59.441Z] [0m- FAIL - [littlefs.test_lfs_perf] duration = 1.228 seconds[0m
[2023-04-19T01:25:59.441Z] [0m- PASS - [littlefs.test_util_path_extend] duration = 0.001 seconds[0m
[2023-04-19T01:25:59.441Z] [0m- PASS - [littlefs.test_util_path_extend_overrun] duration = 0.001 seconds[0m
[2023-04-19T01:25:59.441Z] [0m- PASS - [littlefs.test_util_path_extend_up] duration = 0.001 seconds[0m
[2023-04-19T01:25:59.441Z] [0m- PASS - [littlefs.test_util_path_init_base] duration = 0.001 seconds[0m
[2023-04-19T01:25:59.441Z] [0m- PASS - [littlefs.test_util_path_init_extended] duration = 0.001 seconds[0m
[2023-04-19T01:25:59.441Z] [0m- PASS - [littlefs.test_util_path_init_overrun] duration = 0.001 seconds[0m
[2023-04-19T01:25:59.441Z] [0m------ TESTSUITE SUMMARY END ------[0m
[2023-04-19T01:25:59.441Z] [0m===================================================================[0m
[2023-04-19T01:25:59.441Z] [0mRunID: f000198d053bba6ec0a8c1fe95c8ebf7[0m
[2023-04-19T01:25:59.441Z] [0mPROJECT EXECUTION FAILED[0m
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.15.2
 - Commit SHA or Version used: zephyr-v3.3.0-2791-g4bbd91a9083a

de-nordic, 2023-06-05:
I am still investigating the issue, but it seems that the problem is not with LittleFS, but rather with resetting back the write enable bit after erase returns.
The function `qspi_nor_write_protections_set(... , true)`, issued after erase fails on timeout, when waiting for command to complete.

nordic-piks, 2023-09-04:
Still occurs, please remove stale label.
@jhedberg @carlescufi @de-nordic

nordic-piks, 2023-11-02:
Currently cannot reproduce.

closed 2023-11-02 by nordic-piks