zephyrproject-rtos/zephyr #57022: LE Audio: CAP Initiator unicast sample

Thalley, 2023-04-19:
**Is your enhancement proposal related to a problem? Please describe.**
We do not have a sample that uses the CAP Initiator unicast sample.

**Describe the solution you'd like**
Create a sample application that uses the CAP Initiator.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

kartben, 2023-08-21:
TMAP Central and TMAP BMS seem to use CAP initiator unicast, is that good enough to call victory?

Thalley, 2023-08-21:
> TMAP Central and TMAP BMS seem to use CAP initiator unicast, is that good enough to call victory?

I think that's a fair argument. Let's close this, and reopen if someone requests this again

closed 2023-08-21 by Thalley