zephyrproject-rtos/zephyr #57041: Building on STM32 boards fails due to Kconfig issue after #56350

semihalf-duda-patryk, 2023-04-19:
**Describe the bug**
After #56350 building Zephyr fails with the following
```
warning: HAS_SEGGER_RTT (defined at modules/segger/Kconfig:7) has direct dependencies 0 with value n, but is currently being y-selected by the following symbols:
 - SOC_FAMILY_STM32 (defined at soc/arm/st_stm32/Kconfig:6), with value y, direct dependencies y (value: y)
```

**To Reproduce**
`west build --pristine -b nucleo_f412zg samples/hello_world`

**Expected behavior**
Hello world sample builds successfully for `nucleo_f412zg`

**Impact**
I think that all STM32 boards are affected, so it's serious.

**Logs and console output**
N/A

**Environment (please complete the following information):**
N/A

**Additional context**
If I revert https://github.com/zephyrproject-rtos/zephyr/commit/1cfc85bab89deffea303e115f7d9d8a81a7b8cb0 It will work.

henrikbrixandersen, 2023-04-19:
Did you run `west update`?

semihalf-duda-patryk, 2023-04-19:
> Did you run `west update`?

I tried this and it works now. Could you tell my why it fixed the problem?

henrikbrixandersen, 2023-04-19:
> I tried this and it works now. Could you tell my why it fixed the problem?

Your segger module was outdated.

semihalf-duda-patryk, 2023-04-19:
Ok. Thanks!