zephyrproject-rtos/zephyr #57043: Unexpected thread context switch in BSIM_NRFXX platform

alxelax, 2023-04-19:
**Describe the bug**
There is a BLE Mesh function that sends mesh packet. Data should be encrypted both transport and network with authentication on each layer where nonce includes the sequence number. Delayed work from system queue interrupts the process in between app encryption and network encryption. Handler of this work sends another packet (with sequence number incrimination). Then the code goes back to the previous procedure and continue encryption in network (but with already incremented sequence). Finally, this packet cannot pass authentication on recipient, that is correct since transport and network used different nonce for authentication (what is the wrong).

BLE Mesh has been designed to work in cooperative environment and doesn't expect context switch somewhere in between functionality. 

Further investigation showed that the system work start execution inside this call: https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/ecb.c#L39
Seems `k_busy_wait` function causes switching thread context if there is pended system work.

Explanation of the attached log of wrong behavior:
1. Test runs mesh packet sending.
2. Mesh stack goes into encryption data on transport level (top red area).
3. Encryption functionality goes into ble controller until `k_busy_wait` function (first green area)
4. Context is switched and sending another BLE Mesh packet is triggered with encryption on network layer (yellow area). This is sending of Friend Poll that is triggered by `k_work_delayable` that was planned earlier.
5. After finishing step 4 the context goes back to controller encryption (makes 2 runs in loop) (second green are). Completes encryption.
6. Goes further to the network layer to start encryption there (bottom red area). 

**Expected behavior**
BLE Mesh thread is not preempted by system queue work. BLE Mesh performs partially in the system works and partially in thread. Mixing of contexts causes the competition for variables (like security materials in this use case). 

Platform: nrf5_bsim 
Workaround: Disable controller encryption\decryption. Use Host based on tinycrypt enc\dec instead.

**To Reproduce**
Steps to reproduce the behavior:
1. checkout main 326eb860b8
7. apply attached patch (need to modify bsim mesh test scenario and add logging into controller encryption)
8. `bash compile.sh` in `zephyr/tests/bsim/bluetooth/mesh`
9. `bash msg_mesh.sh` in `zephyr/tests/bsim/bluetooth/mesh/tests_scripts/friendship`
10. See logs (attached log with explanation of wrong behavior)

**Impact**
BLE Mesh can be broken in any new PR(related to Mesh) if system work fires during encryption\decryption.
Issue is not related to mesh changes, just makes coincidence of system queue work and encryption. In general, bsim tests for mesh are not trustable with such behavior, since Mesh has been designed for cooperative behavior.

**Logs and console output**
Log has been attached as picture:
[image]

Patch with necessary changes is attached too: 
[context_switch_bug.txt](https://github.com/zephyrproject-rtos/zephyr/files/11274427/context_switch_bug.txt)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: main 326eb860b8

aescolar, 2023-04-20:
@alxelax What seems to be happening there is the following:
The do_ecb() call in green area happens in the main thread context (preemptible priority 0).
While that is being executing, an interrupt readies eventually the BT receive thread (Cooperative at CONFIG_BT_DRIVER_RX_HIGH_PRIO (-10)), which goes into executing its own do_ecb()

At the very moment in which the "yellow" part enters into do_ecb() this is
the main (interrupted) thread stack:
```
k_busy_wait() at include/generated/syscalls/kernel.h:133 0x8166e51	
do_ecb() at subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/ecb.c:45 0x81671bc	
ecb_encrypt_be() at subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/ecb.c:69 0x8167480	
bt_encrypt_be() at subsys/bluetooth/controller/crypto/crypto.c:42 0x81568b0	
ccm_auth() at subsys/bluetooth/host/aes_ccm.c:119 0x80a4551	
bt_ccm_encrypt() at subsys/bluetooth/host/aes_ccm.c:228 0x80a5000	
bt_mesh_ccm_encrypt() at subsys/bluetooth/mesh/crypto_tc.c:45 0x812d21d	
bt_mesh_app_encrypt() at subsys/bluetooth/mesh/crypto.c:423 0x80d07db	
trans_encrypt() at subsys/bluetooth/mesh/transport_legacy.c:619 0x80f09e1	
bt_mesh_trans_send() at subsys/bluetooth/mesh/transport_legacy.c:680 0x80f130a	
bt_mesh_access_send() at subsys/bluetooth/mesh/access.c:1,097 0x80d542d	
bt_mesh_model_send() at subsys/bluetooth/mesh/access.c:1,324 0x80d6bc1	
bt_mesh_test_send_async() at tests/bsim/bluetooth/mesh/src/mesh_test.c:429 0x804ff94	
bt_mesh_test_send() at tests/bsim/bluetooth/mesh/src/mesh_test.c:457 0x8050213	
test_lpn_msg_mesh() at tests/bsim/bluetooth/mesh/src/test_friendship.c:545 0x8058b40	
bst_main() at boards/posix/nrf52_bsim/bstests_entry.c:216 0x80994de	
```

And this the BT RX thread stack:
```
Thread #8 [bs_nrf52_bsim_t] 319568 [core: 22] (Suspended : Breakpoint)	
	do_ecb() at subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/ecb.c:35 0x8167078	
	ecb_encrypt_be() at subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/ecb.c:69 0x8167480	
	bt_encrypt_be() at subsys/bluetooth/controller/crypto/crypto.c:42 0x81568b0	
	ccm_auth() at subsys/bluetooth/host/aes_ccm.c:119 0x80a4551	
	bt_ccm_encrypt() at subsys/bluetooth/host/aes_ccm.c:228 0x80a5000	
	bt_mesh_ccm_encrypt() at subsys/bluetooth/mesh/crypto_tc.c:45 0x812d21d	
	bt_mesh_net_encrypt() at subsys/bluetooth/mesh/crypto.c:355 0x80cfd42	
	net_encrypt() at subsys/bluetooth/mesh/net.c:482 0x80c3a47	
	bt_mesh_net_send() at subsys/bluetooth/mesh/net.c:572 0x80c4288	
	send_unseg() at subsys/bluetooth/mesh/transport_legacy.c:195 0x80ede33	
	bt_mesh_ctl_send() at subsys/bluetooth/mesh/transport_legacy.c:1,048 0x80f3ae6	
	send_friend_poll() at subsys/bluetooth/mesh/lpn.c:481 0x80fe0cb	
	bt_mesh_lpn_msg_received() at subsys/bluetooth/mesh/lpn.c:607 0x80fe966	
	bt_mesh_trans_recv() at subsys/bluetooth/mesh/transport_legacy.c:1,599 0x80f7813	
	bt_mesh_net_recv() at subsys/bluetooth/mesh/net.c:871 0x80c6557	
	bt_mesh_scan_cb() at subsys/bluetooth/mesh/adv.c:299 0x80bc30d	
	le_adv_recv() at subsys/bluetooth/host/scan.c:470 0x80b707c	
	bt_hci_le_adv_ext_report() at subsys/bluetooth/host/scan.c:608 0x80b7c49	
	handle_event_common() at subsys/bluetooth/host/hci_core.c:164 0x80a727d	
	handle_event() at subsys/bluetooth/host/hci_core.c:176 0x80a731d	
	hci_le_meta_event() at subsys/bluetooth/host/hci_core.c:2,339 0x80a8e6d	
	handle_event_common() at subsys/bluetooth/host/hci_core.c:164 0x80a727d	
	handle_event() at subsys/bluetooth/host/hci_core.c:176 0x80a731d	
	hci_event() at subsys/bluetooth/host/hci_core.c:2,425 0x80a9184	
	bt_recv() at subsys/bluetooth/host/hci_core.c:3,517 0x80ac4fd	
	recv_thread() at subsys/bluetooth/controller/hci/hci_driver.c:664 0x813b13e	
```

Meaning, either that app main thread is wrongfully not configured as cooperative, or the main thread code is calling an API that should not be called from a preemptible thread; or the mesh stack is the one calling directly into an internal API it should not in the caller context.

alxelax, 2023-04-24:
Seems there is an issue in mesh bsim tests. They are run from the main thread that is preemptible. BLE Mesh should call all functions(including API) only from cooperative, non preemptible threads.
Mesh bsim should be fixed.

closed 2023-05-02 by jgl-meta