zephyrproject-rtos/zephyr #57058: HAS_SEGGER_RTT dependency breaks build for STM32 soc

jyeons, 2023-04-20:
Hi!
I am trying to switch to `main` from `v3.3.0` for some newly supported drivers.
Then I found this dependency issue that breaks the build.
This error is from building blinky sample for nucleo_f302r8.
The same error was thrown from building an application for stm32f427.

```
warning: HAS_SEGGER_RTT (defined at modules/segger/Kconfig:7) has direct dependencies 0 with value n, but is currently being y-selected by the following symbols:
 - SOC_FAMILY_STM32 (defined at soc/arm/st_stm32/Kconfig:6), with value y, direct dependencies y (value: y)

error: Aborting due to Kconfig warnings
```

In `soc/arm/st_stm32/Kconfig`, SOC_FAMILY_STM32 selects HAS_SEGGER_RTT which I understand why it being y-selected.
```
  config SOC_FAMILY_STM32                                                   
      bool
      select HAS_SEGGER_RTT                  
      select BUILD_OUTPUT_HEX
```

However,  where is the direct dependencies 0 with value n from?
There's no default value assigned to HAS_SEGGER_RTT in `modules/segger/Kconfig`
```
 config HAS_SEGGER_RTT
     bool
     help
     Indicates that the platform supports SEGGER J-Link RTT.
```

After bisecting between v3.3.0 and current main, I found the commit 1cfc85b is the first bad one.
Yet, I cannot pick up why this commit causes this problem.
Can anyone check if this error is reproducible and give me a hint where to look at?
Thanks!

nordicjm, 2023-04-20:
You did not do `west update`, the `modules/debug/segger` repo must be on commit `4bfaf28a11c3e5ec29badac744fab6d2f342749e`

Closing since user error, not a bug.

jyeons, 2023-04-20:
Hi, thanks for your reply.
Indeed, we do have our own build system.  After your answer, a question comes into my mind.
We do not use anything from segger module. Does it mean that the segger repository is required as zephyr module by default even if nothing from it is required for the application?
Or am I missing something here?

nordicjm, 2023-04-20:
Give https://github.com/zephyrproject-rtos/zephyr/pull/57062 a try and see if it resolves your issue

jyeons, 2023-04-20:
Thanks a lot! I will try it!

jyeons, 2023-04-20:
Hi! It indeed resolved the problem. Thanks!

PR #57062 (2023-04-20): soc: Only select HAS_SEGGER_RTT if module is available
This prevents configuration errors if a board is configured when the SoC indicates segger RTT support but the segger module is not available.

Fixes #57058
closed 2023-04-20 by carlescufi