zephyrproject-rtos/zephyr #57088: LE Audio: Config QoS with low max transport latency fails

mkasenberg, 2023-04-20:
While running BAP test cases, e.g. BAP/UCL/SCC/BV-035-C, I've met an issue while performing Config QoS with parameters with low max transport latency e.g. 8_1_1 where:
SDU Interval(µs): 7500
Framing: 0x00
Maximum SDU Size (octets): 26
Retransmission Number: 2
Max Transport Latency (us): 8000

If the controller was built with CONFIG_BT_CTLR_CONN_ISO_RELIABILITY_POLICY=y (set by default) then it fails on assertion:
https://github.com/zephyrproject-rtos/zephyr/blob/02655e5b15fe5bf06bcb00410c412c61ab645d5b/subsys/bluetooth/controller/ll_sw/ull_central_iso.c#L411

I printed state of cig object at the moment of the assert:
```gdb
(gdb) print (struct ll_conn_iso_group)*cig
$5 = {
  ull = {
    ref = 0 '\000',
    {
      ticks_active_to_start = 0,
      ticks_prepare_to_start = 0,
      ticks_preempt_to_start = 0,
      ticks_slot = 0
    },
    disabled_cb = 0x0,
    disabled_param = 0x0
  },
  lll = {
    hdr = {
      parent = 0x21006610 <cig_pool>
    },
    handle = 0,
    num_cis = 1 '\001',
    role = 0 '\000',
    paused = 0 '\000',
    resume_cis = 65535,
    window_widening_periodic_us_frac = 0,
    window_widening_prepare_us_frac = 0,
    window_widening_event_us_frac = 0,
    window_widening_max_us = 0
  },
  c_sdu_interval = 7500,
  p_sdu_interval = 7500,
  c_latency = 8000,
  p_latency = 8000,
  cig_ref_point = 0,
  sync_delay = 1016,
  iso_interval = 6,
  cig_id = 0 '\000',
  started = 0 '\000',
  sca_update = 0 '\000',
  {
    central = {
      sca = 0 '\000',
      packing = 0 '\000',
      framing = 0 '\000',
      test = 0 '\000'
    }
  }
}
```
and values of the local variables:
```gdb
(gdb) info locals

c_latency = 8516
p_latency = 8516
i = 0 '\000'
cis = 0x21002538 <cis_pool>
cig = 0x21006610 <cig_pool>
iso_interval_us = <optimized out>
cig_sync_delay = 1016
max_se_length = <optimized out>
c_max_latency = 0
p_max_latency = 0
handle_iter = <optimized out>
cis_count = <optimized out>
se = {{length = 508, total_count = 3 '\003'}, {length = 8, total_count = 139 '�'}}
total_time = <optimized out>
```

Bluetooth Protocol Viewer logs from the BAP/UCL/SCC/BV-035-C test case:
[BAP_UCL_SCC_BV_035_C.cfa.txt](https://github.com/zephyrproject-rtos/zephyr/files/11288039/BAP_UCL_SCC_BV_035_C.cfa.txt)

Debug logs from IUT board:
[BAP_UCL_SCC_BV-035-C_iutctl.log](https://github.com/zephyrproject-rtos/zephyr/files/11288078/BAP_UCL_SCC_BV-035-C_iutctl.log)

I found that setting CONFIG_BT_CTLR_CONN_ISO_LOW_LATENCY_POLICY=y works fine as a workaround, but maybe it is not a right solution here. Let me know if you need more logs.

cvinayak, 2023-04-21:
@mtpr-ot is this something you can look at?

cvinayak, 2023-04-21:
@mkasenberg could you check with the LL_ASSERT commented out? 

@mtpr-ot is the expectation being the use of framed PDUs ?

cvinayak, 2023-06-01:
Setting `CONFIG_BT_CTLR_CONN_ISO_LOW_LATENCY_POLICY=y` is an acceptable solution. Closing issue.

closed 2023-06-01 by cvinayak