zephyrproject-rtos/zephyr #57103: drivers: serial: ns16550: test uart async failed due to undefined structure

cs3806, 2023-04-21:
**Describe the bug**
* Running `/tests/drivers/uart/uart_async_api` for UART async API testing, failed on every testsuite.
* After looking into `drivers/serial/uart_ns16550.c` and found that the driver did not define the parameters needed for the test, which resulted in all of the testsuites failing.
* For instance, the code as shown below is from the callback function of the first testsuite **uart_chained_read**, where 'evt->data.rx.buf', 'evt->data.rx.offset' and 'evt->data.rx.len' are both not exist in ns16550.c, which brought about an assertion failure during the test. 
https://github.com/zephyrproject-rtos/zephyr/blob/9eae6fda9930eb1ae674d02f6992dbbc274fbd2a/tests/drivers/uart/uart_async_api/src/test_uart_async.c#L246
https://github.com/zephyrproject-rtos/zephyr/blob/9eae6fda9930eb1ae674d02f6992dbbc274fbd2a/tests/drivers/uart/uart_async_api/src/test_uart_async.c#L247
* There are other similar issues that caused other testsuites in this test to fail.
* UART driver: ns16550
[https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/serial/uart_ns16550.c](url)

**Expected behavior**
Pass all the tests.

**Environment:**
 * OS: Ubuntu 22.04.2
 * Toolchain: Zephyr SDK 0.15.2
 * UART driver: ns16550

dcpleung, 2023-05-11:
@akanisetti Please fix the ns16550 UART async implementation. If it is not fixed before the merge window is closed, I will have to revert it.

akanisetti, 2023-05-12:
> * FCR

Sure, will fix the issue.