zephyrproject-rtos/zephyr #57120: Running Twister tests on RTT connectd devices

IvanVnucec, 2023-04-21:
I have a Zephyr device that lacks Serial port so I'm using Segger J-Link RTT instead.
I would like to run Twister tests on my device and check the Serial console output.
I didn't considered any alternatives.

kmeinhar, 2023-06-16:
The workaround to achieve this for me was to start `JLinkRTTLogger` on the machine executing the testing.
This makes it possible to read RTT output over telnet port 19021.
Twister can then use the Python script below to read the RTT output with argument ` --device-serial-pty script.py`.

For example:
`./scripts/twister -p BOARD -T path/to/tests  --device-serial-pty script.py --device-testing`

There is probably a better way than starting `JLinkRTTLogger` on the testing machine. `JLinkGDBServer` already supports flashing and RTT output on port 19021 at the same time. So instead of `west flash` twister could use `west debug`. But I did not find a good way to integrate that quickly.

```
#!/usr/bin/env python3
import telnetlib3

HOST = 'localhost'
PORT = 19021

async def read_telnet_output():
    try:
        # Connect to the telnet host
        reader, writer = await telnetlib3.open_connection(HOST, PORT)
        
        # Read the output until the connection is closed
        async for line in reader:
            print(line)
        
    except ConnectionRefusedError:
        print(f"Connection refused: Unable to connect to {HOST}:{PORT}")
        
    except Exception as e:
        print(f"An error occurred: {str(e)}")
        
    finally:
        writer.close()

# Run the event loop and call the function to start reading the telnet output
import asyncio
asyncio.run(read_telnet_output())

```

IvanVnucec, 2023-07-04:
@nashif Do you have an idea how this feature might be implemented? I'm wiling to get on on this if it's reasonably simple.

IvanVnucec, 2023-07-04:
> 

In your code, i think that the `writer.close()` line should be in an `else`, not in the `finally` because writer is not created when we have an exception and the `UnboundLocalError: local variable 'writer' referenced before assignment` error could happen.

IvanVnucec, 2023-07-04:
> Also, I get `Connection refused: Unable to connect to localhost:19021` error. I've tried `telnet localhost 19021` but it also refused to establish connection, although J-Link RTT Viewer with Existing Session and Auto Reconnect can indeed establish the connection no problem. I'm sure I'm not trying to connect to the J-Link when the Viewer is connected at the same time.
I've managed to successfully connect by re-connecting jlink device into pc and retrying.

kmeinhar, 2023-07-04:
> @kmeinhar In your code, i think that the `writer.close()` line should be in an `else`, not in the `finally` because writer is not created when we have an exception and the `UnboundLocalError: local variable 'writer' referenced before assignment` error could happen.
> 
> Also, I get `Connection refused: Unable to connect to localhost:19021` error. I've tried `telnet localhost 19021` but it also refused to establish connection, although J-Link RTT Viewer with Existing Session and Auto Reconnect can indeed establish the connection no problem. I'm sure I'm not trying to connect to the J-Link when the Viewer is connected at the same time.

I am honestly not a good Python developer. So the Code above is simply the output from ChatGPT. It should be more an inspiration than the production ready code.

> Also, I get `Connection refused: Unable to connect to localhost:19021` error. I've tried `telnet localhost 19021` but it also refused to establish connection, although J-Link RTT Viewer with [...]

Hm I can not reproduce that for my platform. Both RTT Viewer and RTT Logger expose the 19021 port for me.
You could see if GDB Server has the same problem. Just start debug session with  `west debug`, now port 19021 should also output RTT logs over telnet.

nashif, 2023-08-17:
> @nashif Do you have an idea how this feature might be implemented? I'm wiling to get on on this if it's reasonably simple.

using PTY and a script is how I would implement this.

rego21, 2023-09-18:
Hi @kmeinhar ,

I tried using the python scrip with twister as suggested, but it fails when trying to flash the device - timeout error. When advice how I should proceed ? 
Thanks!

IvanVnucec, 2023-09-18:
> Hi @kmeinhar ,
> 
> I tried using the python scrip with twister as suggested, but it fails when trying to flash the device - timeout error. When advice how I should proceed ? 
> Thanks!

Can you provide the script you're mentioned?

rego21, 2023-09-18:
> The workaround to achieve this for me was to start `JLinkRTTLogger` on the machine executing the testing. This makes it possible to read RTT output over telnet port 19021. Twister can then use the [...]

This one @IvanVnucec

IvanVnucec, 2023-09-18:
> This one @IvanVnucec

And where is here the code that flashes the device? Can you also post full error printout?

rego21, 2023-09-18:
I think this is done by twister, so here goes the twister log:

```
2023-09-18 17:59:53,402 - twister - DEBUG - Flash command: ['ninja', '-C', '<twsiter output folder>', 'flash']
2023-09-18 17:59:54,456 - twister - DEBUG - DEVICE: SEGGER J-Link V7.86h - Real time terminal output
2023-09-18 17:59:54,457 - twister - DEBUG - DEVICE: 
2023-09-18 17:59:54,458 - twister - DEBUG - DEVICE: J-Link OB-nRF5340-NordicSemi compiled Nov  7 2022 16:22:01 V1.0, SN=1050248111
2023-09-18 17:59:54,459 - twister - DEBUG - DEVICE: 
2023-09-18 17:59:54,459 - twister - DEBUG - DEVICE: Process: JLinkRTTLoggerExe
2023-09-18 17:59:54,459 - twister - DEBUG - DEVICE: 
2023-09-18 18:00:53,441 - twister - WARNING - Flash operation timed out.
2023-09-18 18:00:54,058 - twister - DEBUG - Timed out while monitoring serial output on nrf52840dk_nrf52840
2023-09-18 18:00:54,061 - twister - DEBUG - Process /home/miguel/Documents/omniaActions/rtt_serial.py terminated outs: None errs None
```
rtt_serial.py is the script mentioned above.

Thanks @IvanVnucec !

kmeinhar, 2023-09-18:
To me that looks like the problem happens before the Python script is even executed. It does not handle flashing of the device.

Could you try to execute test that do not require the RTT serial output?

Twister also generates other log files in the `twister-out` directory, e.g. `handler.log`. Could you check for all `*.log` files and see if there is anything relevant to the problem?

rego21, 2023-09-18:
Yeah I agree, seems it is trying to flash using the following device according to the logs.: 
`Using serial device /dev/pts/6 @ 115200 baud` 
And this seems wrong to me...

 When you tried this do you know if twister tried to flash the device ? And if so do you have any logs ?
 Thanks @kmeinhar !

kmeinhar, 2023-09-19:
Yes twister tries and succeeds to flash my device. The log below is sanitized output from our private project that runs in CI
As you can see in the log below it uses the `west flash` command -> Please try if west flash works with the hello-world example, without twister

One thing to note here is that we have our targets connected to a remote device. Thats why we need a `gdbport` runner.
This works the same has the JLink Runner just that the JLinkGDBServer is started on remote machine and connected over IP.
I am considering creating a PR for this `gdbport` runner, but did not find the opportunity yet.

```
2023-09-19 08:05:15,676 - twister - DEBUG - run test: efr32bg22_brd4184a/<...>
2023-09-19 08:05:15,677 - twister - DEBUG - Using serial device /dev/pts/80 @ 115200 baud
2023-09-19 08:05:15,678 - twister - DEBUG - Flash command: ['west', 'flash', '--skip-rebuild', '-d', '/home/<...>/projects/zephyr/zephyr-3.4/twister-out/efr32bg22_brd4184a/<...>', '--', '--gdb-port=3332', '--gdb-ip=<...>']
2023-09-19 08:05:18,495 - twister - DEBUG - -- west flash: using runner gdbport
['/home/<...>r/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb']
Remote debugging using 172.14.1.41:3332
0x00008f1a in ?? ()
Reading symbols from /home/<...>/twister-out/efr32bg22_brd4184a/<...>/zephyr/zephyr.elf...
Loading section rom_start, size 0x400 lma 0x0
Loading section text, size 0xeb78 lma 0x400
Loading section .ARM.exidx, size 0x8 lma 0xef78
Loading section initlevel, size 0xb0 lma 0xef80
Loading section device_area, size 0x140 lma 0xf030
Loading section sw_isr_table, size 0x1e0 lma 0xf170
Loading section ztest, size 0x38 lma 0xf350
Loading section log_const_area, size 0xb0 lma 0xf388
Loading section log_backend_area, size 0x10 lma 0xf438
Loading section zephyr_dbg_info, size 0x40 lma 0xf448
Loading section rodata, size 0x2dd0 lma 0xf488
Loading section .ramfunc, size 0x1c0 lma 0x12258
Loading section datas, size 0x2cae lma 0x12418
Loading section device_states, size 0x20 lma 0x150c6
Loading section log_mpsc_pbuf_area, size 0x44 lma 0x150e8
Loading section log_msg_ptr_area, size 0x4 lma 0x1512c
Loading section k_mutex_area, size 0x14 lma 0x15130
Loading section k_sem_area, size 0x18 lma 0x15144
Loading section .last_section, size 0x4 lma 0x1515c
Start address 0x00007e54, load size 86366
Transfer rate: 2720 KB/sec, 3925 bytes/write.

2023-09-19 08:05:19,771 - twister - DEBUG - DEVICE: Connected to 172.14.1.41:19023
2023-09-19 08:05:19,773 - twister - DEBUG - DEVICE: SEGGER J-Link V7.82f - Real time terminal output
2023-09-19 08:05:19,773 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:19,773 - twister - DEBUG - DEVICE: Silicon Labs J-Link OB compiled Nov 25 2020 13:01:14 V1.0, SN=440237457
2023-09-19 08:05:19,773 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:19,774 - twister - DEBUG - DEVICE: Process: JLinkGDBServerCLExe
2023-09-19 08:05:19,776 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:19,776 - twister - DEBUG - DEVICE: *** Booting Zephyr OS build zephyr-v3.4.0-1986-g4d304a2b9eab ***
2023-09-19 08:05:19,776 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:19,776 - twister - DEBUG - DEVICE: Running TESTSUITE <...>
2023-09-19 08:05:19,777 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:19,777 - twister - DEBUG - DEVICE: ===================================================================
2023-09-19 08:05:19,777 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:19,777 - twister - DEBUG - DEVICE: START - test_minimal
2023-09-19 08:05:19,777 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:19,777 - twister - DEBUG - DEVICE: PASS - test_minimal in 0.001 seconds
2023-09-19 08:05:19,777 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:19,778 - twister - DEBUG - DEVICE: ===================================================================
2023-09-19 08:05:19,778 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:19,778 - twister - DEBUG - DEVICE: TESTSUITE <...> succeeded
2023-09-19 08:05:19,778 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:20,743 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:20,743 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:20,744 - twister - DEBUG - DEVICE: ------ TESTSUITE SUMMARY START ------
2023-09-19 08:05:20,745 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:20,745 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:20,745 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:21,767 - twister - DEBUG - DEVICE: SUITE PASS - 100.00% [<...>]: pass = 1, fail = 0, skip = 0, total = 1 duration = 0.001 seconds
2023-09-19 08:05:21,767 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:22,781 - twister - DEBUG - DEVICE: - PASS - [<...>] duration = 0.001 seconds
2023-09-19 08:05:22,781 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:23,804 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:23,804 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:24,822 - twister - DEBUG - DEVICE: ------ TESTSUITE SUMMARY END ------
2023-09-19 08:05:24,823 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:24,823 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:24,823 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:25,853 - twister - DEBUG - DEVICE: ===================================================================
2023-09-19 08:05:25,853 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:25,853 - twister - DEBUG - DEVICE: RunID: <...>
2023-09-19 08:05:25,853 - twister - DEBUG - DEVICE: 
2023-09-19 08:05:25,853 - twister - DEBUG - DEVICE: PROJECT EXECUTION SUCCESSFUL
2023-09-19 08:05:25,855 - twister - DEBUG - Process <...> terminated outs: None errs None
2023-09-19 08:05:25,855 - twister - DEBUG - Expected suite names:['<...>']
2023-09-19 08:05:25,855 - twister - DEBUG - Detected suite names:['<...>']
2023-09-19 08:05:25,855 - twister - DEBUG - run status: efr32bg22_brd4184a/<...> passed
```

IvanVnucec, 2023-09-19:
> Yes twister tries and succeeds to flash my device. The log below is sanitized output from our private project that runs in CI As you can see in the log below it uses the `west flash` command -> [...]

Is this printout generated from RTT?

kmeinhar, 2023-09-19:
This is the relevant part of the `twister-out/twister.log` file. But this includes the RTT log.
The RTT log is in `twister-out/efr32bg22_brd4184a/<...>/handler.log`

```
Connected to <...>:19023

SEGGER J-Link V7.82f - Real time terminal output

Silicon Labs J-Link OB compiled Nov 25 2020 13:01:14 V1.0, SN=440237457

Process: JLinkGDBServerCLExe

*** Booting Zephyr OS build zephyr-v3.4.0-1986-g4d304a2b9eab ***

Running TESTSUITE <...>

===================================================================

START - test_minimal

PASS - test_minimal in 0.001 seconds

===================================================================

TESTSUITE <...> succeeded

------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [<...>]: pass = 1, fail = 0, skip = 0, total = 1 duration = 0.001 seconds

- PASS - [<...>] duration = 0.001 seconds

------ TESTSUITE SUMMARY END ------

===================================================================

RunID: <...>

PROJECT EXECUTION SUCCESSFUL
```

rego21, 2023-09-19:
> Yes twister tries and succeeds to flash my device. The log below is sanitized output from our private project that runs in CI As you can see in the log below it uses the `west flash` command -> [...]

With hello world example you mean a simple program ? Im using a nrf52840DK for testing this and yeah I'm able to flash the device with serial port (USB) without any problems. I also have used twister with the option --device-serial /dev/tty<dev> and it works. But with RTT I'm following the approach you suggested but so far I'm not able to flash the device... To clarify what I'm doing is:
* On my computer have one nrf52840 connect to it. The DK is connected to other DK by SWD - debug in and debug out pins
* On the same computer I start JLinkRTTLogger
*  Then I run twister with the script you suggested

Probably Im missing something...

Thanks for the help!

kmeinhar, 2023-09-19:
> 2023-09-18 17:59:54,456 - twister - DEBUG - DEVICE: SEGGER J-Link V7.86h - Real time terminal output
> 2023-09-18 17:59:54,457 - twister - DEBUG - DEVICE:
> 2023-09-18 17:59:54,458 - twister - DEBUG - [...]

Took a second look at this log and missed that connecting to RTT log actually worked. This is indicated by `DEBUG - DEVICE:` log. This should also be be in the `handler.log` of your test.
Can you please validate that your handler.log says something like this:
```
SEGGER J-Link V7.86h - Real time terminal output
J-Link OB-nRF5340-NordicSemi compiled Nov  7 2022 16:22:01 V1.0, SN=105024811
Process: JLinkRTTLoggerExe
```

But it looks like the RTT log is simply empty. There could be multiple reasons:
* RTT log is not configured in the test project
  * Sometimes tests have additional confgiurations, please check the generated `.config` in twister-out build directory
* The Python script might shutdown too early
  * Try to but a sleep before connecting to Telnet port
* Maybe `JLinkRTTLogger` blocks `JLinkGDBServer` to connect to target
  * That would mean the Python script would needs to start  `JLinkRTTLogger` after the flash process, with a delay
  * Working around this restriction that does not look like a hack is probably not trivial, if  I find the time will see what is possible there.

rego21, 2023-09-19:
Yeah It has that message. The only missing thing here is the flash process. Can you take a look at what I mentioned here : 

Meanwhile I will try your suggestions, thanks @kmeinhar !

> > Yes twister tries and succeeds to flash my device. The log below is sanitized output from our private project that runs in CI As you can see in the log below it uses the `west flash` command -> Please try if west flash works with the hello-world example, without twister
> > One thing to note here is that we have our targets connected to a remote device. Thats why we need a `gdbport` runner. This works the same has the JLink Runner just that the JLinkGDBServer is started on remote machine and connected over IP. I am considering creating a PR for this `gdbport` runner, but did not find the opportunity yet.
> > ```
> > 2023-09-19 08:05:15,676 - twister - DEBUG - run test: efr32bg22_brd4184a/<...>
> > 2023-09-19 08:05:15,677 - twister - DEBUG - Using serial device /dev/pts/80 @ 115200 baud
> > 2023-09-19 08:05:15,678 - twister - DEBUG - Flash command: ['west', 'flash', '--skip-rebuild', '-d', '/home/<...>/projects/zephyr/zephyr-3.4/twister-out/efr32bg22_brd4184a/<...>', '--', '--gdb-port=3332', '--gdb-ip=<...>']
> > 2023-09-19 08:05:18,495 - twister - DEBUG - -- west flash: using runner gdbport
> > ['/home/<...>r/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb']
> > Remote debugging using 172.14.1.41:3332
> > 0x00008f1a in ?? ()
> > Reading symbols from /home/<...>/twister-out/efr32bg22_brd4184a/<...>/zephyr/zephyr.elf...
> > Loading section rom_start, size 0x400 lma 0x0
> > Loading section text, size 0xeb78 lma 0x400
> > Loading section .ARM.exidx, size 0x8 lma 0xef78
> > Loading section initlevel, size 0xb0 lma 0xef80
> > Loading section device_area, size 0x140 lma 0xf030
> > Loading section sw_isr_table, size 0x1e0 lma 0xf170
> > Loading section ztest, size 0x38 lma 0xf350
> > Loading section log_const_area, size 0xb0 lma 0xf388
> > Loading section log_backend_area, size 0x10 lma 0xf438
> > Loading section zephyr_dbg_info, size 0x40 lma 0xf448
> > Loading section rodata, size 0x2dd0 lma 0xf488
> > Loading section .ramfunc, size 0x1c0 lma 0x12258
> > Loading section datas, size 0x2cae lma 0x12418
> > Loading section device_states, size 0x20 lma 0x150c6
> > Loading section log_mpsc_pbuf_area, size 0x44 lma 0x150e8
> > Loading section log_msg_ptr_area, size 0x4 lma 0x1512c
> > Loading section k_mutex_area, size 0x14 lma 0x15130
> > Loading section k_sem_area, size 0x18 lma 0x15144
> > Loading section .last_section, size 0x4 lma 0x1515c
> > Start address 0x00007e54, load size 86366
> > Transfer rate: 2720 KB/sec, 3925 bytes/write.
> > 
> > 2023-09-19 08:05:19,771 - twister - DEBUG - DEVICE: Connected to 172.14.1.41:19023
> > 2023-09-19 08:05:19,773 - twister - DEBUG - DEVICE: SEGGER J-Link V7.82f - Real time terminal output
> > 2023-09-19 08:05:19,773 - twister - DEBUG - DEVICE: 
> > 2023-09-19 08:05:19,773 - twister - DEBUG - DEVICE: Silicon Labs J-Link OB compiled Nov 25 2020 13:01:14 V1.0, SN=440237457
> > 2023-09-19 08:05:19,773 - twister - DEBUG - DEVICE: 
> > 2023-09-19 08:05:19,774 - twister - DEBUG - DEVICE: Process: JLinkGDBServerCLExe
> > 2023-09-19 08:05:19,776 - twister - DEBUG - DEVICE: 
> > 2023-09-19 08:05:19,776 - twister - DEBUG - DEVICE: *** Booting Zephyr OS build zephyr-v3.4.0-1986-g4d304a2b9eab ***
> > 2023-09-19 08:05:19,776 - twister - DEBUG - DEVICE: 
> > 2023-09-19 08:05:19,776 - twister - DEBUG - DEVICE: Running TESTSUITE <...>
> > 2023-09-19 08:05:19,777 - twister - DEBUG - DEVICE: 
> > 2023-09-19 08:05:19,777 - twister - DEBUG - DEVICE: ===================================================================
> > 2023-09-19 08:05:19,777 - twister - DEBUG - DEVICE: 
> > 2023-09-19 08:05:19,777 - twister - DEBUG - DEVICE: START - test_minimal
> > 2023-09-19 08:05:19,777 - twister - DEBUG - DEVICE: 
> > 2023-09-19 08:05:19,777 - twister - DEBUG - DEVICE: PASS - test_minimal in 0.001 seconds
> > 2023-09-19 08:05:19,777 - twister - DEBUG - DEVICE: 
> > 2023-09-19 08:05:19,778 - twister - DEBUG - DEVICE: ===================================================================
> > 2023-09-19 08:05:19,778 - twister - DEBUG - DEVICE: 
> > 2023-09-19 08:05:19,778 - twister - DEBUG - DEVICE: TESTSUITE <...> succeeded
> > 2023-09-19 08:05:19,778 - twister - DEBUG - DEVICE: 
> > 2023-09-19 08:05:20,743 - twister - DEBUG - DEVICE: 
> > 2023-09-19 08:05:20,743 - twister - DEBUG - DEVICE: 
> > 2023-09-19 08:05:20,744 - twister - DEBUG - DEVICE: ------ TESTSUITE SUMMARY START ------
> > 2023-09-19 08:05:20,745 - twister - DEBUG - DEVICE: 
> > 2023-09-19 08:05:20,745 - twister - DEBUG - DEVICE: 
> > 2023-09-19 08:05:20,745 - twister - DEBUG - DEVICE: 
> > 2023-09-19 08:05:21,767 - twister - DEBUG - DEVICE: SUITE PASS - 100.00% [<...>]: pass = 1, fail = 0, skip = 0, total = 1 duration = 0.001 seconds
> > 2023-09-19 08:05:21,767 - twister - DEBUG - DEVICE: 
> > 2023-09-19 08:05:22,781 - twister - DEBUG - DEVICE: - PASS - [<...>] duration = 0.001 seconds
> > 2023-09-19 08:05:22,781 - twister - DEBUG - DEVICE: 
> > 2023-09-19 08:05:23,804 - twister - DEBUG - DEVICE: 
> > 2023-09-19 08:05:23,804 - twister - DEBUG - DEVICE: 
[... 2 lines cut ...]
> > 2023-09-19 08:05:24,823 - twister - DEBUG - DEVICE: 
> > 2023-09-19 08:05:24,823 - twister - DEBUG - DEVICE: 
> > 2023-09-19 08:05:25,853 - twister - DEBUG - DEVICE: ===================================================================
[... cut ...]
> With hello world example you mean a simple program ? Im using a nrf52840DK for testing this and yeah I'm able to flash the device with serial port (USB) without any problems. I also have used twister with the option --device-serial /dev/tty and it works. But with RTT I'm following the approach you suggested but so far I'm not able to flash the device... To clarify what I'm doing is:
> 
>     * On my computer have one nrf52840 connect to it. The DK is connected to other DK by SWD - debug in and debug out pins
> 
>     * On the same computer I start JLinkRTTLogger
> 
>     * Then I run twister with the script you suggested
> 
> 
> Probably Im missing something...
> 
> Thanks for the help!

kmeinhar, 2023-09-19:
> With hello world example you mean a simple program ? Im using a nrf52840DK for testing this and yeah I'm able to flash the device with serial port (USB) without any problems. I also have used [...]

I think I am able to replicate the behavior that you see on my side as well.
But I also think I found a temporary workaround.
Line 334 to 339 state: https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/west_commands/runners/jlink.py#L334
```
        # Reset the Debug Port CTRL/STAT register
        # Under normal operation this is done automatically, but if other
        # JLink tools are running, it is not performed.
        # The J-Link scripting layer chains commands, meaning that writes are
        # not actually performed until after the next operation. After writing
        # the register, read it back to perform this flushing.
```
Since we actually are running other JLink tools we do not want to reset CTRL/STAT register.
I am not really sure why it worked for me previously this commit is from Jun 2020: https://github.com/zephyrproject-rtos/zephyr/commit/9c39a930148d3ab843c11e071c2741251cc75764
So you could try to comment them out. After that the setup works for me. Could you check that?
```
diff --git a/scripts/west_commands/runners/jlink.py b/scripts/west_commands/runners/jlink.py
index 80b7de2dff..5d8e0d0e13 100644
--- a/scripts/west_commands/runners/jlink.py
+++ b/scripts/west_commands/runners/jlink.py
@@ -337,8 +337,8 @@ class JLinkBinaryRunner(ZephyrBinaryRunner):
         # The J-Link scripting layer chains commands, meaning that writes are
         # not actually performed until after the next operation. After writing
         # the register, read it back to perform this flushing.
-        lines.append('writeDP 1 0')
-        lines.append('readDP 1')
+        # lines.append('writeDP 1 0')
+        # lines.append('readDP 1')
 
         lines.append('q') # Close the connection and quit
 
```

kmeinhar, 2023-09-19:
Hm I am kind of out of ideas right now. You would need to investigate yourself a bit more to locate the issue.

Here are some pointers:
* Your error is printed here: https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/pylib/twister/twisterlib/handlers.py#L665
* You could see what command is created here and see if this one also fails when executing directly on CLI: https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/pylib/twister/twisterlib/handlers.py#L651
  * I assume it executes `west flash`. Enable in the following area debug logging: https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/west_commands/runners/jlink.py#L278
* The serial PTY Python script is started here. Comment this line out to see if flashing is now possible: https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/pylib/twister/twisterlib/handlers.py#L644
  * If this is the case that it is indeed my RTT python script

rego21, 2023-09-19:
@kmeinhar Thanks for helping me on this! Meanwhile I solved the issues, here what I did:
* Use nrfjprog as west runner
* Create a new script using pylink lib, so you dont need to open JLink before

Thanks again!

marinjurjevic, 2023-09-21:
We've been using https://github.com/codecoup/tools-rtt2pty for bridging RTT channels to pty devices which can be opened with any serial monitor tool.
I'm wondering if this tool can be integrated with twister for easier RTT handling. SEGGER tools are really bad, especially CLI ones.

carriegong, 2023-09-21:
Is there something needed to change on the port number of that script? If so where do i get the port number? 
i get this on handler.log
`SEGGER J-Link V7.80c - Real time terminal output
V0.0, SN=0
Process: jlinkarm_nrf_worker_osx`

carriegong, 2023-09-22:
> Is there something needed to change on the port number of that script? If so where do i get the port number? i get this on handler.log `SEGGER J-Link V7.80c - Real time terminal output V0.0, SN=0 Process: jlinkarm_nrf_worker_osx`

nevermind I see the logs when i open RTT viewer...Is there a way to get this in twister? My twister times out...

rego21, 2023-09-22:
> nevermind I see the logs when i open RTT viewer...Is there a way to get this in twister? My twister times out...

I would suggest to use nrjprog and pylink lib for the script (you can easily finds example of the library). But I don't know what setup you are using. Maybe you can expose your use case so we can help you in a better way :)

carriegong, 2023-09-22:
> I would suggest to use nrjprog and pylink lib for the script (you can easily finds example of the library). But I don't know what setup you are using. Maybe you can expose your use case so we can [...]

I am just using a custom nrf52840 board 
I just tried having RTTlogger open but i get ERROR: Failed to read data.
I can see the logs on RTTviewer though

rego21, 2023-09-22:
So if you have something like I have nrf52840DK connect to a custom board (Debug out/Debug In pins). I would suggest to do what I mentioned above. I had the same issues as you and switched to pylink, got better results with it. But all depends of what you want to achieve...

carriegong, 2023-09-22:
> So if you have something like I have nrf52840DK connect to a custom board (Debug out/Debug In pins). I would suggest to do what I mentioned above. I had the same issues as you and switched to pylink, got better results with it. But all depends of what you want to achieve...

Is it possible you can share you script? Ty

rego21, 2023-09-22:
Sadly I cant, but like I said you can easily find really similar an workable script on pylink examples

rego21, 2023-09-25:
> So i got it working in handler.log however twister always Errors out as device error (flash error)
> 
> I am using pylink... but there is an error with twister.. how to resolve it? `INFO - Adding tasks to the queue... INFO - Added initial list of jobs to queue Exception in thread Thread-1 (monitor_serial): Traceback (most recent call last): File "/usr/local/Cellar/python@3.10/3.10.13/Frameworks/Python.framework/Versions/3.10/lib/python3.10/threading.py", line 1016, in _bootstrap_inner self.run() File "/usr/local/Cellar/python@3.10/3.10.13/Frameworks/Python.framework/Versions/3.10/lib/python3.10/threading.py", line 953, in run self._target(*self._args, **self._kwargs) File "/opt/nordic/ncs/v2.4.0/zephyr/scripts/pylib/twister/twisterlib/handlers.py", line 363, in monitor_serial ser.close() File "/usr/local/lib/python3.10/site-packages/serial/serialposix.py", line 533, in close os.close(self.fd) OSError: [Errno 9] Bad file descriptor`

Have you tried to run only the script without twister ? Try to do it, if it works it should output the RTT log

carriegong, 2023-09-26:
nvm have it working. somehow test's that didn't pass on that board caused that issue. i am not sure how.. anyways thank you!!

moose-rivieh, 2024-01-05:
@carriegong could you share how you will able to get it working? I am using the same Python script shared here but I am getting a permission error when I run Twister. The way I am doing it is I start JLinkRTTLogger, then I run the command `./../zephyr/scripts/twister -W -T tests/lib/syc_lib/ -p rivieh550p2@3.0.0 --board-root boards/ --device-serial-pty scripts/rtt_telnet.py --device-testing`

Am I missing anything? I tried to run in a Python virtual environment and still got the error. Please note Twister works if I just perform a project build and I have been able to connect to the hardware using RTT over telnet at localhost port 19021. Below is a screenshot of the error.

[image]

carriegong, 2024-02-01:
> @carriegong could you share how you will able to get it working? I am using the same Python script shared here but I am getting a permission error when I run Twister. The way I am doing it is I [...]

maybe allow permission for that file chmod?

fab-uleuh, 2024-05-27:
https://github.com/fab-uleuh/zephyr-twister-rtt