zephyrproject-rtos/zephyr #57129: drivers: dma: emul driver

cfriedt, 2023-04-21:
**Is your feature request related to a problem? Please describe.**
Many Zephyr drivers incorporate what we call an `emul` driver.

Covered to some extent by @sjg20 's and @cfriedt's talks at ZDS 2022

Overall, `emul` drivers provide a number of benefits
* prototypical model for Driver API for the specific class of driver
  * physical drivers must mimic this behaviour
* driver-specific DT bindings to allow for customization on a per-instance basis
* **Fast, Functional Testing** for application code (regression testing, prototyping, ..)
* private API extensions
  * customizable backend functionality (e.g. "wiring" for `gpio_emul`)
  * test and cover very specific corner cases
  * such corner cases can be very difficult to reproduce in reality
  * coverage in safety-critical and high-reliability systems can be extremely important (think Automotive, Avionics, Infrastructure, ...)
* Can be instantiated on any board, regardless of architecture (`native_posix`, `qemu*`, actual physical boards)
* Can be used for more than just testing (e.g. `gpio_emul_sdl.c`)

**Describe the solution you'd like**
* Probably an RFC to start off with
  * Outline DT bindings, testsuite integration, strategies
  * Buy-in from DMA subsystem maintainer @teburd
  * Suggestions / feedback from other `emul` driver authors

**Describe alternatives you've considered**
Not having a DMA `emul` driver (which is the case today). Not that anything is wrong with the DMA subsystem today, but we want **Fast Functional Testing**!

**Additional context**
May coincide with existing activities by @jgl-meta

teburd, 2023-04-23:
Which dma are you looking to emulate? They all behave differently

cfriedt, 2023-04-23:
> Which dma are you looking to emulate? They all behave differently 

All of them :-)

teburd, 2023-04-27:
> > Which dma are you looking to emulate? They all behave differently
> 
> All of them :-)

I'm fine with adding emulators to DMA generally. So if you are suggesting emulating specific DMA controllers no objections from me.

cfriedt, 2023-05-03:
I've got a branch going. Will make some additional progress on it shortly.

closed 2023-12-04 by cfriedt