zephyrproject-rtos/zephyr #57133: Compile error building SNTP library for native_posix while NET_SOCKETS_POSIX_NAMES=n

heinwessels, 2023-04-21:
**Describe the bug**
It's not possible to build an application with the SNTP library using the `native_posix` board when `CONFIG_NET_SOCKETS_POSIX_NAMES=n` and `CONFIG_POSIX_API=n`. The error shown is

```
In file included from /home/hein/zephyrproject/zephyr/include/zephyr/posix/unistd.h:10,
                 from /home/hein/zephyrproject/zephyr/include/zephyr/net/sntp.h:15,
                 from /home/hein/zephyrproject/zephyr/samples/net/sockets/sntp_client/src/main.c:11:
/home/hein/zephyrproject/zephyr/include/zephyr/posix/sys/stat.h:40:10: fatal error: sys/_timespec.h: No such file or directory
   40 | #include <sys/_timespec.h>
      |          ^~~~~~~~~~~~~~~~~
```
A previous PR #50805 was merged by a member of our team to allow building SNTP for the same scenario, but I'm relatively confident that this functionality was broken again with a necessary change in https://github.com/zephyrproject-rtos/zephyr/commit/6f4e96bc2482e868590f3f72cfed92d14dcd6de1. 

After conferring with @cfriedt he mentioned that this could be fixed by changing the headers in `sntp.h`, something like:
```
#ifdef CONFIG_ARCH_POSIX
#include <sys/socket.h>
#include <unistd.h>
#include <poll.h>
#else
#include <zephyr/posix/sys/socket.h>
#include <zephyr/posix/unistd.h>
#include <zephyr/posix/poll.h>
#endif
```

I could get it to work, but it turned into a little oil spill where it required fixing a few files named below. I'm not sure what is the right way to fix it, and the fix seemed quite messy. If no one else is willing I will create a draft PR with these fixes soon.
```
include/zephyr/net/sntp.h
include/zephyr/net/socketutils.h
subsys/net/lib/sntp/sntp.c
subsys/net/lib/sntp/sntp_simple.c
subsys/net/lib/utils/addr_utils.c
```

**To Reproduce**
Steps to reproduce the behavior:
1. In `zephyr/samples/net/sockets/sntp_client/prj.conf` change to `CONFIG_NET_SOCKETS_POSIX_NAMES=n`.
2. `west build -p always -b native_posix samples/net/sockets/sntp_client`
4. Witness the mentioned compiler error

**Expected behavior**
For the application to build without warnings.

**Impact**
This is currently a showstopper in our attempt to migrate from Zephyr 2.6 to the upcoming Zephyr 3.4.

**Environment (please complete the following information):**
 - Linux
 - Zephyr SDK 0.16
 - Zephyr commit: 3707014f20f7c9fcb71e4f13fd23cf897b1dcbdc

---
Here is my attempt to fix it, which reverted what was done in the mentioned #50805
```diff
From 206743e3d574fd457eb0aebb0caf4f5386e8cbd9 Mon Sep 17 00:00:00 2001
From: Hein Wessels <heinwessels93@gmail.com>
Date: Fri, 21 Apr 2023 16:18:40 +0200
Subject: [PATCH] Fix building sntp with or without
 CONFIG_NET_SOCKETS_POSIX_NAMES

---
 include/zephyr/net/sntp.h                | 14 ++++++++++++--
 include/zephyr/net/socketutils.h         |  7 +++++--
 samples/net/sockets/sntp_client/prj.conf |  2 +-
 subsys/net/lib/sntp/sntp.c               | 15 ++++++++-------
 subsys/net/lib/sntp/sntp_simple.c        |  1 +
 subsys/net/lib/utils/addr_utils.c        |  6 ++++++
 6 files changed, 33 insertions(+), 12 deletions(-)

diff --git a/include/zephyr/net/sntp.h b/include/zephyr/net/sntp.h
index c069c6e777..3586e2b0bc 100644
--- a/include/zephyr/net/sntp.h
+++ b/include/zephyr/net/sntp.h
@@ -8,8 +8,18 @@
 #ifndef ZEPHYR_INCLUDE_NET_SNTP_H_
 #define ZEPHYR_INCLUDE_NET_SNTP_H_
 
-#ifdef CONFIG_NET_SOCKETS_POSIX_NAMES
-#include <zephyr/net/socket.h>
+#ifdef CONFIG_ARCH_POSIX
+#include <sys/socket.h>
+#include <unistd.h>
+#include <poll.h>
+#include <zephyr/sys_clock.h>
+#include <stdlib.h>
+#include <arpa/inet.h>
+#include <netinet/in.h>
+
 #else
 #include <zephyr/posix/sys/socket.h>
 #include <zephyr/posix/unistd.h>
diff --git a/include/zephyr/net/socketutils.h b/include/zephyr/net/socketutils.h
index 98bc8b3c58..d76bfe13a2 100644
--- a/include/zephyr/net/socketutils.h
+++ b/include/zephyr/net/socketutils.h
@@ -7,8 +7,11 @@
 #ifndef ZEPHYR_INCLUDE_NET_SOCKETUTILS_H_
 #define ZEPHYR_INCLUDE_NET_SOCKETUTILS_H_
 
-#ifdef CONFIG_NET_SOCKETS_POSIX_NAMES
-#include <zephyr/net/socket.h>
+#ifdef CONFIG_ARCH_POSIX
+#include <sys/socket.h>
+#include <stdlib.h>
+#include <stddef.h>
+#include <netdb.h>
 #else
 #include <zephyr/posix/netdb.h>
 #endif
diff --git a/samples/net/sockets/sntp_client/prj.conf b/samples/net/sockets/sntp_client/prj.conf
index 8b2265efd8..9ec3c66312 100644
--- a/samples/net/sockets/sntp_client/prj.conf
+++ b/samples/net/sockets/sntp_client/prj.conf
@@ -10,7 +10,7 @@ CONFIG_NET_SHELL=y
 
 # Sockets
 CONFIG_NET_SOCKETS=y
-CONFIG_NET_SOCKETS_POSIX_NAMES=y
+CONFIG_NET_SOCKETS_POSIX_NAMES=n
 CONFIG_NET_SOCKETS_POLL_MAX=4
 
 # Network driver config
diff --git a/subsys/net/lib/sntp/sntp.c b/subsys/net/lib/sntp/sntp.c
index b85fbe877a..bc3e573c1a 100644
--- a/subsys/net/lib/sntp/sntp.c
+++ b/subsys/net/lib/sntp/sntp.c
@@ -9,6 +9,7 @@
 LOG_MODULE_REGISTER(net_sntp, CONFIG_SNTP_LOG_LEVEL);
 
 #include <zephyr/net/sntp.h>
+#include <zephyr/net/net_core.h>
 #include "sntp_pkt.h"
 
 #define SNTP_LI_MAX 3
@@ -106,7 +107,7 @@ static int sntp_recv_response(struct sntp_ctx *sntp, uint32_t timeout,
 	int status;
 	int rcvd;
 
-	status = zsock_poll(sntp->sock.fds, sntp->sock.nfds, timeout);
+	status = poll(sntp->sock.fds, sntp->sock.nfds, timeout);
 	if (status < 0) {
 		NET_ERR("Error in poll:%d", errno);
 		return -errno;
@@ -116,7 +117,7 @@ static int sntp_recv_response(struct sntp_ctx *sntp, uint32_t timeout,
 		return -ETIMEDOUT;
 	}
 
-	rcvd = zsock_recv(sntp->sock.fd, (uint8_t *)&buf, sizeof(buf), 0);
+	rcvd = recv(sntp->sock.fd, (uint8_t *)&buf, sizeof(buf), 0);
 	if (rcvd < 0) {
 		return -errno;
 	}
@@ -150,15 +151,15 @@ int sntp_init(struct sntp_ctx *ctx, struct sockaddr *addr, socklen_t addr_len)
 
 	memset(ctx, 0, sizeof(struct sntp_ctx));
 
-	ctx->sock.fd = zsock_socket(addr->sa_family, SOCK_DGRAM, IPPROTO_UDP);
+	ctx->sock.fd = socket(addr->sa_family, SOCK_DGRAM, IPPROTO_UDP);
 	if (ctx->sock.fd < 0) {
 		NET_ERR("Failed to create UDP socket %d", errno);
 		return -errno;
 	}
 
-	ret = zsock_connect(ctx->sock.fd, addr, addr_len);
+	ret = connect(ctx->sock.fd, addr, addr_len);
 	if (ret < 0) {
-		(void)zsock_close(ctx->sock.fd);
+		(void)close(ctx->sock.fd);
 		NET_ERR("Cannot connect to UDP remote : %d", errno);
 		return -errno;
 	}
@@ -186,7 +187,7 @@ int sntp_query(struct sntp_ctx *ctx, uint32_t timeout, struct sntp_time *time)
 	ctx->expected_orig_ts = get_uptime_in_sec() + OFFSET_1970_JAN_1;
 	tx_pkt.tx_tm_s = htonl(ctx->expected_orig_ts);
 
-	ret = zsock_send(ctx->sock.fd, (uint8_t *)&tx_pkt, sizeof(tx_pkt), 0);
+	ret = send(ctx->sock.fd, (uint8_t *)&tx_pkt, sizeof(tx_pkt), 0);
 	if (ret < 0) {
 		NET_ERR("Failed to send over UDP socket %d", ret);
 		return ret;
@@ -198,6 +199,6 @@ int sntp_query(struct sntp_ctx *ctx, uint32_t timeout, struct sntp_time *time)
 void sntp_close(struct sntp_ctx *ctx)
 {
 	if (ctx) {
-		(void)zsock_close(ctx->sock.fd);
+		(void)close(ctx->sock.fd);
 	}
 }
diff --git a/subsys/net/lib/sntp/sntp_simple.c b/subsys/net/lib/sntp/sntp_simple.c
index 9399f578f8..873dff80dc 100644
--- a/subsys/net/lib/sntp/sntp_simple.c
+++ b/subsys/net/lib/sntp/sntp_simple.c
@@ -5,6 +5,7 @@
  */
 #include <errno.h>
 
+#include <zephyr/kernel.h>
 #include <zephyr/net/sntp.h>
 #include <zephyr/net/socketutils.h>
 
diff --git a/subsys/net/lib/utils/addr_utils.c b/subsys/net/lib/utils/addr_utils.c
index f6b7dc38b2..bf715f3759 100644
--- a/subsys/net/lib/utils/addr_utils.c
+++ b/subsys/net/lib/utils/addr_utils.c
@@ -7,6 +7,8 @@
 #ifdef CONFIG_NET_SOCKETS
 
 #include <zephyr/net/socketutils.h>
+#include <errno.h>
+#include <string.h>
 
 const char *net_addr_str_find_port(const char *addr_str)
 {
@@ -35,7 +37,11 @@ int net_getaddrinfo_addr_str(const char *addr_str, const char *def_port,
 			     struct addrinfo **res)
 {
 	const char *port;
+#ifdef NI_MAXHOST
 	char host[NI_MAXHOST];
+#else
+	char host[64];
+#endif
 
 	if (addr_str == NULL) {
 		errno = EINVAL;
-- 
2.34.1
```

cfriedt, 2023-04-21:
@heinwessels - are you able to open a PR?

heinwessels, 2023-04-21:
> Also major kudos making the leap from v2.6.0 to v3.4.0

Thanks! It's definitely not been easy, but this is the last hurdle I believe. Want to jump onto the next LTS where most of our functionality is in upstream Zephyr.

And I created the PR. Added you as co-author since you gave me the starting code. Not sure if that's the correct etiquette, so please feel to tell me if I should remove your name from the commit.

cfriedt, 2023-04-22:
> > Also major kudos making the leap from v2.6.0 to v3.4.0
> 
> Thanks! It's definitely not been easy, but this is the last hurdle I believe. Want to jump onto the next LTS where most of our functionality is in upstream Zephyr.

Would be great if you had a few notes to share about that experience. An LTSv2 to LTSv3 migration guide is still very much in the planning stages cc @kartben

heinwessels, 2023-04-24:
> Would be great if you had a few notes to share about that experience. An LTSv2 to LTSv3 migration guide is still very much in the planning stages cc @kartben

I don't mind sharing if there's an existing document and everything, but don't currently have the time create such a migration guide. Most of our time was only spent moving our changes from our fork into upstream though, so also don't know how helpful that would be. :)

PR #57144 (2023-04-24): net: sntp: support building without NET_SOCKETS_POSIX_NAMES
Fixes the compiler error that occurs when attempting to build the SNTP library with CONFIG_NET_SOCKETS_POSIX_NAMES=n, while using native_posix.

Closes #57133
commit 2689e166 (2023-04-24): net: sntp: support building without NET_SOCKETS_POSIX_NAMES

Fixes the compiler error that occurs when attempting to build
the SNTP library with CONFIG_NET_SOCKETS_POSIX_NAMES=n, while
using native_posix.

Closes #57133
closed 2023-04-24 by cfriedt