zephyrproject-rtos/zephyr #57138: Automate find missing SPDX tags in source files

romkell, 2023-04-21:
## Introduction

Running the SPDX tool with west on a build target generating the SPDX report revealed, that not all source files were assigned a license.
At a closer look, source files (from an external source, HAL) had a license statement in the file but no SPDX tag.

According to

https://github.com/zephyrproject-rtos/zephyr/issues/new?assignees=&labels=TSC&template=ext-source.md&title=

also external sources shall be SPDX tagged.

### Problem description

We have customers demanding prove of only using permissive licensed code (no GPL), if we use open source code in our products that we sell them, in order not to have to disclose the full sources.
The prove would be the SPDX report for a build firmware. If that has major gaps due to missing SPDX tag in the source files, the SPDX report looses its value.

### Proposed change

Since source files slipped in, the CI should run a check, that all sources carry the SPDX tag and that the license is a permissive kind.

## Detailed RFC

- Configure the CI to run SPDX over the full code base also including git sub modules (HAL, tensor flow etc.).
- Write a script evaluating the SPDX report extracting the un-tagged files, creating a "missing SPDX report" and one or several bug issue(s) (split by maintainer or module or "tbd")
- The CI check frequency could be weekly 

### Proposed change (Detailed)

See "Detailed RFC"

### Dependencies

- None.

### Concerns and Unresolved Questions

- None

## Alternatives

- None

nashif, 2023-05-01:
> According to
> 
> [zephyrproject-rtos/zephyr/issues/new?assignees=&labels=TSC&template=ext-source.md&title=](https://github.com/zephyrproject-rtos/zephyr/issues/new?assignees=&labels=TSC&template=ext-source.md&title=)
> 
> also external sources shall be SPDX tagged.

Where does it say that? The ask in the template is to provide the license name of the module in form of an SPDX identifier, it is not asking for the source files to be SPDX tagged.

romkell, 2023-05-02:
> Where does it say that? The ask in the template is to provide the license name of the module in form of an SPDX identifier, it is not asking for the source files to be SPDX tagged.

>## License
>
>Please use an SPDX identifier (https://spdx.org/licenses/), such as
``BSD-3-Clause``

It does not explicitly demand to tag each file, as it is written, it rather is a polite request to do so.

At least to 

https://spdx.dev/ids/

the IDs were meant to be used with the `SPDX-License-Identifier:` field ID in the actual documents/files itself.

That is how I understand https://spdx.dev/ids/ and therefore the ask in the template.

Looking at the hal_stm32 (just the micro that we happen use in that customer project), there is a central license file naming 2 of those SPDX IDs somewhere in the free text. The sources referring to that central license file.

```  
  * Copyright (c) 2022 <vendor>.
  * All rights reserved.
  *
  * This software is licensed under terms that can be found in the LICENSE file
  * in the root directory of this software component.
  * If no LICENSE file comes with this software, it is provided AS-IS.
  *
  ******************************************************************************
  */
```

This is just one example and I do not know if that is the general approach how it is done by this vendor also in his environment or if that was invented for the use as an zephyrproject-rtos module.

Nevertheless generating an SPDX for a built binary, created some 30 - 40% empty license information and therefore makes the report near to useless.
In Discord before the RFC, I suggested to add a mechanism with a central license info file per module (see copied text above).
I see the following solutions:

1. do nothing - then the quality of the SPDX report high depends on what of the eco system you use
2. have a non-standard add-on mechanism implemented in our version of SPDX tool - as suggested in Discord and copied in this RFC, feels a bit like adding a zephyr specific balcony to SPDX and the SPDX tool.
3. Ask (demand) the module source be tagged properly - can we really demand that? Possibly for the module in zephyrproject-rtos, but rather not upstream)
4. provide a script as a supporting tool (basically find/replace), that can add a proper SPDX tag at a given location in the source files for the downstream sources in zephyrproject-rtos - no 100% automation, since rather error prone, rather only support to the module maintainer to easily distribute such tags, still manual checking required.

The RFC asks for a CI supported automated verification that for all source files a license is found by the SPDX tool, to identify gaps.

We possibly need to do one step back and first clarify how to handle SPDX tagging specially for zephyrproject-rtos modules in general.

nashif, 2023-05-02:
> That is how I understand [spdx.dev/ids](https://spdx.dev/ids/) and therefore the ask in the template.

the ask in the template is to specify a license, and to be accurate, use the SPDX ID of the license instead of some free form, there is no requirement for modules to tag have every file tagged with SPDX.

romkell, 2023-05-03:
> the ask in the template is to specify a license, and to be accurate, use the SPDX ID of the license instead of some free form, there is no requirement for modules to tag have every file tagged with SPDX.

What ever the template is asking for, we end up, and that is the current status, with an SPDX tool resp. the output of it that is highly incomplete, dependent on which modules you use. The options I see are listed above.
The question is do we want to leave it in that state or do we want to improve it, because you cannot show that your application does not use non-permissive licensed code. 
One can do all by hand, but then the whole SPDX tool does not make any sense.

nashif, 2023-05-05:
@romkell  The point I am trying to make is that modules were never required to have SPDX headers in licences. The license of the module in general had to be compliant with our license requirements and there was no demand to have SPDX header in each file.

You proposed change:
> Proposed change
>> Since source files slipped in, the CI should run a check, that all sources carry the SPDX tag and that the license is a >> permissive kind.

can only apply to the zephyr code. Our license checker actually requires SPDX lisence to be present, so this is the first line of deference we have, if items were missed, we need to address those individualy and running SPDX tools on the whole tree is something we do AFAIK per release and can be extended to run mo frequently to audit the state of the tree.

Asking module owners to have SPDX headers in their code would be a new thing that we need to agree on and document and make it part of the acceptance criteria, so I would propose this as a separate item with the caveat that we also need to address all existing modules we already have as part of the project.

romkell, 2023-05-05:
@nashif 
> @romkell The point I am trying to make is that modules were never required to have SPDX headers in licences. The license of the module in general had to be compliant with our license requirements and there was no demand to have SPDX header in each file.

Fully understood.
Consequence of it: SPDX run and report of a build firmware will, dependent on which modules the build used, have some 30- 40% file with no license information.

The RFC asks for a CI supported automated verification that for all source files a license is found by the SPDX tool, to identify gaps. I feel, for the zephyr repo that is good enough. No need to increase frequency or anything.

The issue here is that
1. the modules do not necessarily carry SPDX tags in the source (since not requested/demanded)
2. the SPDX tool only supports that one way to identify the files license

We possibly need to do one step back and first clarify how to handle SPDX tagging specially for zephyrproject-rtos modules in general.

I see the following solutions (copied from above):
1. do nothing - then the quality of the SPDX report high depends on what of the eco system you use
2. have a non-standard add-on mechanism implemented in our version of SPDX tool - as suggested in Discord and copied in this RFC, feels a bit like adding a zephyr specific balcony to SPDX and the SPDX tool.
3. Ask (demand) the module source be tagged properly - can we really demand that? Possibly for the module in zephyrproject-rtos, but rather not upstream)
    3.1. provide a script as a supporting tool (basically find/replace), that can add a proper SPDX tag at a given location in the source files for the downstream sources in zephyrproject-rtos - no 100% automation, since rather error prone, rather only support to the module maintainer to easily distribute such tags, still manual checking required.

I see this RFC rather as bringing that circumstance to TSC attention, start a discussion, find a solution and close it.
If it required to reword the issue because it requests to "Automate find missing SPDX tags in source files", I can do so.

nashif, 2023-05-10:
removing TSC label for now, if you want to bring this back to the TSC, please relabel.

romkell, 2024-02-28:
> @carlescufi to look into using Nordic's manifest plugin, still TBD.

@carlescufi: Any decision on whether Nordic management agreed to open source your SPDX extension evaluating the license of not SPDX tagged sources?

nashif, 2024-06-05:
@carlescufi  any updates.