zephyrproject-rtos/zephyr #57174: STM32C0 - Add i2c support

Ereaey, 2023-04-24:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**
Current stm32c0 SOC description does not implement I2C

**Describe why you are asking for this support?**
It is a missing element on a platform which will be used commonly in the future and which is partially implemented.

**Additional context**
[Documentation](https://www.st.com/en/evaluation-tools/nucleo-c031c6.htm)

erwango, 2023-04-24:
@benjaminbjornsson I guess this is still in your plans ?

benjaminbjornsson, 2023-04-25:
> @benjaminbjornsson I guess this is still in your plans ?

It is, I’ll try to make room for testing the I2C and SPI this week. I also have a branch with DMA supported.

Ereaey, 2023-05-15:
@benjaminbjornsson Any news for this subject ? Thanks for your help

erwango, 2023-05-16:
@Ereaey Would you mind testing #57905 and provide a feedback ?

Ereaey, 2023-05-17:
I will try to look once I have managed to make my configuration for my custom board with a stm32c031k6t6.

For the moment, I have problems with the usart (with hsi)
```
LL_USART_IsActiveFlag_TXE_TXFNF (USARTx=<error reading variable: Cannot access memory at address 0x8005008>) at /home/.../modules/hal/stm32/stm32cube/stm32c0xx/drivers/include/stm32c0xx_ll_usart.h:3021
3021      return ((READ_BIT(USARTx->ISR, USART_ISR_TXE_TXFNF) == (USART_ISR_TXE_TXFNF)) ? 1UL : 0UL);

```

benjaminbjornsson, 2023-05-25:
Thanks for the help with testing @Ereaey

closed 2023-05-25 by benjaminbjornsson