zephyrproject-rtos/zephyr #57180: out-of-tree config to save time in CI/CD Pipelines

hf81-ble, 2023-04-24:
**Is your feature request related to a problem? Please describe.**
I followed the example on https://github.com/zephyrproject-rtos/example-application

But all the zephyr "stuff" will be in the project for that exclusive. If i will build multiple project i will have a plenty of storage waste receiving again and again at the same zephyr stuff into the each project separately. 

on my local and CI buildsystem  the will be in the order like here:

|->zepyhr SDK
|->zephyr (for example. 3.3.0)
|->project A
|->project B
|->project C

**Describe the solution you'd like**
each project should be in a stand-a-lone git repo. But they share the same version of zephyr on the build process. I want to set a local relative west config for each project. In that case ../zephyr but is could also be ../zephyr33 ../zephyr31 if they are based on different zephyr version.  

I dont want to have all the projects into one big container repo. 

**Describe alternatives you've considered**
there is maybe already a solution for that, but i didn't find in with my words in the documentation. Maybe any hints would already solve my issue.

**Additional context**

nordicjm, 2023-04-24:
Did you look up a freestanding application? https://docs.zephyrproject.org/latest/develop/application/index.html#zephyr-freestanding-app

hf81-ble, 2023-04-24:
yes, but how to set the west config or manifest to that project? 

```
west: error: argument <command>: invalid choice: 'build' (choose from 'init', 'update', 'list', 'manifest', 'diff', 'status', 'forall', 'help', 'config', 'topdir', 'selfupdate')
```

hf81-ble, 2023-04-24:
i got the solution from here: 

https://devzone.nordicsemi.com/f/nordic-q-a/89008/west-build-is-invalid-choice

before west build 

`export ZEPHYR_BASE=/opt/nordic/ncs/v2.0.0/zephyr`

i add this to my build scripts and everything is fine!

closed 2023-04-24 by hf81-ble